arendst/Tasmota #10673: GPIO "Not supported"  - gpio fail on devices configured with template

majherek, 2021-01-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff MINI
- [x] Tasmota binary firmware version number used: 9.2.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: tismotizer, OTA + reset 2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
cmnd/sonoff15/Backlog Template; Module; GPIO 255
stat/sonoff15/RESULT {"NAME":"Sonoff Mini","GPIO":[32,0,0,0,160,0,0,0,224,320,0,0,1,0],"FLAG":0,"BASE":1}
stat/sonoff15/RESULT {"Module":{"0":"Sonoff Mini"}}
stat/sonoff15/RESULT {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"33":"Button2"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
cmnd/sonoff15/status 0
stat/sonoff15/STATUS {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"sonoff15","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
stat/sonoff15/STATUS1 {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://sonoff-ota.majchrowski.waw.pl:8888/api/arduino/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:55","StartupUTC":"2021-01-23T11:32:22","Sleep":50,"CfgHolder":4617,"BootCount":3,"BCResetTime":"2021-01-23T12:09:50","SaveCount":9,"SaveAddress":"FB000"}}
stat/sonoff15/STATUS2 {"StatusFWR":{"Version":"9.2.0.1(sensors)","BuildDateTime":"2021-01-22T21:18:17","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"434/699"}}
stat/sonoff15/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.199.251","LogPort":514,"SSId":["atomix","htconeM7"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000002C0","00206000","00000000"]}}
stat/sonoff15/STATUS4 {"StatusMEM":{"ProgramSize":677,"Free":324,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA87C7","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
stat/sonoff15/STATUS5 {"StatusNET":{"Hostname":"sonoff15","IPAddress":"192.168.20.15","Gateway":"192.168.20.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.199.251","Mac":"D8:F1:5B:8E:3C:A8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
stat/sonoff15/STATUS6 {"StatusMQT":{"MqttHost":"mqtt.majchrowski.waw.pl","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8E3CA8","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
stat/sonoff15/STATUS7 {"StatusTIM":{"UTC":"2021-01-23T11:42:17","Local":"2021-01-23T12:42:17","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:29","Sunset":"16:05"}}
stat/sonoff15/STATUS10 {"StatusSNS":{"Time":"2021-01-23T12:42:17","Switch1":"ON"}}
stat/sonoff15/STATUS11 {"StatusSTS":{"Time":"2021-01-23T12:42:17","Uptime":"0T00:09:55","UptimeSec":595,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"atomix","BSSId":"18:8B:45:08:B4:C1","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
12:42:53.201 MQT: stat/sonoff15/RESULT = {"WebLog":4}
12:42:53.705 CFG: Saved to flash at FA, Count 10, Bytes 4096
12:42:55.592 CMD: gpio
12:42:55.595 SRC: WebConsole from 192.168.199.251
12:42:55.598 CMD: Group 0, Index 1, Command "GPIO", Data ""
12:42:55.608 MQT: stat/sonoff15/RESULT = {"GPIO":"Not supported"}
```

### TO REPRODUCE
On every device configured by template when you perform:
```lua
reset 2
backlog template; module 0
gpio
```
E.g.:
```lua
majherek@atom:~$ mosquitto_pub -h mqtt -u sonoff -P PASS  -p 8883 --capath /etc/ssl/certs -t "cmnd/sonoff15/reset" -m 2
majherek@atom:~$ mosquitto_pub -h mqtt -u sonoff -P PASS -p 8883 --capath /etc/ssl/certs -t "cmnd/sonoff15/backlog" -m 'template {"NAME":"Sonoff Mini","GPIO":[17,0,0,0,9,0,0,0,21,56,0,0,255],"FLAG":0,"BASE":1}; module 0'
majherek@atom:~$ mosquitto_pub -h mqtt -u sonoff -P PASS -p 8883 --capath /etc/ssl/certs -t "cmnd/sonoff15/gpio" -n
```
And the results:
```lua
tele/sonoff15/LWT Online
cmnd/sonoff15/reset 2
stat/sonoff15/RESULT {"Reset":"Erase, Reset and Restarting"}
[... 37 lines cut ...]
I can change the GPIO in Web UI, from drop down menu.
I can't see correct result of command GPIO through MQTT or console using GPIO command.

I can change the GPIO16 on sonoff MINI, but I get still "Not supported" message.

```lua
cmnd/sonoff15/gpio16 33
stat/sonoff15/RESULT {"GPIO":"Not supported"}
tele/sonoff15/LWT Offline
tele/sonoff15/LWT Online
cmnd/sonoff15/POWER (null)
tele/sonoff15/INFO1 {"Module":"Sonoff Mini","Version":"9.2.0.1(sensors)","FallbackTopic":"cmnd/DVES_8E3CA8_fb/","GroupTopic":"cmnd/tasmotas/"}
tele/sonoff15/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff15","IPAddress":"192.168.20.15"}
tele/sonoff15/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff15/RESULT {"POWER":"ON"}
stat/sonoff15/POWER ON
tele/sonoff15/STATE {"Time":"2021-01-23T12:32:35","Uptime":"0T00:00:13","UptimeSec":13,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"atomix","BSSId":"18:8B:45:08:B4:C1","Channel":1,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:07"}}
tele/sonoff15/SENSOR {"Time":"2021-01-23T12:32:35","Switch1":"ON"}
cmnd/sonoff15/gpio all
stat/sonoff15/RESULT {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"160":"Switch1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"33":"Button2"},"GPIO17":{"0":"None"}}
cmnd/sonoff15/gpio (null)
stat/sonoff15/RESULT {"GPIO":"Not supported"}
```

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2021-01-23:
I've tested a little more and I can confirm that the command `gpio` fails reporting if GPIO17 (ADC) is set to None. If it i sset to USer, then every thing works as expected.
I haven't yet looked into the code to fully understand the reason

Only reporting is failing. In th eabove use case, trying to change gpio16 by `gpio16 33` does change the configuration and restart, but the command still reports "Not Supported".
`gpio` without any argument should report all pins listed as 'User' in the template but doesn't, unless GPIO17 is set to 'User'

majherek, 2021-01-23:
I set GPIO17 to 1 in template (NEW style) and now GPIO command work...
```lua
cmnd/sonoff15/reset 2
stat/sonoff15/RESULT {"Reset":"Erase, Reset and Restarting"}
tele/sonoff15/LWT Offline
tele/sonoff15/LWT Online
cmnd/sonoff15/POWER (null)
tele/sonoff15/INFO1 {"Module":"Sonoff Basic","Version":"9.2.0.1(sensors)","FallbackTopic":"cmnd/DVES_8E3CA8_fb/","GroupTopic":"cmnd/tasmotas/"}
tele/sonoff15/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff15","IPAddress":"192.168.20.15"}
tele/sonoff15/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff15/RESULT {"POWER":"OFF"}
stat/sonoff15/POWER OFF
cmnd/sonoff15/backlog template {"NAME":"Sonoff Mini","GPIO":[32,0,0,0,160,0,0,0,224,320,0,0,1,1],"FLAG":0,"BASE":1}; module 0
stat/sonoff15/RESULT {"NAME":"Sonoff Mini","GPIO":[32,0,0,0,160,0,0,0,224,320,0,0,1,1],"FLAG":0,"BASE":1}
stat/sonoff15/RESULT {"Module":{"0":"Sonoff Mini"}}
tele/sonoff15/LWT Offline
tele/sonoff15/LWT Online
cmnd/sonoff15/POWER (null)
tele/sonoff15/INFO1 {"Module":"Sonoff Mini","Version":"9.2.0.1(sensors)","FallbackTopic":"cmnd/DVES_8E3CA8_fb/","GroupTopic":"cmnd/tasmotas/"}
tele/sonoff15/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff15","IPAddress":"192.168.20.15"}
tele/sonoff15/INFO3 {"RestartReason":"Software/System restart"}
stat/sonoff15/RESULT {"POWER":"OFF"}
stat/sonoff15/POWER OFF
cmnd/sonoff15/gpio (null)
stat/sonoff15/RESULT {"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

ascillato2, 2021-01-24:
Closing this issue as it has been fixed.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-24 by ascillato2