arendst/Tasmota #10679: Exception 28 on Novostella B22 bulbs and Shelly 1 on 9.2.0.4

imduffy15, 2021-01-24:
### PROBLEM DESCRIPTION

I did a mass upgrade of ~20 Novostella B22 bulbs and 4 Shellys after the update completed they all started rebooting over and over again stating Exception 28

I have managed to recover all devices without issue, I hammered all devices with:

```
until curl http://{}/cm?cmnd=Reset%206; do echo 'retrying {}'; done
```

Which allowed them to stay online, on configuring them again though they went back to throw Exception 28 again. Ended up restoring the firmware to 9.2.0 from the github releases page and then restoring config.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Novostella B22 and Shelly 1
- [x] eTasmota binary firmware version number used: 9.2.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA / Tuya Convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
Device already recovered so can't provide
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
Device already recovered so can't provide
```
- [ ] Provide the output of this command: `Status 0`:
```lua
Device already recovered so can't provide
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
Device already recovered so can't provide
```

### TO REPRODUCE

Upgrade a Shelly1 or Novostella B22 bulb to 9.2.0.4

### EXPECTED BEHAVIOUR

The upgrade goes successfully and the device does not reboot with Exception 28

imduffy15, 2021-01-24:
I have some devices that are not in use, I'll test further on this tomorrow, its very late for me at the moment and just trying to recover the house before going to bed.

imduffy15, 2021-01-24:
Looks like my Athom LB017W are doing this too, managed to grab the logs before my auto recover process hit it:

```
01:33:13.123 MQT: kitchen_light/tele/INFO3 = {"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40243b40","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40249f30","40243b3c","40249f30","4028a8d4","4024c6d8","4022932f","4024cf50","4024c4ee","40289bdb","40289bdb","4024bc8d","40243f5d","4024bcd0","4020338c","4024bc8d","4024bcd0","4024bc8d","4024bd44","402077bc","40207911","40101be1","40100020","4024bd54","40207a3f","40240064","401018db","40249f30","40249f30","402864ff","40289c74","4022eecc"]}}
```

imduffy15, 2021-01-24:
It seems to be related to having devicegroups (setoption85) enabled.

arendst, 2021-01-24:
Looking at the code of the latest device group change https://github.com/arendst/Tasmota/pull/10664 I think there is an issue regarding wrong parameter count resulting in this exception.

@pcdiem pls have a look.

arendst, 2021-01-24:
Thx for pointing in the correct direction.

For some reason the change from #10616 leads to an exception. This change has been reverted for the Device Group driver.

closed 2021-01-24 by ascillato2

imduffy15, 2021-01-24:
Wow so cool! Thank you for maintaining this and having quick actions to fix
things.

pcdiem, 2021-01-24:
I missed this thread but was already working on a fix for this after I pulled the latest dev. The issue is related to the IPAddress.tostring() method when invoked on the device group member structure. I'm not sure if it's because the structure is allocated from dynamic pool or not. My approach to the fix was to copy the IPAddress to stack space before invoking the tostring() method.

arendst, 2021-01-24:
Paul don't bother anymore.

I've restored previous functionality and today we also introduced a new snprintf function able to parse any uint32_t as an ip address. The new function is called ext_vsnprintf_P and it uses %_I as ip address replacement. It is already the default for AddLog and AddLog_P functions so give it a try.

arendst, 2021-01-27:
Update. 

As the new function adds to the code size I decided to keep the current solution using the original ``IPAddressToString()`` function.