arendst/Tasmota #10683: Inconsistent iFan03 RESULT for cmnd/tasmotas/SetOption19

bilogic, 2021-01-25:
### PROBLEM DESCRIPTION
1. When I send `cmnd/tasmotas/SetOption19`, value 0, my `Sonoff iFan03 (71)`, returns a response that alternates between A and B below.
2. I gone through all my settings and could not anything that might be causing this behavior.
3. When it responds with B, my Home Assistant becomes unable to control the light (`POWER1`)

Response A,
```
{
    "Time": "2021-01-25T18:55:19",
    "Uptime": "1T04:58:34",
    "UptimeSec": 104314,
    "Heap": 26,
    "SleepMode": "Dynamic",
    "Sleep": 50,
    "LoadAvg": 19,
    "MqttCount": 5,
    "POWER1": "OFF",
    "FanSpeed": 2,
    "Wifi": {
        "AP": 1,
        "SSId": "*******",
        "BSSId": "**:**:**:**:**:**",
        "Channel": 1,
        "RSSI": 94,
        "Signal": -53,
        "LinkCount": 3,
        "Downtime": "0T00:01:41"
    }
}
```

Response B
```
{
    "FanSpeed":2
}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff iFan03
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:07:45 MQT: stat/tasmota_8FA7C0/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:07:45 MQT: stat/tasmota_8FA7C0/RESULT = {"Module":{"71":"Sonoff iFan03"}}
19:07:46 MQT: stat/tasmota_8FA7C0/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"256":"Relay_i1"},"GPIO10":{"512":"Buzzer_i"},"GPIO12":{"226":"Relay3"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"225":"Relay2"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
19:08:26 MQT: stat/tasmota_8FA7C0/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:08:26 MQT: stat/tasmota_8FA7C0/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
19:08:27 MQT: stat/tasmota_8FA7C0/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
[... 111 lines cut ...]
19:10:29 LOG: Home Assistant MQTT Discovery disabled.
19:10:29 MQT: tasmota/discovery/D8F15B8FA7C0/config = {"ip":"192.168.9.168","dn":"ifan03","fn":["Ceiling Light","Tasmota2","Tasmota3","Tasmota4",null,null,null,null],"hn":"tasmota_8FA7C0-1984","mac":"D8F15B8FA7C0","md":"Sonoff iFan03","ty":0,"if":1,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.2.0","t":"tasmota_8FA7C0","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[2,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":1,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
19:10:29 MQT: tasmota/discovery/D8F15B8FA7C0/sensors = {"sn":{"Time":"2021-01-25T19:10:29"},"ver":1} (retained)
19:10:31 MQT: Data Size 0, "cmnd/tasmota_8FA7C0/STATE "
19:10:31 SRC: MQTT
19:10:31 CMD: Group 0, Index 1, Command "STATE", Data ""
19:10:31 MQT: tele/tasmota_8FA7C0/STATE = {"Time":"2021-01-25T19:10:31","Uptime":"1T05:13:46","UptimeSec":105226,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":5,"POWER1":"OFF","FanSpeed":2,"Wifi":{"AP":1,"SSId":"rtfckut","BSSId":"78:24:AF:E4:F8:F0","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":3,"Downtime":"0T00:01:41"}}
19:10:31 MQT: stat/tasmota_8FA7C0/RESULT = {"Time":"2021-01-25T19:10:31","Uptime":"1T05:13:46","UptimeSec":105226,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":5,"POWER1":"OFF","FanSpeed":2,"Wifi":{"AP":1,"SSId":"rtfckut","BSSId":"78:24:AF:E4:F8:F0","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":3,"Downtime":"0T00:01:41"}}
19:10:48 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
I'm expecting the JSON to consistently be JSON A

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-01-25:
No problem there, "response A" is just the regular /STATE message (not /RESULT) published by at TelePeriod intervals. Use it if you like, or just ignore it.

ascillato2, 2021-01-25:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-01-25 by ascillato2

bilogic, 2021-01-25:
@sfromis sorry, I don't quite follow. I have other tasmota devices, they don't become `Entity not available` after I fire 'cmnd/tasmotas/SetOption19', '0' only once. Is there a way to make iFan03 behave the same? Thank you.

sfromis, 2021-01-25:
Well I answered to your point about "inconsistent", which it is not. No issue needing fixing. For help understanding how it works with HA (which I'm not using), better ask in the "Chat" link posted above by ascillato2.

bilogic, 2021-01-25:
Hmm, but I do have a /RESULT with response A. See the last 2nd line of from the `weblog` = `4` output.
```
19:10:31 MQT: stat/tasmota_8FA7C0/RESULT = {"Time":"2021-01-25T19:10:31","Uptime":"1T05:13:46","UptimeSec":105226,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":5,"POWER1":"OFF","FanSpeed":2,"Wifi":{"AP":1,"SSId":"rtfckut","BSSId":"78:24:AF:E4:F8:F0","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":3,"Downtime":"0T00:01:41"}}
```