arendst/Tasmota #10696: high load with ESP32 cam

andreasloe, 2021-01-25:
### PROBLEM DESCRIPTION
I am experiencing high load with tasmota webcam although no other „fancy“ stuff is running - just continuous streaming to a synology drive.  LoadAvg | 1143 after one day uptime and increasing.

### REQUESTED INFORMATION

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): ESP32 Cam
- [X ] Tasmota binary firmware version number used: 9.2.0.4
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: esptool
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

22:16:39.357 MQT: stat/tasmota_F2BD54/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
22:16:39.587 MQT: stat/tasmota_F2BD54/RESULT = {"Module":{"1":"ESP32-DevKit"}}
22:16:40.470 MQT: stat/tasmota_F2BD54/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"}}
22:16:40.473 MQT: stat/tasmota_F2BD54/RESULT = {"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: 

```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:18:06.679 MQT: stat/tasmota_F2BD54/STATUS = {"Status":{"Module":1,"DeviceName":"ESPCam","FriendlyName":["ESPCam"],"Topic":"tasmota_F2BD54","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:18:06.683 MQT: stat/tasmota_F2BD54/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"":"cmnd/tasmotas/"}","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"1T06:47:41","StartupUTC":"2021-01-24T14:30:25","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-01-23T19:50:35","SaveCount":31}}
22:18:06.685 MQT: stat/tasmota_F2BD54/STATUS2 = {"StatusFWR":{"Version":"9.2.0.4(webcam)","BuildDateTime":"2021-01-23T17:22:11","Core":"1_0_5_RC6","SDK":"v3.3.4-402-gcd59d107b","CpuFrequency":240,"Hardware":"ESP32","CR":"430/699"}}
22:18:06.687 MQT: stat/tasmota_F2BD54/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLAN-12095",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
22:18:06.691 MQT: stat/tasmota_F2BD54/STATUS4 = {"StatusMEM":{"ProgramSize":1301,"Free":1984,"Heap":91,"PsrMax":4095,"PsrFree":3345,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","80004000","00200000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,50,81","Sensors":"1,2,3,5,6"}}
22:18:06.695 MQT: stat/tasmota_F2BD54/STATUS5 = {"StatusNET":{"Hostname":"tasmota_F2BD54-7508","IPAddress":"192.168.2.68","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"7C:9E:BD:F2:BD:54","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:18:06.697 MQT: stat/tasmota_F2BD54/STATUS6 = {"StatusMQT":{"MqttHost":"rasp.fritz.box","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F2BD54","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:18:06.701 MQT: stat/tasmota_F2BD54/STATUS7 = {"StatusTIM":{"UTC":"2021-01-25T21:18:06","Local":"2021-01-25T22:18:06","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:28","Sunset":"17:36"}}
22:18:06.705 MQT: stat/tasmota_F2BD54/STATUS10 = {"StatusSNS":{"Time":"2021-01-25T22:18:06"}}
22:18:06.709 MQT: stat/tasmota_F2BD54/STATUS11 = {"StatusSTS":{"Time":"2021-01-25T22:18:06","Uptime":"1T06:47:41","UptimeSec":110861,"Heap":91,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":1036,"MqttCount":2,"Wifi":{"AP":1,"SSId":"WLAN-12095","BSSId":"F2:9F:C2:A7:4D:00","Channel":1,"RSSI":98,"Signal":-51,"LinkCount":2,"Downtime":"0T00:01:01"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
just run ESP32 and provide stream continuously

### EXPECTED BEHAVIOUR
shouldn‘t the load be below 100?

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-01-25:
Yes, that's as expected. Webcam is a very strenuous task for ESP32. Ensure good power + cooling, otherwise reliability tends to suffer, in my experience.

ascillato2, 2021-01-25:
> shouldn‘t the load be below 100?

No, it is not a percentage. If you are not experiencing any issue, don't pay attention to that value. It is for debug only. In my ESPCAM is 320 at this moment with just one mjpeg connection from my NVR.

closed 2021-01-26 by ascillato2