arendst/Tasmota #10698: Mqtt connection failing on long password

rmoosa, 2021-01-25:
### PROBLEM DESCRIPTION
I am connecting to the adafruit.io Mqtt server. The password is 32 characters long. 
Connection works perfectly on 9.1.0. But fails to connect on subsequent versions. 
After connection failed on latest build, I downgraded to 9.1.0 and it connected successfully again. 
Tested on 2 separate devices. Same issue. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): sonoff pow and sonoff th
- [ ] Tasmota binary firmware version number used: 9.2.0.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: firmware update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade to latest build and connect to Mqtt server using 32 character long password. 

### EXPECTED BEHAVIOUR
_connect sussessfully to Mqtt server_

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-01-25:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you? Please also provide all the logs.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2021-01-27:
Hi, any news on this?

arendst, 2021-01-27:
I don't think the password lenght is the issue here. 

Searching the changelog it seems since v9.1.0 the following two MQTT related changes have been introduced:

- v9.1.0.2: MQTT Wifi connection timeout from 5000 to 200 mSec (#9886)
- v9.2.0.3: PubSubClient MQTT_SOCKET_TIMEOUT from 15 to 4 seconds

As your issue starts with v9.2 I suggest you try to change back the Wifi connection timeout from 200 mSec to 5000 mSec and see if it solves your issue. If soo try to lower it again (as it was way too high for most users) to find a workable solution for Adafruit MQTT.

ascillato2, 2021-01-29:
@rmoosa

Hi, any news on this?

harryd31, 2021-01-29:
Running into the same issue when trying to connect to aws iot also

ascillato, 2021-01-29:
@harryd31

Please, help us solving this issue you are experiencing. Can you try as Theo has explained above?

rmoosa, 2021-01-29:
Apologies for taking so long to respond. Life came in the way of tinkering. 
It is unlikely to be the wifi timeout, cause the device is successfully connected to the wifi. 
I am able to get to the device on my network. 
And the adafruit Mqtt connection is normally established in less than 1 second, so an Mqtt socket timeout of 4 secs also should have any impact. 
I have not compiled this code myself yet, so gonna do some reading and will try the changes suggested.

rmoosa, 2021-01-29:
I have added all of the missing data on the template.

PS thanks for all the efforts in creating and maintaining this great software.

rmoosa, 2021-01-29:
This is the log after a restart:

00:00:00.049 CFG: Loaded from flash at FB, Count 500
00:00:00.060 Project tasmota TasmotaPower Version 9.2.0.4(tasmota)-2_7_4_9(2021-01-29T15:17:55)
00:00:00.163 SER: Set to 8E1 4800 bit/s
00:00:00.166 SNS: Hardware Serial
00:00:00.772 WIF: Connecting to AP1 myWiFly Channel 1 BSSId 18:E8:29:A7:60:7E in mode 11n as TasmotaPower...
00:00:01.539 WIF: Connected
00:00:01.756 HTP: Web server active on TasmotaPower with IP address 192.168.1.220
19:35:25.576 MQT: Attempting connection...
19:35:25.799 MQT: Connect failed to io.adafruit.com:1883, rc -2. Retry in 10 sec
19:35:36.754 MQT: Attempting connection...
19:35:36.970 MQT: Connect failed to io.adafruit.com:1883, rc -2. Retry in 20 sec
19:35:55.143 RSL: rmoosa/feeds/PowerMeter/STATE = {"Time":"2021-01-29T19:35:55","Uptime":"0T00:00:34","UptimeSec":34,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"myWiFly","BSSId":"18:E8:29:A7:60:7E","Channel":1,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:35:55.152 RSL: rmoosa/feeds/PowerMeter/SENSOR = {"Time":"2021-01-29T19:35:55","ENERGY":{"TotalStartTime":"2020-10-01T05:55:45","Total":23.196,"Yesterday":3.333,"Today":6.174,"Period":11,"Power":1330,"ApparentPower":1380,"ReactivePower":369,"Factor":0.96,"Voltage":228,"Current":6.057}}
19:35:55.164 RUL: TELE-ENERGY#POWER performs "publish rmoosa/feeds/PowerMeter.wattage 1330"
19:35:55.168 RSL: rmoosa/feeds/PowerMeter.wattage = 1330
19:35:57.753 MQT: Attempting connection...

rmoosa, 2021-01-29:
Could the issue have been caused by this FIX in 9.1.0.1:
GUI MqttUser and MqttPassword updates when TLS is compiled in (#9825)

rmoosa, 2021-01-31:
Hi. Why is is this issue marked as "duplicated". What is the duplicate ref number?

ascillato2, 2021-01-31:
> Hi. Why is is this issue marked as "duplicated". What is the duplicate ref number?

https://github.com/arendst/Tasmota/issues/6237
https://github.com/arendst/Tasmota/issues/3048
https://github.com/arendst/Tasmota/issues/1776
https://github.com/arendst/Tasmota/issues/3257
https://github.com/arendst/Tasmota/issues/1767
https://github.com/arendst/Tasmota/issues/1827
https://github.com/arendst/Tasmota/issues/1725
https://github.com/arendst/Tasmota/issues/958
https://github.com/arendst/Tasmota/issues/10163

harryd31, 2021-02-09:
@ascillato Sorry for the long wait, My issue was with connecting using the aws IOT, however I found the answer I was looking for. It wasn't a bug sorry.

ascillato2, 2021-02-09:
Great to know that now it is working for you.

rmoosa, 2021-02-09:
Hi. I logged this issue and it is still a problem. As per my initial description.

rmoosa, 2021-02-11:
@ascillato This is still an issue. 
I have tested on 9.2.0.7 and the issue still exists there. 
I then flashed the device with 9.1.0 without making any changes to any parameters and it started working fine. 
Can you please advise.

ascillato2, 2021-02-11:
Reopening as requested

arendst, 2021-02-11:
As said before (https://github.com/arendst/Tasmota/issues/10698#issuecomment-768141796) execute these changes and try again.

Trying to explain it (as you did before) without testing makes no sense.

jaymemaurice, 2021-02-13:
@rmoosa can you share more information about the MQTT server and any configuration you have on it? I think getting a packet capture may lead to some clues. Maybe I can attempt to reproduce..

rmoosa, 2021-02-13:
> @rmoosa can you share more information about the MQTT server and any configuration you have on it? I think getting a packet capture may lead to some clues. Maybe I can attempt to reproduce..

@jaymemaurice it is a public Mqtt broker. I don't have control over the settings. 
You can create a free adafruit io account here:
https://io.adafruit.com
Details on using the Mqtt api are here:
https://io.adafruit.com/api/docs/mqtt.html#adafruit-io-mqtt-api
Thanks for the assistance

barbudor, 2021-02-16:
@rmoosa I'm also interested to see if I can use Adafruit.IO for a specific use case
I will probably look into it in the next days.
Are you trying to connect insecure 1883 or with TLS ?
How are you expecting to work around Adafruit very specific topic structure ?
Except for rule-publish like I see you are doing, you won't be able to support tasmota's stat tele and cmnd prefixes. So no remote control.

rmoosa, 2021-02-17:
@barbudor hi. I'm connecting to port 1883. 
I only using the rule-publish. 
Mine was also a very specific use case - I just needed to collect sensor data. I did not need to control the device via Mqtt. 
I could probably also use a websend rule to adafruit api to accomplish the same.

barbudor, 2021-02-17:
OK. I'll do some tests by the end of the week and report back.

barbudor, 2021-02-17:
@rmoosa I can confirm immediate connection to AIO broker in unsecure mode (1883) using latest 9.2.0.7
[image]

logs : 
```
00:00:00.049 CFG: Loaded from flash at F7, Count 13
00:00:00.054 QPC: Count 1
00:00:00.056 CFG: CR 460/699, Busy 0
00:00:00.058 SRC: Restart
00:00:00.059 Project tasmota Tasmota Version 9.2.0.7(lite)-2_7_4_9(2021-02-17T17:38:00)
00:00:00.161 WIF: Attempting connection...
00:00:00.497 WIF: Connecting to AP1 ******** Channel 6 BSSId ******* in mode 11n as ************-5763...
00:00:01.751 WIF: Connected
00:00:02.003 HTP: Web server active on *******-5763 with IP address 192.168.169.195
00:00:03.537 RTC: UTC 2021-02-17T21:25:12, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
22:25:12.301 MQT: Attempting connection...
22:25:13.178 MQT: Connected
22:25:13.181 MQT: AIOUSERNAME/feeds/tele/LWT = Online (retained)
22:25:13.183 MQT: AIOUSERNAME/feeds/cmnd/POWER = 
22:25:13.184 MQT: Subscribe to AIOUSERNAME/feeds/cmnd/#
22:25:13.186 MQT: Subscribe to tasmotas/feeds/cmnd/#
22:25:13.187 MQT: Subscribe to cmnd/DVES_A6D683_fb/#
22:25:13.190 MQT: AIOUSERNAME/feeds/tele/INFO1 = {"Module":"Generic","Version":"9.2.0.7(lite)","FallbackTopic":"cmnd/DVES_A6D683_fb/","GroupTopic":"tasmotas/feeds/cmnd/"}
22:25:13.199 MQT: AIOUSERNAME/feeds/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"AIOUSERNAME-5763","IPAddress":"192.168.169.195"}
22:25:13.212 MQT: AIOUSERNAME/feeds/tele/INFO3 = {"RestartReason":"Software/System restart"}
22:25:15.449 QPC: Reset
22:25:17.448 APP: Boot Count 6
22:25:17.455 MQT: AIOUSERNAME/feeds/tele/STATE = {"Time":"2021-02-17T22:25:17","Uptime":"0T00:00:09","UptimeSec":9,"Vcc":3.050,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"BNETWLAN","BSSId":"F8:D1:11:BD:3B:A0","Channel":6,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}
22:25:18.346 CFG: Saved to flash at F6, Count 14, Bytes 4096
22:26:53.102 CMD: publish AIOUSERNAME/feeds/temp 21.24
22:26:53.104 SRC: WebConsole from 192.168.169.106
22:26:53.106 CMD: Group 0, Index 1, Command "PUBLISH", Data "AIOUSERNAME/feeds/temp 21.24"
22:26:53.110 MQT: AIOUSERNAME/feeds/temp = 21.24
```

Results in my feeds:
[image]

Of course errors on subscribing on topics that do not conform to AIO standard (errors shown on AIO web site:)
```
MQTT ERROR: DVES_A6D683 90.112.255.205 SUBSCRIBE to topic `tasmotas/feeds/cmnd/#` rejected, resource not available or not authorized
MQTT ERROR: DVES_A6D683 90.112.255.205 SUBSCRIBE to topic `cmnd/DVES_A6D683_fb/#` rejected, not a valid resource topic
```

And standard Tasmota messsages publications such as `22:25:13.199 MQT: AIOUSERNAME/feeds/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"AIOUSERNAME-5763","IPAddress":"192.168.169.195"}` are going nowhere as they don't conform to AIO standard.

But as you can see the "manual" publish goes correctly in my feeds

Now, need to test with TLS.

May be a specific AIO mode to conform Tasmota topics to AIO standard would be interresting.
I will investigate that after TLS.

rmoosa, 2021-02-18:
@barbudor can you ping the io.adafruit.com server from you network. 
I am based in South Africa. I wanna compare network latency with yours. 
Strangely enough though, yoI logs seem to indicate about 800 milliseconds to establish connection. Mine is failing in  less than 300. 
Perhaps I should try and test on an alternate network. Will try and post feedback.

barbudor, 2021-02-18:
Located in France, ADSL access in countryside location (around 4Mbps bw) 
Ping is around 120 to 140ms in general. Twice 620-640ms on 45 tests

rmoosa, 2021-02-18:
@barbudor my pings to that server are about 400ms. I'm going to try and recompile with new settings as previously suggested. Will test over the weekend.

barbudor, 2021-02-18:
There is one recent change regarding the TCP Timeout that is used when establishing the connection with the MQTT socket
The value was reduced from 15 seconds to 4 seconds
Still this doesn't seem to be the problem as your connection is rejected within 300ms.
I would guess this is more like an authentication problem. Are you using your AIOKEY as MQTT password (and not your account password) ?

I've just tested TLS. 1st time for me I tried TLS and it worked on 1st try too. So I strongly suggest that you use it too as transporting credential in clear on the Internet is not the safest.

rmoosa, 2021-02-18:
@barbudor i am using the correct auth details and settings. 
I downgraded to 9.1.0 without adjusting any of the settings and it starts working. 
Ap grade to 9.2.0.7 without changing any settings and it starts fails. 
Downgrade again to 9.1.0, again not making any changes to setting (username/password etc) and it starts failing again.

barbudor, 2021-02-18:
Downgrading may leads to unexpected behavior as flash mapping of settings have changed

Can you try 9.2.0.7 fresh (serial erase and flash and reconfigure from scratch) ?

Are you sure that after all those downgrades and upgrades your device still has proper access to the Internet and the DNS ?
When I want to check Internet access, my typical test sequence is:
`ping -4 google.com` on your pc, note the IP address (use ping -4 if needed to have an IPv4 address) and use it in the 1st test below :
In Tasmota console `websend [ip.of.google.com] /`  if replies "Done", you have Internet access, if not => fix it (gateway ?)
Now to test DNS, use google.com name:
In Tasmota console `websend [google.com] /`   if replies "Done" you have DNS, if not => fix it (use  1.1.1.1 or 8.8.8.8 for DNS for example)

rmoosa, 2021-02-18:
@barbudor thanks for that info. 
I will try that over the weekend.

jaymemaurice, 2021-02-26:
@rmoosa - did you get a chance to try again?

ascillato2, 2021-03-02:
Please, when you have time, tell us if you could solve your issue as suggested. Thanks.

closed 2021-03-02 by ascillato2