arendst/Tasmota #10699: Zigbee: Wrong argument leads to unexpected behavior.

CrazyMan2000, 2021-01-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When I send the command "zbsend { "Device":"0xA8E7", "send": {"power": "?"}} the zigbee device will turn off. 
It seems like Tasmota parses every invalid argument to 0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____ZBBridge
- [x] Tasmota binary firmware version number used: _____9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
01:01:04 MQT: stat/tasmota_E2A955/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E2A955","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:21","StartupUTC":"2021-01-25T23:51:43","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-01-05T15:39:33","SaveCount":277,"SaveAddress":"FA000"}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS2 = {"StatusFWR":{"Version":"9.2.0(zbbridge)","BuildDateTime":"2020-12-21T15:06:32","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"409/699"}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!BoxT","TP-Link"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS5 = {"StatusNET":{"Hostname":"tasmota_E2A955-2389","IPAddress":"192.168.2.150","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"98:F4:AB:E2:A9:55","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.83","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E2A955","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS7 = {"StatusTIM":{"UTC":"2021-01-26T00:01:04","Local":"2021-01-26T01:01:04","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:27","Sunset":"17:37"}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS10 = {"StatusSNS":{"Time":"2021-01-26T01:01:04"}}
01:01:04 MQT: stat/tasmota_E2A955/STATUS11 = {"StatusSTS":{"Time":"2021-01-26T01:01:04","Uptime":"0T00:09:21","UptimeSec":561,"Vcc":3.521,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":2,"SSId":"TP-Link","BSSId":"D8:47:32:60:C9:4A","Channel":4,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
01:05:33 CMD: zbsend { "Device":"0xA8E7", "send": {"power": "?"}}
01:05:33 SRC: WebConsole from 192.168.2.35
01:05:33 CMD: Group 0, Index 1, Command "ZBSEND", Data "{ "Device":"0xA8E7", "send": {"power": "?"}}"
01:05:33 ZIG: guessing endpoint 11
01:05:33 SendZCLCommand_P: zcl_cmd = 
01:05:33 ZigbeeZCLSend device: 0xA8E7, group: 0x0000, endpoint:11, cluster:0x0006, cmd:0x00, send:""
01:05:33 ZbSend: shortaddr 0xA8E7, groupaddr 0x0000, cluster 0x0006, endpoint 0x0B, cmd 0x00, data 
01:05:33 ZIG: ZbEZSPSend 340000E7A804010600010B40010000120103011200
01:05:33 ZIG: adding packet to_send, to_ack:6, to_send:6, to_end:6
01:05:33 MQT: stat/tasmota_E2A955/RESULT = {"ZbSend":"Done"}
01:05:33 ZIG: Something to_send, to_ack:6, to_send:6, to_end:7
01:05:33 ZIG: ZbEZSPSentRaw 632E0001340000E7A804010600010B40010000120103011200
01:05:33 ZIG: {"ZbEZSPReceived2":"372E8001340000F6"}
01:05:33 ZIG: new ack/data received, was 6 now 7
01:05:33 ZIG: freeing packet 6 from memory
01:05:33 ZIG: ZbEZSPSentRaw 84
01:05:33 ZIG: {"ZbEZSPReceived":"340000F6"}
01:05:33 ZIG: {"ZbEZSPReceived2":"472E90013F0000E7A804010600010B40010000F6010000"}
01:05:33 ZIG: ZbEZSPSentRaw 85
01:05:33 ZIG: {"ZbEZSPReceived":"3F0000E7A804010600010B40010000F6010000"}
01:05:34 ZIG: {"ZbEZSPReceived2":"572E9001450000040106000B0100010000EC64B5E7A8FFFF0518120B0000"}
01:05:34 ZIG: ZbEZSPSentRaw 86
01:05:34 ZIG: {"ZbEZSPReceived":"450000040106000B0100010000EC64B5E7A8FFFF0518120B0000"}
01:05:34 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xA8E7","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":31,"securityuse":0,"seqnumber":236,"fc":"0x18","manuf":"0x0000","transact":18,"cmdid":"0x0B","payload":"0000"}}
01:05:34 MQT: tele/tasmota_E2A955/RESULT = {"ZbResponse":{"Device":"0xA8E7","Command":"0006!00","Status":0,"StatusMessage":"SUCCESS","Endpoint":11,"LinkQuality":31}}
01:05:34 ZIG: ZbEZSPSend 340000E7A804010600010B400100001301050013000000
01:05:34 ZIG: adding packet to_send, to_ack:7, to_send:7, to_end:7
01:05:34 ZIG: Something to_send, to_ack:7, to_send:7, to_end:0
01:05:34 ZIG: ZbEZSPSentRaw 762F0001340000E7A804010600010B400100001301050013000000
01:05:34 ZIG: {"ZbEZSPReceived2":"602F8001340000F7"}
01:05:34 ZIG: new ack/data received, was 7 now 0
01:05:34 ZIG: freeing packet 7 from memory
01:05:34 ZIG: ZbEZSPSentRaw 87
01:05:34 ZIG: {"ZbEZSPReceived":"340000F7"}
01:05:34 ZIG: {"ZbEZSPReceived2":"702F90013F0000E7A804010600010B40010000F7010000"}
01:05:34 ZIG: ZbEZSPSentRaw 80
01:05:34 ZIG: {"ZbEZSPReceived":"3F0000E7A804010600010B40010000F7010000"}
01:05:34 ZIG: {"ZbEZSPReceived2":"002F9001450000040106000B0100010000ED64B5E7A8FFFF081813010000001000"}
01:05:34 ZIG: ZbEZSPSentRaw 81
01:05:34 ZIG: {"ZbEZSPReceived":"450000040106000B0100010000ED64B5E7A8FFFF081813010000001000"}
01:05:34 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0xA8E7","srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":31,"securityuse":0,"seqnumber":237,"fc":"0x18","manuf":"0x0000","transact":19,"cmdid":"0x01","payload":"0000001000"}}
01:05:34 ZIG: ZbZCLRawReceived: {"0xA8E7":{"0006/0000":0,"Endpoint":11,"LinkQuality":31}}
01:05:34 MQT: tele/tasmota_E2A955/SENSOR = {"ZbReceived":{"0xA8E7":{"Device":"0xA8E7","Power":0,"Endpoint":11,"LinkQuality":31}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
First make sure the device is turned on with:
zbsend { "Device":"0xA8E7", "send": {"power": 1}}
After the device turned on send the following request:
zbsend { "Device":"0xA8E7", "send": {"power": "?"}}
It doesn't matter if there is a question mark or any other invalid argument.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
After sending "zbsend { "Device":"0xA8E7", "send": {"power": "?"}}" I would expect a response with the current device state at the .../SENSOR topic and no change in the state of the device. 
When sending a request with an invalid argument I would expect an error message like "wrong argument" or just nothing.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I'm controlling a Phillips Hue RGB Bulb over mqtt but the issue also happens in the console.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-01-26:
Due to Tasmota being tuned for low memory footprint, it is quite normal for invalid arguments not resulting in some nice error message, but being treated as some variant of nil. This happens "all over", and changing this generally would mean lots of extra code and memory usage.

CrazyMan2000, 2021-01-26:
> Due to Tasmota being tuned for low memory footprint, it is quite normal for invalid arguments not resulting in some nice error message, but being treated as some variant of nil. This happens "all over", and changing this generally would mean lots of extra code and memory usage.

Sounds reasonable but I thinks it's not the right decision to translate every invalid command to 0 when using zigbee.
In Tasmota without zigbee every invalid command will result in a status report of the device. 
Isn't it possible to let Tasmota behave the same when using zigbee?

sfromis, 2021-01-26:
Outside of Zigbee it is _by far_ not the case that "every invalid command will result in a status report". Side effects of bad input is AFAIK not different here, meaning that the behavior is already basically "the same". I deliberately wrote "nil" instead of 0, as there is not some sort of logic to assign 0 for invalid values, more of a side effect of code not being able to use bad arguments.

CrazyMan2000, 2021-01-26:
Ok maybe I said it wrong.
Let me try again:
My problem is that Tasomta itself ignores invalid arguments (at least for power arguments, that's all I testet right now) but zigbee2Tasmota does not and instead changes the state of the device.

In my opinion that's not right. Zigbee2Tasmota should only change the format a message should have to be processed but not the result of a command.

sfromis, 2021-01-26:
Again, it is very common that Tasmota does *not* just ignore bad arguments, but does something with it. I just tested a bulb with `color zzaa000000` which gave me green light, skipping the invalid red value. State of device was changed, and that is what I already expected. Just take care to not use invalid arguments for normal operation :-)

s-hadinger, 2021-01-26:
Unfortunately there are many corner cases. For ex: "Power" expects a number, which can be a JSON number `1`, an hex value as string `"0x01"` or a synonym as string `"On"`. Moreover the tests are different whether it's a `Write` or `Send` which adds even more logic. It's much more compact to accept anything as a number and invalid is considered as zero - makes code much simpler.

It's doable but very down far in the priority list and don't expect it any time soon, if ever. And as mentioned, if the code size impact is too big, it will not be integrated.

closed 2021-01-26 by ascillato2