arendst/Tasmota #10718: Sonoff Zigbee Bridge not responding - reboots

valki2, 2021-01-27:
### PROBLEM DESCRIPTION
A new sonoff zigbee bridge was successfully flashed using tasmotizer and following this documentation (official link in the tasmota docs): https://zigbee.blakadder.com/Sonoff_ZBBridge.html
The device ends up with tasmota running - but very unresponsive (. Each page in the webinterface needs minutes to build up) and rebooting constantly every 2.3 minutes or so. 

There is no bootloop directly during startup, wifi and pings are good.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [x] Tasmota binary firmware version number used: 9.2.0.4(zbbridge)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer 1.2
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

too  unresponsive for console output... thats what i can squeeze out of the console:
---------------------------------
00:00:00.032 CFG: Loaded from flash at FC, Count 54
00:00:00.035 QPC: Count 1
00:00:00.038 ZIG: ZBBridge EEPROM found at address 0x50
00:00:00.039 SNS: Hardware Serial
00:00:00.041 Project tasmota Tasmota Version 9.2.0.4(zbbridge)-2_7_4_9(2021-01-27T16:44:46)
00:00:00.041 ZIG: Resetting EZSP device
00:00:00.527 WIF: Connecting to AP1 Mafia_IoT Channel 7 BSSId 4A:8F:5A:C5:17:AC in mode 11n as tasmota_E2B35B-4955...
00:00:01.755 WIF: Connected
00:00:02.005 HTP: Web server active on tasmota_E2B35B-4955 with IP address 192.168.3.241
00:00:23.548 ZIG: Resetting EZSP device
--------------------------
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
too  unresponsive for console output

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
too  unresponsive for console output

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
too  unresponsive for console output

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash a Sonoff ZB Bridge with the precompiled binary.

Program Version | 9.2.0.4(zbbridge) 
build 2021-01-27T16:44:46
core 2_7_4_9/2.2.2-dev(38a443e)
(also tried 9.2.0. stable release)

tried both available zigbee module ota's

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
working as a ZB bridge :-)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-01-27:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

valki2, 2021-01-27:
dear ascillato2, 
I would love to give more infos... but the console is absolutely not responding so that i could catch some outputs as ordered in the template.

I may add:
- fresh flash (again)
- no configuration entered except wifi
- no rules added

valki2, 2021-01-27:
I tried some versions. For me it seems that it works for 8.5 and 9.1 - but the zigbee module ota wont work on those older versions.
9.2+ is unusable (but with enough patience it seems that the zigbee is flashable. tried this earlier this day)

Log:

flashing 8.5 (via tasmotizer 1.2, erasing always on)
configuring wifi
```
00:00:00 CFG: Loaded from flash at F7, Count 5
00:00:00 QPC: Count 1
00:00:00 SNS: Hardware Serial
00:00:00 Project tasmota Tasmota Version 8.5.1(zbbridge)-2_7_4_1
00:00:00 WIF: Connecting to AP1 Mafia_IoT in mode 11N as tasmota_E2B35B-4955...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota_E2B35B-4955 with IP address 192.168.3.241
22:34:41 RSL: tele/tasmota_E2B35B/INFO1 = {"Module":"Sonoff ZbBridge","Version":"8.5.1(zbbridge)","FallbackTopic":"cmnd/DVES_E2B35B_fb/","GroupTopic":"cmnd/tasmotas/"}
22:34:41 RSL: tele/tasmota_E2B35B/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_E2B35B-4955","IPAddress":"192.168.3.241"}
22:34:41 RSL: tele/tasmota_E2B35B/INFO3 = {"RestartReason":"Software/System restart"}
22:34:41 QPC: Reset
22:34:45 ZIG: Resetting EZSP device
22:34:45 RSL: tele/tasmota_E2B35B/STATE = {"Time":"2021-01-27T22:34:45","Uptime":"0T00:00:12","UptimeSec":12,"Vcc":3.524,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:34:46 RSL: tele/tasmota_E2B35B/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
22:34:47 ZIG: timeout, goto label 99
22:34:47 RSL: tele/tasmota_E2B35B/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
22:34:47 ZIG: Abort
22:34:47 ZIG: Stopping (99)
22:35:08 UPL: File ncp-uart-sw_6.7.6_115200.ota ...
22:35:11 UPL: Successful 190272 bytes
```
completely responsive... but upload Failed (for the ota file)

reflashing 9.1 
configuring wifi
```
00:00:00 CFG: Loaded from flash at F7, Count 5
00:00:00 QPC: Count 1
00:00:00 SNS: Hardware Serial
00:00:00 Project tasmota Tasmota Version 9.1.0(zbbridge)-2_7_4_5
00:00:00 ZIG: Resetting EZSP device
00:00:00 WIF: Connecting to AP1 Mafia_IoT in mode 11N as tasmota_E2B35B-4955...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota_E2B35B-4955 with IP address 192.168.3.241
22:43:34 RSL: tele/tasmota_E2B35B/INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.1.0(zbbridge)","FallbackTopic":"cmnd/DVES_E2B35B_fb/","GroupTopic":"cmnd/tasmotas/"}
22:43:34 RSL: tele/tasmota_E2B35B/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_E2B35B-4955","IPAddress":"192.168.3.241"}
22:43:34 RSL: tele/tasmota_E2B35B/INFO3 = {"RestartReason":"Software/System restart"}
22:43:34 QPC: Reset
22:43:38 ZIG: Resetting EZSP device
22:43:38 RSL: tele/tasmota_E2B35B/STATE = {"Time":"2021-01-27T22:43:38","Uptime":"0T00:00:12","UptimeSec":12,"Vcc":3.532,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:43:39 RSL: tele/tasmota_E2B35B/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
22:43:40 ZIG: timeout, goto label 99
22:43:40 RSL: tele/tasmota_E2B35B/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
22:43:40 ZIG: Abort
22:43:40 ZIG: Stopping (99)
22:44:20 UPL: File ncp-uart-sw_6.7.8_115200.ota ...
22:44:22 UPL: Successful 190756 bytes
22:45:13 UPL: File ncp-uart-sw_6.7.6_115200.ota ...
22:45:16 UPL: Successful 190272 bytes
```
upload Failed (for both ota files)

reflashing 9.2 stable release (should be 9.2.0.4)
configuring wifi
pings ok
sloooooow UI. waiting 5mins to open the console or config page. "website not reachable" in between (while ping is ok). no led is blinking as it would when bootlooping.
unusable.

flashing 9.2 latest (5h old)
same situation. unusable slow with reboots inbetween.

valki2, 2021-01-29:
Update:
As the 9.1+ versions clearly differ in usability during its responsiveness i played around with the 8.5 version. 
To be sure, I searched for a 8.5 documentation and found https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

So far the installation works fine but I get a **Sync Timeout** during the flashing of the zigbee module.  Does this pinpoint anything for you?

Again just to be sure - I downloaded ncp-uart-sw_6.7.8_115200.ota and its MD5 hash is CA68D8B59F6958D1F3372B8FFBB3A61A.

```
00:00:00 CFG: Loaded from flash at FA, Count 20
00:00:00 SNS: Hardware Serial
00:00:00 SRC: Restart
00:00:00 Project tasmota ZbBridge Version 8.5.1(zbbridge)-2_7_4_1
00:00:00 CFG: Saved to flash at F9, Count 21, Bytes 4096
00:00:00 ZbEZSPReceived: time-out, discarding FC, 1
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 Mafia_IoT Channel 7 BSSId 4A:8F:5A:C5:17:AC in mode 11N as ZbBridge...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Connected
09:14:39 NTP: UTC 2021-01-29T08:14:38, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
09:14:39 HTP: Web server active on ZbBridge with IP address 192.168.3.241
09:14:40 RSL: tele/tasmota_E2B35B/INFO1 = {"Module":"Sonoff ZbBridge","Version":"8.5.1(zbbridge)","FallbackTopic":"cmnd/DVES_E2B35B_fb/","GroupTopic":"cmnd/tasmotas/"}
09:14:40 RSL: tele/tasmota_E2B35B/INFO2 = {"WebServerMode":"Admin","Hostname":"ZbBridge","IPAddress":"192.168.3.241"}
09:14:40 RSL: tele/tasmota_E2B35B/INFO3 = {"RestartReason":"Software/System restart"}
09:14:42 HTP: Main Menu
09:14:43 APP: Boot Count 6
09:14:43 CFG: Saved to flash at F8, Count 22, Bytes 4096
09:14:43 HTP: Firmware Upgrade
09:14:44 RSL: tele/tasmota_E2B35B/STATE = {"Time":"2021-01-29T09:14:44","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.523,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}
09:14:45 ZIG: Resetting EZSP device
09:14:46 RSL: tele/tasmota_E2B35B/RESULT = {"ZbState":{"Status":1,"Message":"EFR32 booted","RestartReason":"Power-on","Code":2}}
09:14:46 ZIG: ZbEZSPSend 000008
09:14:47 ZIG: timeout, goto label 99
09:14:47 RSL: tele/tasmota_E2B35B/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
09:14:47 ZIG: Abort
09:14:47 ZIG: Stopping (99)
09:14:55 CFG: Saved to flash at FB, Count 23, Bytes 4096
09:14:55 UPL: File ncp-uart-sw_6.7.8_115200.ota ...
09:14:56 UPL: File type 4
09:14:58 UPL: Successful 190756 bytes
09:14:58 HTP: Upload transfer
09:14:58 XMD: Init bootloader
09:14:59 XMD: Init sync
09:15:29 XMD: SYNC timeout
09:15:29 XMD: Restarting
09:15:30 HTP: Upload done
09:16:11 HTP: Main Menu
09:16:12 HTP: Firmware Upgrade
09:16:16 CFG: Saved to flash at FB, Count 24, Bytes 4096
09:16:16 UPL: File ncp-uart-sw_6.7.6_115200.ota ...
09:16:16 UPL: File type 4
09:16:20 UPL: Successful 190272 bytes
09:16:20 HTP: Upload transfer
09:16:21 XMD: Init bootloader
09:16:22 XMD: Init sync
09:16:52 XMD: SYNC timeout
09:16:52 XMD: Restarting
09:16:53 HTP: Upload done
09:19:50 RSL: tele/tasmota_E2B35B/STATE = {"Time":"2021-01-29T09:19:50","Uptime":"0T00:05:17","UptimeSec":317,"Vcc":3.531,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:05"}}
```

valki2, 2021-01-29:
This timeout also seems to happen when flashing via xterm / sx (running 8.5:
```
09:55:38 TCP: Starting TCP server on port 8881
09:55:38 RSL: stat/tasmota_E2B35B/RESULT = {"TCPStart":"Done"}
09:56:04 RSL: tele/tasmota_E2B35B/STATE = {"Time":"2021-01-29T09:56:04","Uptime":"0T00:05:11","UptimeSec":311,"Vcc":3.530,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Mafia_IoT","BSSId":"4A:8F:5A:C5:17:AC","Channel":7,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:05"}}
09:56:20 SRC: WebGui from 192.168.2.103
09:56:20 RSL: stat/tasmota_E2B35B/RESULT = {"POWER1":"OFF"}
09:56:20 RSL: stat/tasmota_E2B35B/POWER1 = OFF
09:56:21 CFG: Saved to flash at F7, Count 45, Bytes 4096
09:56:21 SRC: WebGui from 192.168.2.103
09:56:21 RSL: stat/tasmota_E2B35B/RESULT = {"POWER1":"ON"}
09:56:21 RSL: stat/tasmota_E2B35B/POWER1 = ON
09:56:21 TCP: from MCU: 0D0A4765636B6F20426F6F746C6F616465722076312E392E312E30340D0A312E2075706C6F61642067626C0D0A322E2072756E0D0A332E2065626C20696E666F0D0A424C203E200D0A4765636B6F20426F6F746C6F616465722076312E392E312E30340D0A312E2075706C6F61642067626C0D0A322E2072756E0D0A332E20
09:56:21 TCP: to MCU/1: 0101FEEB17A603080000000000000301010000FA0606FA1000000098E80200771D2E8907881DA728EF7F31F90707F9240000006A7AB563DDB509D755692C568E06B90FF3A7BB2907452A79A287ECD872C726CAE89A6FDAF90707F974E80200973AB65154103B93A22A25C1E524F90C88D9F757D0A3D4DA2E752BBD9DF1221E
09:56:22 CFG: Saved to flash at F6, Count 46, Bytes 4096
09:57:22 TCP: to MCU/1: 0101FEEB17A603080000000000000301010000FA0606FA1000000098E80200771D2E8907881DA728EF7F31F90707F9240000006A7AB563DDB509D755692C568E06B90FF3A7BB2907452A79A287ECD872C726CAE89A6FDAF90707F974E80200973AB65154103B93A22A25C1E524F90C88D9F757D0A3D4DA2E752BBD9DF1221E
09:58:21 TCP: to MCU/1: 0101FEEB17A603080000000000000301010000FA0606FA1000000098E80200771D2E8907881DA728EF7F31F90707F9240000006A7AB563DDB509D755692C568E06B90FF3A7BB2907452A79A287ECD872C726CAE89A6FDAF90707F974E80200973AB65154103B93A22A25C1E524F90C88D9F757D0A3D4DA2E752BBD9DF1221E
09:59:21 TCP: to MCU/1: 0101FEEB17A603080000000000000301010000FA0606FA1000000098E80200771D2E8907881DA728EF7F31F90707F9240000006A7AB563DDB509D755692C568E06B90FF3A7BB2907452A79A287ECD872C726CAE89A6FDAF90707F974E80200973AB65154103B93A22A25C1E524F90C88D9F757D0A3D4DA2E752BBD9DF1221E
10:00:21 TCP: to MCU/1: 0101FEEB17A603080000000000000301010000FA0606FA1000000098E80200771D2E8907881DA728EF7F31F90707F9240000006A7AB563DDB509D755692C568E06B90FF3A7BB2907452A79A287ECD872C726CAE89A6FDAF90707F974E80200973AB65154103B93A22A25C1E524F90C88D9F757D0A3D4DA2E752BBD9DF1221E
```

meanwhile, on the client:
```
Sending ncp-uart-sw_6.7.8_115200.ota, 1490 blocks: Give your local XMODEM receive command now.
Xmodem sectors/kbytes sent:   0/ 0kRetry 0: Got 65 for sector ACK
Retry 0: Timeout on sector ACK
Retry 0: Timeout on sector ACK
Retry 0: Timeout on sector ACK
```

s-hadinger, 2021-01-29:
Most of the time this is due to a hardware problem. Pads are very fragile and it's easy to break something.

You ESP TX -> MCU RX seems broken here, the MCU does not respond to any command sent.

valki2, 2021-01-29:
@s-hadinger  *Brilliant* 
You nailed it - there was a bad pad at the outer side of R13... - unbelievable!
Resoldered it and voila - Firmware update is OK!

I will try now an update to 9.2 to check if the issue with the performance is gone too before closing this incident! Stay tuned :-)

valki2, 2021-02-03:
Final Update:

Got a new device - **everything is working** there out of the box. 

For future googlers:

I had definitively an issue with the TX/RX connection between the 2 chips - although there were "to MCU/1: 0101FE...." and "XMD" messages in the output.

I resoldered the relevant vias and measured a good connection.
Then I could flash the MG21 chip BUT things went strange and tasmota versions are behaving different. 

SO: double check that your connections are fine before flashing to prevent an inconsistant firmware!

closed 2021-02-03 by valki2

voodooboy3000, 2021-04-06:
To others who may also be having this issue I found another work around by using some breadboard wires to connect the appropriate pins

I was not having any issues with Tasmota itself and this was working as expected however I was receiving the same error message as @valki2 when attempting to flash the UART firmware. 

I tried v 9.1.0/9.2.0/9.3.1.2

However re-reading your post I assumed it was a TX/RX issue, so I used two breadboard wires to connect the ETX > PB0O-RX and ERX > PB01-TX as per the pin out found [here](https://github.com/CoolKit-Technologies/DevDocs/blob/master/Zigbee/SM-011%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC%E4%B9%A6.md)

This immediately allowed me to do the firmware flash via the Tasmota Interface.

[Pictures](https://imgur.com/a/P0E2kbC)

s-hadinger, 2021-04-07:
@voodooboy3000 thanks for sharing. In your case you had most certainly a cut trace. It happens sometimes when people de-solder the wires and mistakenly cut the trace.

Be aware that you should connect ETX - PB00 and ERX - PB01 through 2x 1KOhm resistors, not directly one another. This prevents bad things to happen when pins are set to OUTPUT on both chips at the same time.

kallisti5, 2021-04-12:
I just had this exact issue (no response from ZigBee module), failure entering bootloader

The ERX via on the bottom of the ZBBridge is *extremely* delicate.
After pulling out my magnifying glass.. the trace "ends" just shy of the pad.  I was able to fix mine via:

* Scrape back some coating over the trace running up to the VIA
* Bridge the via over the bare area to the trace.

You can test the continuity with a multi-meter between the ERX, ETX, and R12,R13 resistors.

voodooboy3000, 2021-04-12:
I will break out the soldering iron at the weekend. I need to practice first though as only just got one. Makes wonder if the unit would have worked to begin with

kallisti5, 2021-04-12:
yeah... I didn't really test the ZBBridge before flashing Tasmota on it. It could be an issue on a lot of devices as shipped resulting in a lot of RMA returns (seeing some comments all saying generally the same thing)

Just carefully look at ERX (bottom side) with a high power magnifying glass.  See if the trace connects completely to the via.  You also could sanity check with a multi-meter to see if the via reaches the resistors next to Zigbee module.  it's how I debugged / fixed mine.

voodooboy3000, 2021-04-12:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEQAAABECAYAAAA4E5OyAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAoSURBVHic7cEBAQAAAIIg/69uSEABAAAAAAAAAAAAAAAAAAAAAACPBkiEAAHzeIrHAAAAAElFTkSuQmCC###24x24:true###][Tracking image][https://github.com/notifications/beacon/ABFZZRZQHBBI6E7FNVLWOE3TINUPVA5CNFSM4WV5DPT2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGDC5EBI.gif]

Yes true I only connected mine to the app to see if it was working but don't have any Zigbee Sonoff devices to test.

I will grab a decent magnifying glass and make sure I use my meter to check