arendst/Tasmota #10719: Connection restrictions when using AVM Mesh

tholler3012, 2021-01-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Hello, I have 1x Fritz!Box 7590 as Mesh Master and 1x Fritz!Box 7490 as Mesh Client in operation to provide the complete apartment with WLAN. The connection between both routers is via WLAN 5 GHz.
I have flashed some switches from Sonoff, Gosund and no name to Tasmota 9.2.0. All switches connected directly to the Mash Master FritzBox 7590 work fast and without problems. Problematic are all switches that are WLAN connected to the Mesh Client FritzBox 7490. These switches react sluggishly, partly only on the 3rd or 4th response.
I have now switched off the Mesh Client 7490 as a test. All switches are now connected to the main router 7590 and respond quickly and directly. Unfortunately, the WLAN level in the living room is now so low that working, watching TV or just surfing is not possible or only slowly and with interference.
Question: Is the problem with the switches, the network with a repeater or with Tasmota, because with the original firmware the connection and control of the switches via the repeater worked without problems, flawlessly and quickly.
Would be nice if this would be fixed!

Many greetings
Th. Holler

If you have any questions, please do not hesitate to contact me.

Translated with www.DeepL.com/Translator (free version)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP111 10A, AVM Fritz!Box 7590 & 7490
- [x] Tasmota binary firmware version number used: _____9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled -NO!
- [x] Flashing tools used: _____tuya-convert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:06:53 CMD: Backlog Template; Module; GPIO 255
23:06:53 RSL: stat/tasmota/RESULT = {"NAME":"Gosund SP111","GPIO":[321,1,320,1,0,2720,0,0,2624,32,2656,224,0,0],"FLAG":0,"BASE":45}
23:06:54 RSL: stat/tasmota/RESULT = {"Module":{"0":"Gosund SP111"}}
23:06:54 RSL: stat/tasmota/RESULT = {"GPIO0":{"321":"Led_i2"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
23:06:54 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-Schrank-25012021","FriendlyName":["Schrank"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:06:54 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:44:06","StartupUTC":"2021-01-27T21:22:48","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2021-01-25T19:22:02","SaveCount":82,"SaveAddress":"F4000"}}
23:06:54 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.21 15:09:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"420/699"}}
23:06:54 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EV24-FB7590",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000088","00006000","00000000"]}}
23:06:54 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
23:06:54 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2028","IPAddress":"192.168.1.166","Gateway":"192.168.1.10","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.10","Mac":"CC:50:E3:D4:67:EC","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:06:54 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D467EC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:06:54 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2021-01-27T22:06:54","Local":"2021-01-27T23:06:54","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:26","Sunset":"17:39"}}
23:06:54 RSL: stat/tasmota/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
23:06:54 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2021-01-27T23:06:54","ENERGY":{"TotalStartTime":"2021-01-25T19:10:51","Total":0.387,"Yesterday":0.101,"Today":0.235,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
23:06:54 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2021-01-27T23:06:54","Uptime":"0T00:44:06","UptimeSec":2646,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EV24-FB7590","BSSId":"44:4E:6D:26:16:12","Channel":11,"RSSI":34,"Signal":-83,"LinkCount":2,"Downtime":"0T00:00:23"}}}
23:07:00 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-Schrank-25012021","FriendlyName":["Schrank"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:07:00 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:44:12","StartupUTC":"2021-01-27T21:22:48","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2021-01-25T19:22:02","SaveCount":82,"SaveAddress":"F4000"}}
23:07:00 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.21 15:09:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"420/699"}}
23:07:00 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EV24-FB7590",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000088","00006000","00000000"]}}
23:07:00 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
23:07:00 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2028","IPAddress":"192.168.1.166","Gateway":"192.168.1.10","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.10","Mac":"CC:50:E3:D4:67:EC","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:07:00 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D467EC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:07:00 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2021-01-27T22:07:00","Local":"2021-01-27T23:07:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:26", [... cut ...]
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

luis33338, 2021-01-27:
> ### DESCRIPCIÓN DEL PROBLEMA
> _Una descripción clara y concisa de cuál es el problema. _
> Hola, tengo 1x Fritz! Box 7590 como Mesh Master y 1x Fritz! Box 7490 como Mesh Client en funcionamiento para dotar al apartamento completo con WLAN. La conexión entre ambos routers se realiza mediante WLAN 5 GHz.
> He mostrado algunos conmutadores de Sonoff, Gosund y sin nombre a Tasmota 9.2.0. Todos los interruptores conectados directamente al Mash Master FritzBox 7590 funcionan rápido y sin problemas. Son problemáticos todos los conmutadores que están conectados por WLAN al Mesh Client FritzBox 7490. Estos conmutadores reaccionan con lentitud, en parte sólo en la tercera o cuarta respuesta.
> Ahora apagué el Mesh Client 7490 como prueba. Todos los conmutadores están ahora conectados al enrutador principal 7590 y responden rápida y directamente. Desafortunadamente, el nivel de WLAN en la sala de estar ahora es tan bajo que no es posible trabajar, ver televisión o simplemente navegar o solo de forma lenta y con interferencias.
> Pregunta: ¿Es el problema con los conmutadores, la red con repetidor o con Tasmota, porque con el firmware original la conexión y control de los conmutadores a través del repetidor funcionaba sin problemas, impecable y rápidamente?
> ¡Sería bueno si esto se solucionara!
> 
> Muchos saludos
> Th. Gritar
> 
> Si tiene alguna pregunta, no dude en ponerse en contacto conmigo.
> 
> Traducido con [www.DeepL.com/Translator](http://www.DeepL.com/Translator) (versión gratuita)
> 
> ### INFORMACIÓN SOLICITADA
> _Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
> 
> * [x]  Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
> * [x]  Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
> * [x]  Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
> * [x]  Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
> * [x]  Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
> * [x]  Device used (e.g., Sonoff Basic): Gosund SP111 10A, AVM Fritz!Box 7590 & 7490
> * [x]  Tasmota binary firmware version number used: _____9.2.0
>   
>   * [x]  Pre-compiled
>   * [ ]  Self-compiled -NO!
> * [x]  Flashing tools used: _____tuya-convert
> * [x]  Provide the output of command: `Backlog Template; Module; GPIO 255`:
> 
> ```lua
>   Configuration output here:
> 23:06:53 CMD: Backlog Template; Module; GPIO 255
> 23:06:53 RSL: stat/tasmota/RESULT = {"NAME":"Gosund SP111","GPIO":[321,1,320,1,0,2720,0,0,2624,32,2656,224,0,0],"FLAG":0,"BASE":45}
> 23:06:54 RSL: stat/tasmota/RESULT = {"Module":{"0":"Gosund SP111"}}
> 23:06:54 RSL: stat/tasmota/RESULT = {"GPIO0":{"321":"Led_i2"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
> 23:06:54 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-Schrank-25012021","FriendlyName":["Schrank"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 23:06:54 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:44:06","StartupUTC":"2021-01-27T21:22:48","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2021-01-25T19:22:02","SaveCount":82,"SaveAddress":"F4000"}}
> 23:06:54 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.21 15:09:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"420/699"}}
> 23:06:54 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EV24-FB7590",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000088","00006000","00000000"]}}
> 23:06:54 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
> 23:06:54 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2028","IPAddress":"192.168.1.166","Gateway":"192.168.1.10","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.10","Mac":"CC:50:E3:D4:67:EC","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
> 23:06:54 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D467EC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
[... cut ...]
>   Console output here:
> ```
> 
> ### TO REPRODUCE
> _Steps to reproduce the behavior:_
> 
> ### EXPECTED BEHAVIOUR
> _A clear and concise description of what you expected to happen._
> 
> ### SCREENSHOTS
> _If applicable, add screenshots to help explain your problem._
> 
> ### ADDITIONAL CONTEXT
> _Add any other context about the problem here._
> 
> **(Please, remember to close the issue when the problem has been addressed)**

[arendst/Tasmota] Connection restrictions when using AVM Mesh (#10719)

ascillato, 2021-01-27:
@luis33338

Luis,

Please stop spamming issues

ascillato2, 2021-01-27:
@tholler3012

Hi,

It is a known issue. Some repeaters don't work well with the Arduino Core for ESP8266. Tasmota is built on top of the Arduino Core. And the Arduino core uses the SDK provided by Espressif (the manufacturer of the ESP8266) that is a closed source SDK. So, neither us nor the Arduino core team can solve this issue. Espressif won't open source their libraries and also the actual SDK type (NonOs) is no longer with support/fixes/development by them. Espressif have moved to (RTOS Type and the Arduino core haven't moved to it yet because it requires a full rewrite of all the Arduino Core - too much work for a free open source project)

So, seems that your issue can not be solved as least for now.

closed 2021-01-27 by ascillato2

tholler3012, 2021-01-28:
My solution: Activating the MAC-Filter in my AVM Fritz!Box Client, so that all Tasmota-Clients only connect to the Mesh Master.