arendst/Tasmota #10781: Zigbee Bridge Lost All Devices

hgelpke, 2021-01-31:
### PROBLEM DESCRIPTION
Zigbee devices all show unavilable in Home Assistant. Viewing the Tasmota console, not information is being shown from any zigbee device

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge
- [ ] Tasmota binary firmware version number used: 9.2
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/tasmota_7FC3C7/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"5088":"TCP Tx"},"GPIO2":{"0":"None"},"GPIO3":{"5120":"TCP Rx"},"GPIO4":{"323":"Led_i4"},"GPIO5":{"322":"Led_i3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_7FC3C7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power On","Uptime":"0T00:14:03","StartupUTC":"2021-01-31T16:18:16","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2020-08-16T20:04:49","SaveCount":59,"SaveAddress":"F7000"}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS2 = {"StatusFWR":{"Version":"9.2.0(zbbridge)","BuildDateTime":"2020-12-21T15:06:32","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"365/699"}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["tomato24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00008000","00006000","00000000"]}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":28,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C6","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,4,7,9,10,20,23,41","Sensors":""}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS5 = {"StatusNET":{"Hostname":"tasmota_7FC3C7-0967","IPAddress":"192.168.0.177","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"B4:E6:2D:7F:C3:C7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7FC3C7","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS7 = {"StatusTIM":{"UTC":"2021-01-31T16:32:19","Local":"2021-01-31T17:32:19","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:21","Sunset":"17:45"}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS10 = {"StatusSNS":{"Time":"2021-01-31T17:32:19"}}
17:32:19 RSL: stat/tasmota_7FC3C7/STATUS11 = {"StatusSTS":{"Time":"2021-01-31T17:32:19","Uptime":"0T00:14:03","UptimeSec":843,"Vcc":3.458,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"tomato24","BSSId":"F8:32:E4:6D:95:C0","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Unknown. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Bridge recognizes paired Zigbee devices

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-01-31:
We don't support ZHA here. Tasmota is only a bridge between tcp and serial. 

Please reach out to ZHA forums

closed 2021-01-31 by ascillato2

bbsan2k, 2021-02-09:
@s-hadinger I had the same issue after trying to OTA the ZBBridge. The devices were not LOST; but it seems all data has been removed from ZBBridge. However, the rest of the config was still persistent (like mqtt server, wifi etc).
Is OTA a problem for those devices?

s-hadinger, 2021-02-09:
From what version to what version?

bbsan2k, 2021-02-09:
Basically 9.2 to 9.2

s-hadinger, 2021-02-09:
That's surprising considering that data is saved on eeprom and not touched by OTA.

bbsan2k, 2021-02-10:
> That's surprising considering that data is saved on eeprom and not touched by OTA.

I‘ll give it another try - maybe it was some weird combination of other events...
Is the zigbee device config stored at the same location as the other config? It was especially strange that all „friendly names“ and groups were gone

s-hadinger, 2021-02-10:
On the ZBBridge, with Module 75, all zigbee information is stored on the EEPROM chip connected via I2C.

However, if you don't use Module 75 or don't compile with I2C, then it's stored in the internal ESP8266 Flash.

bbsan2k, 2021-02-10:
I used the official zbbridge binary...
I‘ll try it again and will update based on my findings...

alsa-kode, 2021-09-07:
Hiya... Just upgraded from 9.0.0.3 to 9.5.0.8 and lost all names too :(

BassT23, 2021-09-15:
Here (on 9.5) lost all Device Names after every reboot.