arendst/Tasmota #10784: LILYGO TTGO ESP32: Camera not working

herangstl, 2021-01-31:
### PROBLEM DESCRIPTION
After I configure the webcam and reboot the module the webcam is not working.
With sample code the webcam is working fine, but not with Tasmota.
Error message in the logs:
```
22:06:26.052 CAM: Init failed with error 0x20004
```

The ESP32 module I use is this one:
https://github.com/LilyGO/ESP32-Camera
https://www.banggood.com/LILYGO-TTGO-T-Journal-ESP32-Camera-Development-Board-OV2640-SMA-WiFi-3dbi-Antenna-0_91-OLED-Camera-Board-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-1379925.html

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): LILYGO TTGO ESP32
- [x] Tasmota binary firmware version number used: 9.2.0.4(webcam)-1_0_5_RC6
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:

Backlog:
```
22:22:17.081 CMD: Backlog 
22:22:17.084 RSL: stat/tasmota_6392D4/RESULT = {"Backlog":"Empty"}
```

Template:
```
22:23:00.633 CMD: Template
22:23:00.638 RSL: stat/tasmota_6392D4/RESULT = {"NAME":"ESP32-DevKit","GPIO":[4928,0,0,0,0,5091,0,0,0,608,640,4960,0,5088,5093,5095,0,5184,5120,5056,0,5024,5152,4992,0,0,0,0,0,0,5090,5089,5094,0,0,5092],"FLAG":0,"BASE":2}
```

GPIO 255:
```
22:23:31.032 CMD: GPIO 255
22:23:31.040 RSL: stat/tasmota_6392D4/RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"}}
22:23:31.045 RSL: stat/tasmota_6392D4/RESULT = {"GPIO27":{"5056":"CAM_SIOC"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"0":"None"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}
```

### TO REPRODUCE
Apply template and reboot

### EXPECTED BEHAVIOUR
webcam is working

### SCREENSHOTS
[image]
[image]

sfromis, 2021-01-31:
Looks like you did not apply the [customized template](https://tasmota.github.io/docs/ESP32/#lilygo-ttgo-t-camera-ov2640_v05) for the device.

herangstl, 2021-01-31:
> Looks like you did not apply the [customized template](https://tasmota.github.io/docs/ESP32/#lilygo-ttgo-t-camera-ov2640_v05) for the device.

The template you linked is not for my module. Unfortunately there are lots of different LILYGO TTGO modules out there.

I created my own template, double checked it - but still not working.

Do you know that this "Init failed with error 0x20004" actually means?

sfromis, 2021-01-31:
Don't know, but a quick google search got an [article](https://randomnerdtutorials.com/esp32-cam-troubleshooting-guide/) where someone explains:
_The 0x2004 error means the camera is not supported_
I suspect that bad pin config could cause this, but this is of course just speculation.

mohnen, 2021-02-19:
Your configuration does not match the device. 
I have created a template for this device.
`{"NAME":"TTGO T-Journal ESP32 Camera","GPIO":[4928,1,1,1,1,5091,1,1,1,1,1,4960,1,5088,5093,5095,0,5184,5120,5056,0,5024,5152,4992,0,0,0,0,1,1,5090,5089,5094,0,0,5092],"FLAG":0,"BASE":1}`

mohnen, 2021-02-19:
The camera works, but not the display.

mohnen, 2021-02-22:
Updated version with camera and display. Of course, you need to build your firmware for this manually.
`{"NAME":"TTGO T-Journal ESP32 Camera","GPIO":[4928,1,1,1,1,5091,1,1,1,608,640,4960,1,5088,5093,5095,0,5184,5120,5056,0,5024,5152,4992,0,0,0,0,1,1,5090,5089,5094,0,0,5092],"FLAG":0,"BASE":1}`

ascillato, 2021-02-22:
Great, thanks for sharing. Please, could you add this to https://templates.blakadder.com/ ? Thanks

mohnen, 2021-02-22:
> Great, thanks for sharing. Please, could you add this to https://templates.blakadder.com/ ? Thanks

Already submitted 👍

closed 2021-02-23 by ascillato2