arendst/Tasmota #10796: mqtt topic setting fails over API, MQTT and Console

ikke-t, 2021-02-01:
### PROBLEM DESCRIPTION

I can't set the MQTT topic any other way than GUI. MQTT, Console and API fail.

### REQUESTED INFORMATION

- [x] Device used (e.g., Sonoff Basic): sonoff-dual
- [x] Tasmota binary firmware version number used: 9.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
17:30:09 CMD: Backlog Template; Module; GPIO 255
17:30:09 MQT: stat/tasmota_E76344/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:30:09 MQT: stat/tasmota_E76344/RESULT = {"Module":{"5":"Sonoff Dual"}}
17:30:09 MQT: stat/tasmota_E76344/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
17:31:38 MQT: stat/tasmota_E76344/STATUS = {"Status":{"Module":5,"DeviceName":"Sonoff","FriendlyName":["sonoff-dual","Sonoff2"],"Topic":"tasmota_E76344","ButtonTopic":"0","Power":0,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:31:38 MQT: stat/tasmota_E76344/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:35","StartupUTC":"2021-02-01T16:23:03","Sleep":50,"CfgHolder":4617,"BootCount":366,"BCResetTime":"2021-02-01T16:40:18","SaveCount":392,"SaveAddress":"FB000"}}
17:31:38 MQT: stat/tasmota_E76344/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":4,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"367/699"}}
17:31:38 MQT: stat/tasmota_E76344/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ikenet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
17:31:38 MQT: stat/tasmota_E76344/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:31:38 MQT: stat/tasmota_E76344/STATUS5 = {"StatusNET":{"Hostname":"tasmota_E76344-0836","IPAddress":"192.168.117.70","Gateway":"192.168.117.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.117.65","Mac":"5C:CF:7F:E7:63:44","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:31:38 MQT: stat/tasmota_E76344/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.ikenet","MqttPort":1883,"MqttClientMask":"sonoff-dual","MqttClient":"sonoff-dual","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:31:38 MQT: stat/tasmota_E76344/STATUS7 = {"StatusTIM":{"UTC":"2021-02-01T16:31:38","Local":"2021-02-01T17:31:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:19","Sunset":"17:47"}}
17:31:38 MQT: stat/tasmota_E76344/STATUS10 = {"StatusSNS":{"Time":"2021-02-01T17:31:38"}}
17:31:38 MQT: stat/tasmota_E76344/STATUS11 = {"StatusSTS":{"Time":"2021-02-01T17:31:38","Uptime":"0T00:08:35","UptimeSec":515,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"ikenet","BSSId":"F2:9F:C2:F7:3C:4F","Channel":11,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
17:33:49 CMD: topic sonoff-dual
17:33:49 SRC: WebConsole from 192.168.117.59
17:33:49 CMD: Group 0, Index 1, Command "TOPIC", Data "sonoff-dual"
17:33:49 MQT: stat/tasmota_E76344/RESULT = {"Topic":"tasmota_%06X"}
```

### TO REPRODUCE

Go to web console and type:

```topic sonoff-dual```

### EXPECTED BEHAVIOUR

Go to MQTT config, and see the MQTT Topic shows "sonoff-dual"

### SCREENSHOTS

MQTT Topic in menu shows "tasmota_%06X"

### ADDITIONAL CONTEXT

If I set it from web GUI in MQTT config menu, it works.

ikke-t, 2021-02-01:
upgraded another sonoff, this time the sonoff basic. It behaves the same buggy way.

barbudor, 2021-02-01:
```
MqttClientMask":"sonoff-dual","MqttClient":"sonoff-dual"
```
You can't have MqttClient == Topic

sfromis, 2021-02-01:
Basically a duplicate of issue #9282

ikke-t, 2021-02-01:
thanks for pointing this out. I reconfigured the ansible, it works now.

closed 2021-02-01 by ikke-t