arendst/Tasmota #10802: EZO sensors failing to communicate

carlosmesquita, 2021-02-02:
### PROBLEM DESCRIPTION

Hello.
I am currently trying to use Tasmota with some EZO sensors from Atlas Sensors.
I have compiled Tasmota with support for EZO, flashed an ESP, and the sensor is detected using the "I2Cscan" command, with the right address (0x61 for this case).

Still, the sensor itself is not detected, and when trying to interact with the sensor (Sensor78 command), i get the following:

19:40:59 CMD: Sensor78 R
19:40:59 RSL: stat/tasmota_29B03B/RESULT = {"Command":"Error"}

Anything I try using the Sensor78 command, just returns the same error.

Also, there is a mismatch between the Tasmota EZO sensors documentation (https://tasmota.github.io/docs/EZO/), in the **Build Configuration** part,  the actual #defines that are in the code. In the documentation, the defines are like "USE_EZO_XXX", but in the code, it's actually "USE_EZOXXX", without the last underscore.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP8266 (Wemos Mini D1)
- [x] Tasmota binary firmware version number used: 9.2.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotize
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:32:21 CMD: Backlog Template; Module; GPIO 255
15:32:21 CMD: Group 0, Index 1, Command “BACKLOG”, Data “Template; Module; GPIO 255”
15:32:21 SRC: Backlog
15:32:21 CMD: Group 0, Index 1, Command “TEMPLATE”, Data “”
15:32:21 RSL: stat/tasmota_29B03B/RESULT = {“NAME”:“Generic”,“GPIO”:[1,1,1,1,1,1,1,1,1,1,1,1,1,1],“FLAG”:0,“BASE”:18}
15:32:21 SRC: Backlog
15:32:21 CMD: Group 0, Index 1, Command “MODULE”, Data “”
15:32:21 RSL: stat/tasmota_29B03B/RESULT = {“Module”:{“0":“Generic”}}
15:32:22 SRC: Backlog
15:32:22 CMD: Group 0, Index 1, Command “GPIO”, Data “255”
15:32:22 RSL: stat/tasmota_29B03B/RESULT = {“GPIO0":{“0”:“None”},“GPIO1”:{“0":“None”},“GPIO2":{“0”:“None”},“GPIO3”:{“0":“None”},“GPIO4":{“640”:“I2C SDA”},“GPIO5”:{“608":“I2C SCL”},“GPIO9":{“0”:“None”},“GPIO10”:{“0":“None”},“GPIO12":{“0”:“None”},“GPIO13”:{“0":“None”},“GPIO14":{“0”:“None”},“GPIO15”:{“0":“None”},“GPIO16":{“0”:“None”},“GPIO17”:{“0":“None”}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
15:32:41 CMD: Status 0
15:32:41 CMD: Group 0, Index 1, Command “STATUS”, Data “0”
15:32:41 RSL: stat/tasmota_29B03B/STATUS = {“Status”:{“Module”:0,“DeviceName”:“Tasmota”,“FriendlyName”:[“Tasmota”],“Topic”:“tasmota_29B03B”,“ButtonTopic”:“0”,“Power”:0,“PowerOnState”:3,“LedState”:1,“LedMask”:“FFFF”,“SaveData”:1,“SaveState”:1,“SwitchTopic”:“0",“SwitchMode”:[0,0,0,0,0,0,0,0],“ButtonRetain”:0,“SwitchRetain”:0,“SensorRetain”:0,“PowerRetain”:0}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS1 = {“StatusPRM”:{“Baudrate”:115200,“SerialConfig”:“8N1”,“GroupTopic”:“tasmotas”,“OtaUrl”:“http://ota.tasmota.com/tasmota/release/tasmota.bin.gz”,“RestartReason”:“External System”,“Uptime”:“0T00:01:54”,“StartupUTC”:“2021-02-02T14:30:47”,“Sleep”:50,“CfgHolder”:4617,“BootCount”:8,“BCResetTime”:“2021-02-01T19:53:01",“SaveCount”:19,“SaveAddress”:“F9000”}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS2 = {“StatusFWR”:{“Version”:“9.2.0(tasmota)“,”BuildDateTime”:“2021-02-01T18:47:18”,“Boot”:31,“Core”:“2_7_4_9",“SDK”:“2.2.2-dev(38a443e)“,”CpuFrequency”:80,“Hardware”:“ESP8266EX”,“CR”:“352/699"}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS3 = {“StatusLOG”:{“SerialLog”:2,“WebLog”:4,“MqttLog”:0,“SysLog”:4,“LogHost”:“”,“LogPort”:514,“SSId”:[“WIP”,“”],“TelePeriod”:300,“Resolution”:“558180C0",“SetOption”:[“00008009",“2805C8000100060000005A00000000000000”,“00000000",“00006000”,“00000000"]}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS4 = {“StatusMEM”:{“ProgramSize”:660,“Free”:340,“Heap”:24,“ProgramFlashSize”:4096,“FlashSize”:4096,“FlashChipId”:“164020",“FlashFrequency”:40,“FlashMode”:3,“Features”:[“00000809",“8FDAC787”,“04368001",“000000CF”,“010013C0",“C000F981”,“10004004",“000017FF”],“Drivers”:“1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45”,“Sensors”:“1,2,3,4,5,6,78”}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS5 = {“StatusNET”:{“Hostname”:“tasmota_29B03B-4155",“IPAddress”:“10.0.0.19",“Gateway”:“10.0.0.1",“Subnetmask”:“255.255.255.0",“DNSServer”:“10.0.0.1",“Mac”:“B4:E6:2D:29:B0:3B”,“Webserver”:2,“WifiConfig”:4,“WifiPower”:17.0}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS6 = {“StatusMQT”:{“MqttHost”:“”,“MqttPort”:1883,“MqttClientMask”:“DVES_%06X”,“MqttClient”:“DVES_29B03B”,“MqttUser”:“DVES_USER”,“MqttCount”:0,“MAX_PACKET_SIZE”:1200,“KEEPALIVE”:30}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS7 = {“StatusTIM”:{“UTC”:“2021-02-02T14:32:41”,“Local”:“2021-02-02T15:32:41”,“StartDST”:“2021-03-28T02:00:00”,“EndDST”:“2021-10-31T03:00:00”,“Timezone”:“+01:00”,“Sunrise”:“08:18”,“Sunset”:“17:49”}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS10 = {“StatusSNS”:{“Time”:“2021-02-02T15:32:41"}}
15:32:41 RSL: stat/tasmota_29B03B/STATUS11 = {“StatusSTS”:{“Time”:“2021-02-02T15:32:41”,“Uptime”:“0T00:01:54”,“UptimeSec”:114,“Heap”:25,“SleepMode”:“Dynamic”,“Sleep”:50,“LoadAvg”:19,“MqttCount”:0,“Wifi”:{“AP”:1,“SSId”:“WIP”,“BSSId”:“AC:9E:17:B2:67:C0",“Channel”:6,“RSSI”:94,“Signal”:-53,“LinkCount”:1,“Downtime”:“0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
15:32:51 CMD: Sensor78 R
15:32:51 CMD: Group 0, Index 78, Command “SENSOR”, Data “R”
15:32:51 RSL: stat/tasmota_29B03B/RESULT = {“Command”:“Error”}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Compile Tasmota with EZO support.
- Configure I2C pins.
- Connect EZO Sensor to I2C bus-
- Try to communicate with EZO Sensor. -> This step fails.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- When trying to communicate with EZO sensor, i am expecting not to get an Error, and to be able to read values and change configuration of the Sensor itself.

arendst, 2021-02-02:
Chk output of command ``I2cDriver``

carlosmesquita, 2021-02-02:
Hello, here it is:

16:02:45 CMD: Group 0, Index 1, Command "I2CDRIVER", Data ""
16:02:45 RSL: stat/tasmota_29B03B/RESULT = {"I2CDriver":"7,55"}

Not sure what the "7,55" means actually...

arendst, 2021-02-02:
It means that I2CDriver 7 and 55 are enabled as documented in ``I2CDEVICES.md``.

55 is the EZO driver so that means you should be OK.

I guess it's time for @tichris0 to help out here...

carlosmesquita, 2021-02-02:
Ok, now i understand.

When i was looking in the website (https://tasmota.github.io/docs/I2CDevices/), the I2CDriver 55 was not there, but i have lookged at the I2CDEVICES.md and it's there. Documentation here needs a bump hehe 

Ye, basically right now i don't have many more ideas. I can try placing some logs on the code and stuff, but if @tichris0 has any ideas, i'm willing to go that way first.

ascillato, 2021-02-02:
> Documentation here needs a bump hehe

Please, help us and complete it. You can edit that page and make a pull request. Thanks

carlosmesquita, 2021-02-02:
> > Documentation here needs a bump hehe
> 
> Please, help us and complete it. You can edit that page and make a pull request. Thanks

I will for sure do that! Also regarding the EZO defines, i will update the documentation according to the code.

carlosmesquita, 2021-02-02:
> Please, help us and complete it. You can edit that page and make a pull request. Thanks

While I was updating the documentation, I noticed that the documentation is already up-to-date :D
The site is just using an older version or something, not sure when or how often is the site updated.

tichris0, 2021-02-02:
Yes I had updated the documentation as well.  Happy to help if you hit 
any issue however.  I wasn't able to test some of these devices since I 
don't own every EZO sensors, although they all work very similarly and I 
was able to create tests to validate that it should work as per the spec.

carlosmesquita, 2021-02-02:
> Yes I had updated the documentation as well.  Happy to help if you hit any issue however.  I wasn't able to test some of these devices since I don't own every EZO sensors, although they all work very similarly and I was able to create tests to validate that it should work as per the spec.

Yes, EZO sensors all work the same. I currently own some, and already tested them using the ESP, but standalone, and was trying to use it inside Tasmota, but so far, no luck.
@tichris0 , i've added you on Discord, not sure if you use it or not! Can you tell me how did you use the sensors with Tasmota?

carlosmesquita, 2021-02-22:
Hello!
There was in fact an issue with the DO sensor, @araujo428 has fixed it and opened an PR: https://github.com/arendst/Tasmota/pull/11083
Tested and working for ESP8266 & 32.

closed 2021-02-22 by ascillato2

dpancer, 2023-07-19:
Hi, 

@tichris0 im getting the same issue with an EZO HUM sensor. The values are not shown in the tasmota webserver. 

I2CDRIVER gives:
RSL: RESULT = {"I2CDriver":"7,55"}

i2cscan gives: 

Grp 0, Cmd 'I2CSCAN', Idx 1, Len 0, Pld -99, Data ''
RSL: RESULT = {"I2CScan":"Device(s) found at 0x6f"}

I have been struggling with this for a day and im brand new to tasmota. Is there any chance I can get some help with this?

barbudor, 2023-07-19:
What binary are you using?
If you built it yourself, what have you done exactly?

Please set 'Weblog 4', restart and provide the logs from the 1st logline until the 1st SENSOR message

dpancer, 2023-07-19:
Thank for you quick reply. 
I have built it myself 2 times. 
The first time I built it I  removed "//" directly from the my_user_config.h file:
#define USE_EZOHUM

The second time, following the tasmota documentation I have defined it in user_config_override.h:
#ifndef USE_EZOHUM
#define USE_EZOHUM             
#endif

I have also flashed 2 other builds i found online where users successfully added and configured other EZO sensors. Unfortunally none of the users were using the EZO HUM

Log:
```
16:03:55.394 APP: Boot Count 39
16:03:55.401 RSL: STATE = {"Time":"2023-07-19T16:03:55","Uptime":"0T00:00:10","UptimeSec":10,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"atc admin","BSSId":"02:EC:DA:AD:09:79","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:03:55.612 CFG: Saved to flash at F9, Count 67, Bytes 4096
16:04:00.388 RSL: config = {"ip":"192.168.11.44","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"tasmota-D6C52B-1323","mac":"18FE34D6C52B","md":"Generic","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"13.0.0","t":"tasmota_D6C52B","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[],"sht":[],"ver":1}
16:04:00.456 RSL: sensors = {"sn":{"Time":"2023-07-19T16:04:00"},"ver":1}
```

[tasmota.zip](https://github.com/arendst/Tasmota/files/12096461/tasmota.zip)

barbudor, 2023-07-19:
Yes in user_config_override.h
There seems to be a mechanism to have it automatically defined if any USE_EZOxxx is added but I fear that it doesn't work as expected
So adding it may be the solution

Lack of any EZO related logs at boot up tends to show some piece of code is missing

barbudor, 2023-07-19:
No more ideas for now
May be add some logs in the VARIOUS EZO files around EZO device detection to check if those are called
personality I found this driver code over-engineered and hard to follow

dpancer, 2023-07-19:
Alright, thanks for you help. If you do think of anything else please update me.

dpancer, 2023-08-04:
@tichris0 Still cant figure this issue out. Any chance you can step in and give some assistance?

dpancer, 2023-09-30:
Still no one with success trying to hook up an EZO-Hum Sensor with Tasmota?