arendst/Tasmota #10806: add<x> does crazy things

gine78, 2021-02-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I add 0.01 per count of my gazmeter. I use the conter device paired with a reed sensor. Every Round they ADD3 0.01. But sometimes Tasmota turnover a number. See Bottom.

I usr the rule:
`RULE1
ON Counter#c1>%var1% DO Backlog var1 %value%; Backlog Publish gas/gaszaehler/gas 1; Backlog add3 0.01; Backlog Publish gas/gaszaehler/m3 %var3% ENDON` 

This add every round 0.01 to VAR3.

Now i become the following output in the console:

`22:01:01 RUL: COUNTER#C1>%VAR1% performs "Backlog var1 228; Backlog Publish gas/gaszaehler/gas 1; Backlog add3 0.01; Backlog Publish gas/gaszaehler/m3 152289.80"
22:01:01 MQT: stat/gas/RESULT = {"Var1":"228"}
22:01:01 MQT: gas/gaszaehler/gas = 1
22:01:02 MQT: stat/gas/RESULT = {"Add3":"152289.81"}
22:01:02 MQT: gas/gaszaehler/m3 = 152289.80
22:02:47 RUL: COUNTER#C1>%VAR1% performs "Backlog var1 229; Backlog Publish gas/gaszaehler/gas 1; Backlog add3 0.01; Backlog Publish gas/gaszaehler/m3 152289.81"
22:02:47 MQT: stat/gas/RESULT = {"Var1":"229"}
22:02:48 MQT: gas/gaszaehler/gas = 1
22:02:48 MQT: stat/gas/RESULT = {"Add3":"152289.83"}
22:02:48 MQT: gas/gaszaehler/m3 = 152289.81`

You can see, that Var1 is change correct. But Var3 does not. I miss 152289.**82** - Tasmota ignores or round it in the Background? What is my fault?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Wemos D1 Mini Pro
- [ x] Tasmota binary firmware version number used: 9.2.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: Tasmotize
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
```
22:18:14 MQT: stat/gas/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:18:14 MQT: stat/gas/RESULT = {"Module":{"18":"Generic"}}
22:18:14 MQT: stat/gas/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"352":"Counter1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
```

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

`22:19:10 MQT: stat/gas/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":149,"Free":362,"Rules":"ON Counter#c1>%var1% DO Backlog var1 %value%; Backlog Publish gas/gaszaehler/gas 1; Backlog add3 0.01; Backlog Publish gas/gaszaehler/m3 %var3% ENDON"}`

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I dont know

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2021-02-02:
No sure if the repeated `Backlog` commands are confusing things, but definitely not needed.

`ON Counter#c1>%var1% DO Backlog var1 %value%; Publish gas/gaszaehler/gas 1; add3 0.01; Publish gas/gaszaehler/m3 %var3% ENDON`

gine78, 2021-02-02:
I have delete the double backlog entrys like your code.

`22:48:24 RUL: COUNTER#C1>%VAR1% performs "Backlog var1 232; Publish gas/gaszaehler/gas 1; add3 0.01; Publish gas/gaszaehler/m3 152289.86"`
`22:48:24 MQT: stat/gas/RESULT = {"Var1":"232"}`
`22:48:24 MQT: gas/gaszaehler/gas = 1`
`22:48:24 MQT: stat/gas/RESULT = {"Add3":"152289.88"}`
`22:48:25 MQT: gas/gaszaehler/m3 = 152289.86`
`22:50:16 RUL: COUNTER#C1>%VAR1% performs "Backlog var1 233; Publish gas/gaszaehler/gas 1; add3 0.01; Publish gas/gaszaehler/m3 152289.88"`
`22:50:16 MQT: stat/gas/RESULT = {"Var1":"233"}`
`22:50:16 MQT: gas/gaszaehler/gas = 1`
`22:50:17 MQT: stat/gas/RESULT = {"Add3":"152289.89"}`
`22:50:17 MQT: gas/gaszaehler/m3 = 152289.88`
`22:52:04 RUL: COUNTER#C1>%VAR1% performs "Backlog var1 234; Publish gas/gaszaehler/gas 1; add3 0.01; Publish gas/gaszaehler/m3 152289.89"`
`22:52:04 MQT: stat/gas/RESULT = {"Var1":"234"}`
`22:52:04 MQT: gas/gaszaehler/gas = 1`
`22:52:04 MQT: stat/gas/RESULT = {"Add3":"152289.91"}`
`22:52:05 MQT: gas/gaszaehler/m3 = 152289.89  `

the 90 is not counting - he is jumping from 89 to 91 

`22:58:38 MQT: stat/gas/RESULT = {"CalcRes":2}` is reduced to two number after the comma.

sfromis, 2021-02-02:
The problem is to be that you're expecting too high a resolution of the numbers. Tasmota math is using single precision floating point, up to approx 7 decimal digits. Accuracy may be a bit lower when numbers are not integers, as decimal fractions not composed of negative powers of 2 would take an infinite number of binary digits after the decimal point to be exact, like when 1/3 cannot be fully expressed as a single number of finite length.

gine78, 2021-02-02:
Then is it better i add 0.0100000? Or is there a other way to math the right number?

ascillato2, 2021-02-03:
In a backlog sequence, after a publish command, you can not add any more commands

sfromis, 2021-02-03:
Nope, trailing zeroes makes no difference. The resolution problem is with values like 152289.83 which are 8 digits, near the limit of what can be represented. You'd get a bit better results using integers (multiplying all your numbers by 100), but that does not get around the maximum number of bits to represent the number. Math in rules does not support what you're trying to do.

sfromis, 2021-02-03:
@ascillato2 - the displayed log does show the Add3 operations after the publish command being executed.

gemu2015, 2021-02-03:
you may use the SML driver for your gas counter. This driver internally works with double precision because many meters need it to be completely accurate.

gine78, 2021-02-03:
@gemu2015 
i have check your work. This looks good. Can i use it with a reed sensor? is there a precompiled image?

gemu2015, 2021-02-03:
yes of course you may use a reed sensor.
Unfortunately there is no precompiled binary. you may use the Tasmota online compiler.
you need at least these defines in user_config_override:

#define USE_SML_M
#define USE_SCRIPT

gine78, 2021-02-03:
I give them a try. Thank You.

gine78, 2021-02-03:
@gemu2015 is there a readable howto hox can i count a reed sensor at gpio6 ? 

I think, this is not the right place to talk about this. can i contact you a other way? Gern auch in deutsch.

gemu2015, 2021-02-03:
@gine78
there is a large community  of Tasmota users and also of the sml driver on this german forum: https://forum.creationx.de/forum/

there are also tutorials about the drivers usage there.

if you need further help you can contact me on this forum

gine78, 2021-02-03:
I have contact you on this forum. I close the issue here at the moment. Thank you.

closed 2021-02-03 by gine78