arendst/Tasmota #10830: Problem with type of IR code sended to Domotic via MQTT

saidlm, 2021-02-04:
### PROBLEM DESCRIPTION
I am using Tasmota as IR transmitter / receiver for Domoticz controller. Everything worked properly till upgrade to 9.2.0. After the upgrade I am experiencing problem with string reported to Domoticz via MQTT. 
In previous version which I used the received code was send to Domoticz as hexadecimal string.  Currently Tasmota sends decimal value. It doesn't matter if SetOption29 is se to 0 or 1. Is there any possibility to change this behavior?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic):ESP8266 (Wemos Mini D1)
- [ ] Tasmota binary firmware version number used: 9.2.0 (2020-12-21T15:03:40)
  - [ ] Pre-compiled
- [ ] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:14:11 CMD: Backlog Template; Module; GPIO 255
17:14:11 MQT: stat/tasmota_8C9057/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:14:12 MQT: stat/tasmota_8C9057/RESULT = {"Module":{"18":"Generic"}}
17:14:12 MQT: stat/tasmota_8C9057/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1056":"IRsend"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1088":"IRrecv"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

16:49:41 MQT: tele/tasmota_8C9057/RESULT = {"Time":"2021-02-04T16:49:41","IrReceived":{"Protocol":"NEC","Bits":32,"Data":"0xE12448B7"}}
16:49:41 MQT: domoticz/in = {"idx":28,"nvalue":0,"svalue":"-249280329","Battery":100,"RSSI":3}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-02-04:
> In previous version which I used the received code was sent to Domoticz as hexadecimal string.

Not sure this is true. I've checked the code from latest dev down to v7.2 and I see no change in sending the value to Domoticz:
```
#ifdef USE_DOMOTICZ
      if (iridx) {
        unsigned long value = results.value | (iridx << 28);  // [Protocol:4, Data:28]
        DomoticzSensor(DZ_COUNT, value);                      // Send data as Domoticz Counter value
      }
#endif  // USE_DOMOTICZ
```
It has always sent the data as decimal instead of hex.

saidlm, 2021-02-04:
I am not 100% sure but according to other tasmotized devices which I have in my house I expect I had version 6.2.1. All my scripts which I have in Domoticz compares hex values e.g. "if (device.text == "0XE12448B7") then". Currently id doesn't work at all just because there is decimal value i the variable. Only change was Tasmota upgrade and renewal of the configuration of that device.

arendst, 2021-02-04:
Ditto down to 6.1.1, still decimal.

I guess you will have to change your lua to convert decimal to hex.

saidlm, 2021-02-04:
Well ok, I am little bit confused now. It really had been working for more than year. Strange ...
Anyway thank you for your effort.

closed 2021-02-04 by ascillato2