arendst/Tasmota #10840: rf_switch nSeparationLimit

PepeTheFroggie, 2021-02-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

rf_switch receiv of protocol 4 data not working

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

try to receive protocol 4 433mhz signal
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

i expect the protocol 4 rf message to show up like the protocol 1 does

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

need to change tasmota/lib/lib_rf/src/RCSwitch.cpp nSeparationLimit to 2000 to fix it

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-02-05:
Hi, if you have found already the solution, please help us and provide a pull request. Thanks

arendst, 2021-02-05:
With this change what will be the impact for other protocols?

Jason2866, 2021-02-05:
Change will have side effects for other protocols. From driver code
```
// separationLimit: minimum microseconds between received codes, closer codes are ignored.
// according to discussion on issue #14 it might be more suitable to set the separation
// limit to the same time as the 'low' part of the sync signal for the current protocol.
// should be set to the minimum value of pulselength * the sync signal
```

PepeTheFroggie, 2021-02-05:
>With this change what will be the impact for other protocols?
IDK.
In soft-soft testing (arduino and esp bitbang) i got the impression only protocol 1 and 2 work reliably.
I only got protocol 4 hardware for real-life tests, and receive only works with the mod.

PepeTheFroggie, 2021-02-06:
[image]
[image]
The real gap between messages seems to be 2105 us. I think the gap ist protocol dependant and that is why the "scan all protocols" thing fails. Better fix on one protocol and set the gap accordingly. 
btw recording is tasmota send pin, this command: RFsend {"Data":"0x000014","Bits":24,"Protocol":4,"Repeat":5}

Jason2866, 2021-02-06:
You have noticed the command `RfProtocol` ?
Disabling the not needed protocols enhances the reliablity of the protocol you want to use.

PepeTheFroggie, 2021-02-07:
No i was not aware of RfProtocol.
But i think it wont change nSeparationLimit on receive and it wont stop overwriting pulselen with protocol1 default on send.

I will close this and reopen the pulselen thing in discussion.

The nSeparationLimit thing needs a deeper examination, will reopen in discussion if i get a better workaround.

closed 2021-02-07 by PepeTheFroggie