arendst/Tasmota #10841: Rules dos not Work after Tasmota 8.1.0

coolheizer, 2021-02-05:
An error seems to have crept in in the rules, until Tasmota 8.1.0 the rule described below works properly, everything above leads to wild clocking.

I need a differential circuit for my solar pool heating.

The control is blocked from a pool temperature of 40 °.

If the temperature in the collector is 3 ° higher than in the pool, the pump starts, less than 1 ° the pump is switched off.

Switch off after 30 minutes to be on the safe side.

This rule works up to Tasmota 8.1.0:

mem3 40

rule3
ON DS18B20-1 # temperature DO event t1 =% value% ENDON
ON DS18B20-2 # temperature DO event t2 =% value% ENDON
ON event # t2 <% mem3% DO var1 1 ENDON
ON event # t2> =% mem3% DO var1 0 ENDON
ON event # t2 DO Backlog var2% value%; add2 1 ENDON
ON event # t2 DO Backlog var3% value%; add3 3 ENDON
ON event # t1>% var3% DO Power1% var1% ENDON
ON event # t1 <% var2% DO Power1 0 ENDON
ON power1 # state = 1 Do RuleTimer1 1800 ENDON
ON rules # timer = 1 do Power1 off ENDON

11:23:41 MQT: stat/Solarheizung/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":449,"Free":62,"Rules":"ON DS18B20-1#temperature DO event t1=%value% ENDON ON DS18B20-2#temperature DO event t2=%value% ENDON ON event#t2<%mem3% DO var1 1 ENDON ON event#t2>=%mem3% DO var1 0 ENDON ON event#t2 DO Backlog var2 %value%; add2 1 ENDON ON event#t2 DO Backlog var3 %value%; add3 3 ENDON ON event#t1>%var3% DO Power1 %var1% ENDON ON event#t1<%var2% DO Power1 0 ENDON ON power1#state=1 Do RuleTimer1 1800 ENDON ON rules#timer=1 do Power1 off ENDON"}

11:19:39 CMD: Status 0
11:19:39 SRC: WebConsole from 191.168.178.20
11:19:39 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
11:19:39 MQT: stat/Solarheizung/STATUS = {"Status":{"Module":18,"DeviceName":"Solarheizung","FriendlyName":["Solarheizung"],"Topic":"Solarheizung","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:19:39 MQT: stat/Solarheizung/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power On","Uptime":"0T00:02:56","StartupUTC":"2021-02-05T10:16:43","Sleep":150,"CfgHolder":4617,"BootCount":28,"BCResetTime":"2020-06-11T09:40:40","SaveCount":71,"SaveAddress":"F5000"}}
11:19:39 MQT: stat/Solarheizung/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020.12.21 15:09:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"431/699"}}
11:19:39 MQT: stat/Solarheizung/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wrt3_plus","wrt"],"TelePeriod":60,"Resolution":"AA8180C0","SetOption":["00008009","2805C80001000600005A5AFF000000000000","00000001","00006000","00000000"]}}
11:19:39 MQT: stat/Solarheizung/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":2,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
11:19:39 MQT: stat/Solarheizung/STATUS5 = {"StatusNET":{"Hostname":"Solarheizung-7633","IPAddress":"191.168.178.242","Gateway":"191.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"191.168.178.1","Mac":"BC:DD:C2:24:3D:D1","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
11:19:39 MQT: stat/Solarheizung/STATUS6 = {"StatusMQT":{"MqttHost":"191.168.178.34","MqttPort":1880,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_243DD1","MqttUser":"coolheizer","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:19:39 MQT: stat/Solarheizung/STATUS7 = {"StatusTIM":{"UTC":"2021-02-05T10:19:39","Local":"2021-02-05T11:19:39","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:51","Sunset":"17:08"}}
11:19:39 MQT: stat/Solarheizung/STATUS10 = {"StatusSNS":{"Time":"2021-02-05T11:19:39","DS18B20-1":{"Id":"3C01B556EE0A","Temperature":24.81},"DS18B20-2":{"Id":"3C01B556EF87","Temperature":23.94},"TempUnit":"C"}}
11:19:39 MQT: stat/Solarheizung/STATUS11 = {"StatusSTS":{"Time":"2021-02-05T11:19:39","Uptime":"0T00:02:56","UptimeSec":176,"Heap":24,"SleepMode":"Dynamic","Sleep":150,"LoadAvg":52,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"wrt","BSSId":"52:64:2B:9F:2B:15","Channel":4,"RSSI":100,"Signal":-31,"LinkCount":1,"Downtime":"0T00:00:03"}}}
11:19:39 SRC: Backlog
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-02-05:
Hi, your rules have several syntax errors.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-05 by ascillato2

coolheizer, 2021-02-05:
Syntax error?

Please take a closer look.

I have this rule from:
https://tasmota.github.io/docs/Rules/#solar-heater-control
adopted and adapted to my needs.

Github's original rule is:

mem3 25

rule1
  ON DS18B20-1#temperature DO event t1=%value% ENDON
  ON DS18B20-2#temperature DO event t2=%value% ENDON
  ON event#t2>%mem3% DO var1 1 ENDON
  ON event#t2<=%mem3% DO var1 0 ENDON
  ON event#t1 DO Backlog var2 %value%; add2 1 ENDON
  ON event#t1 DO Backlog var3 %value%; add3 2 ENDON
  ON event#t2>%var3% DO Power1 %var1% ENDON
  ON event#t2<%var2% DO Power1 0 ENDON

I have only adapted this rule to my needs (sensor exchanged, pump / relay blocked / switched off from 40 °)

If it is a syntax error, how do you explain that Tasmota 8.1.0 and the previous versions work with my rule?
Even the original from github does not work with versions greater than 8.1.0

coolheizer, 2021-02-05:
Problem was Fixed, much Thanks to barbudor, he says:

> The problem in the logs was that on DS18BX20 was executed before the last parts of the previous occurence was completed
Total mess
Maybe some things has changed in tasmota regarding order of executions in rules
But anyhow, there is no need to control heater/pump at 1 second period

Now teleperiod is 10, and rule is:

mem3 40
rule3
  ON tele-DS18B20-1#temperature DO event t1=%value% ENDON
  ON tele-DS18B20-2#temperature DO event t2=%value% ENDON
  ON event#t2<%mem3% DO var1 1 ENDON
  ON event#t2>=%mem3% DO var1 0 ENDON
  ON event#t2 DO Backlog var2 %value%; add2 1 ENDON
  ON event#t2 DO Backlog var3 %value%; add3 3 ENDON
  ON event#t1>%var3% DO Power1 %var1% ENDON
  ON event#t1<%var2% DO Power1 0 ENDON
  ON power1#state=1 Do RuleTimer1 1800 ENDON
  ON rules#timer=1 do Power1 off ENDON
  
This works fine.

It would be nice if the rule on https://tasmota.github.io/docs/Rules/#solar-heater-control
could be corrected for other users.

Thanks for pointing out https://discord.gg/Ks2Kzd4
I was given competent help there.