arendst/Tasmota #10845: Error ArduinoIDE

PortScanner, 2021-02-05:
i got the following error:
unishox.h: No such file or directory

i haven't found a solution anywhere.
THX

closed 2021-02-05 by close-issue-app[bot]

PortScanner, 2021-02-05:
Dress

ascillato, 2021-02-05:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)