arendst/Tasmota #10849: Energy Today randomly reset

DarioB92, 2021-02-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Total Today Energy on PZEM-004T V3.0 randomly reset

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): PZEM-004T V3.0 , ESP8266 NodeMCU
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Taasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"18":"Generic"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:16:59 MQT: stat/Fotovoltaico/STATUS = {"Status":{"Module":18,"DeviceName":"Power-Powerplant","FriendlyName":["Power-Powerplant"],"Topic":"Fotovoltaico","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:16:59 MQT: stat/Fotovoltaico/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T05:25:06","StartupUTC":"2021-02-06T04:51:53","Sleep":0,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2020-12-29T23:18:33","SaveCount":62,"SaveAddress":"F6000"}}
11:16:59 MQT: stat/Fotovoltaico/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:17:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"417/699"}}
11:16:59 MQT: stat/Fotovoltaico/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["redacted",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
11:16:59 MQT: stat/Fotovoltaico/STATUS4 = {"StatusMEM":{"ProgramSize":587,"Free":416,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000410","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
11:16:59 MQT: stat/Fotovoltaico/STATUS5 = {"StatusNET":{"Hostname":"Fotovoltaico","IPAddress":"redacted","Gateway":"redacted","Subnetmask":"255.255.255.0","DNSServer":"redacted","Mac":"redacted","Webserver":2,"WifiConfig":4,"WifiPower":redacted}}
11:16:59 MQT: stat/Fotovoltaico/STATUS6 = {"StatusMQT":{"MqttHost":"redacted","MqttPort":redacted,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D05765","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:16:59 MQT: stat/Fotovoltaico/STATUS7 = {"StatusTIM":{"UTC":"2021-02-06T10:16:59","Local":"2021-02-06T11:16:59","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:06","Sunset":"17:28"}}
11:16:59 MQT: stat/Fotovoltaico/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Wait, will happen more time a day

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Energy Today will reset to 0.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-02-06:
Hi, it is already fixed. Please update to latest tasmota from development branch. 

Http://ota.tasmota.com/tasmota.bin

ascillato2, 2021-02-06:
Closing this issue as it is **duplicated**. Sorry.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-06 by ascillato2