arendst/Tasmota #10857: IPAddress4 DNS change does not persist after restarting

rafalcypis, 2021-02-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have sonoff t1 2ch module installed on my sonoff switch, I used latest precompiled dev tasmota.bin 
I issued following command to update my DNS server settings:
IPAddress4 192.168.1.15
It changes fine, but when I issue restart 1, 
It is reverted to 192.168.1.1 which is my gateway IP address.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.2.0.6(tasmota) 2_7_4_9/2.2.2-dev(38a443e)
  - [x] Pre-compiled
  - [] Self-compiled
- [x] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Issue command
IPAddress4 with correct IP address in console
verify DNS change in Information page
issue command restart 1
verify DNS rollbacked to previous setting

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
DNS IP address should remain changed to new IP Address provided

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-02-06:
Have you set also ipaddress1 ?

rafalcypis, 2021-02-06:
OK, Not sure if I have missed that but 
https://tasmota.github.io/docs/Commands/ WIFI section does not contain information
ipaddress1 to be set in order to persist ipaddress4, I understand that now, that with DHCP set automatically IP address you cannot change DNS value permanently.
I have my router static assigned IP address for each device, knowing it won't change did not know must be set on tasmota as ell to make changes to DNS setting.
It is working fine now.
All clear now, thank you ascillato2.

closed 2021-02-06 by rafalcypis