arendst/Tasmota #10860: with 2 shutters configured the second shutter affects the first shutter.

mikep1998, 2021-02-06:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____D1 mini
- [ x] Tasmota binary firmware version number used: ____9206_
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:25:12.492 SRC: Backlog
00:25:12.495 CMD: Group 0, Index 1, Command "GPIO", Data "255"
00:25:12.503 RSL: stat/tasmota_D9C6B8/RESULT = {"GPIO0":{"226":"Relay3"},"GPIO1":{"0":"None"},"GPIO2":{"227":"Relay4"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"416":"PWM1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"417":"PWM2"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:26:30.180 CMD: Group 0, Index 1, Command "STATUS", Data "0"
00:26:30.186 RSL: stat/tasmota_D9C6B8/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_D9C6B8","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:26:30.217 RSL: stat/tasmota_D9C6B8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:36","StartupUTC":"2021-02-06T23:18:54","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-02-06T22:41:50","SaveCount":248,"SaveAddress":"F4000"}}
00:26:30.254 RSL: stat/tasmota_D9C6B8/STATUS2 = {"StatusFWR":{"Version":"9.2.0.6(tasmota)","BuildDateTime":"2021-02-06T17:20:11","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"398/699"}}
00:26:30.273 RSL: stat/tasmota_D9C6B8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mike1.1.2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","2805C8000100060000005A0A000000000000","40000080","00006000","00000000"]}}
00:26:30.303 RSL: stat/tasmota_D9C6B8/STATUS4 = {"StatusMEM":{"ProgramSize":591,"Free":412,"Heap":27.7,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:26:30.338 RSL: stat/tasmota_D9C6B8/STATUS5 = {"StatusNET":{"Hostname":"tasmota_D9C6B8-1720","IPAddress":"192.168.1.131","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:D9:C6:B8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:26:30.359 RSL: stat/tasmota_D9C6B8/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D9C6B8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:26:30.380 RSL: stat/tasmota_D9C6B8/STATUS7 = {"StatusTIM":{"UTC":"2021-02-06T23:26:30","Local":"2021-02-07T00:26:30","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:11","Sunset":"17:57"}}
00:26:30.399 RSL: stat/tasmota_D9C6B8/STATUS10 = {"StatusSNS":{"Time":"2021-02-07T00:26:30","Shutter1":{"Position":96,"Direction":0,"Target":100},"Shutter2":{"Position":96,"Direction":0,"Target":100}}}
00:26:30.419 RSL: stat/tasmota_D9C6B8/STATUS11 = {"StatusSTS":{"Time":"2021-02-07T00:26:30","Uptime":"0T00:07:36","UptimeSec":456,"Heap":27.8,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","PWM":{"PWM1":483,"PWM2":0},"Wifi":{"AP":1,"SSId":"mike1.1.2","BSSId":"9C:D3:6D:9E:41:3C","Channel":10,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:08"}}}
00:26:30.457 RSL: stat/tasmota_D9C6B8/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":5,"Close":5,"50perc":50,"Delay":4,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"5"}},{"SHT1":{"Relay1":3,"Relay2":4,"Open":5,"Close":5,"50perc":50,"Delay":4,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"5"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1) click the open or close button for shutter1 shutter1 relays toggle on and off. Shutter1 open and closes correctly. But shutter2 relay3 turns ON and never turns off.
2) click the open or close button for shutter2 shutter2 opens or closes. But shutter1 also moves.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

clicking shutter1 only affects shutter1. 
Clicking shutter2 should only affects shutter2.

before clicking shutter1 open all relays are off.
[image]

after only clicking shutter1 open relays1 and relay3 are ON.

[image]

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

mikep1998, 2021-02-07:
While testing by pressing the sutter2 open and close buttons. I found this in the logs"
```
02:00:13.677 SHT: Shtr2, Source Shutter, Powerstate 3, RelayMask 4, ManualChange 0
02:00:13.686 RSL: stat/tasmota_D9C6B8/RESULT = {"POWER3":"ON"}
02:00:13.689 RSL: stat/tasmota_D9C6B8/POWER3 = ON
02:00:13.693 RSL: stat/tasmota_D9C6B8/RESULT = {"ShutterPosition2":0}
02:00:13.748 SHT: Shtr2 Real 8842, Start 9592, Stop 0, Dir -1, Delay 4, Rtc 0.10 [s], Freq 500, PWM 454
02:00:13.752 RSL: stat/tasmota_D9C6B8/RESULT = {"Shutter2":{"Position":88,"Direction":-1,"Target":0}}
02:00:13.755 SHT: Time 2, toBA 0, cStop 0, cVelo 500, mVelo 1000, aVelo 250, mRun 50, aPos 8842, nStop 9342, Trgt 0, mVelo 250, Dir -1
02:00:13.817 CFG: Saved to flash at F8, Count 308, Bytes 4096
02:00:13.832 SHT: Time 4, toBA 0, cStop -750, cVelo 1000, mVelo 1000, aVelo 250, mRun 150, aPos 7092, nStop 7342, Trgt 0, mVelo 250, Dir -1
02:00:13.882 SHT: Time 5, toBA 0, cStop -1500, cVelo 1000, mVelo 1000, aVelo 0, mRun 200, aPos 6092, nStop 5592, Trgt 0, mVelo 250, Dir -1
02:00:13.932 SHT: Time 6, toBA 0, cStop -1500, cVelo 1000, mVelo 1000, aVelo 0, mRun 200, aPos 5092, nStop 4592, Trgt 0, mVelo 250, Dir -1
02:00:13.982 SHT: Time 7, toBA 0, cStop -1500, cVelo 1000, mVelo 1000, aVelo 0, mRun 200, aPos 4092, nStop 3592, Trgt 0, mVelo 250, Dir -1
02:00:14.033 SHT: Time 8, toBA 0, cStop -1500, cVelo 1000, mVelo 1000, aVelo 0, mRun 200, aPos 3092, nStop 2592, Trgt 0, mVelo 250, Dir -1
02:00:14.082 SHT: Time 9, toBA 0, cStop -1500, cVelo 1000, mVelo 1000, aVelo 0, mRun 200, aPos 2092, nStop 1592, Trgt 0, mVelo 250, Dir -1
02:00:14.131 SHT: Time 10, toBA 95, cStop -1500, cVelo 763, mVelo 1000, aVelo -237, mRun 200, aPos 1329, nStop 592, Trgt 0, mVelo 250, Dir -1
02:00:14.182 SHT: Time 11, toBA 93, cStop -776, cVelo 531, mVelo 1000, aVelo -232, mRun 152, aPos 798, nStop 553, Trgt 0, mVelo 250, Dir -1
02:00:14.233 SHT: Time 12, toBA 91, cStop -296, cVelo 304, mVelo 1000, aVelo -227, mRun 106, aPos 494, nStop 502, Trgt 0, mVelo 250, Dir -1
02:00:14.282 SHT: Time 13, toBA 85, cStop -28, cVelo 79, mVelo 1000, aVelo -225, mRun 60, aPos 415, nStop 466, Trgt 0, mVelo 250, Dir -1
02:00:14.284 SHT: Stop Shutter 1. Switchmode 0
02:00:14.287 SRC: Shutter
02:00:14.290 SHT: Switched relay 4 by Shutter
02:00:14.293 SHT: Shtr1, Source Shutter, Powerstate 0, RelayMask 4, ManualChange 0
02:00:14.295 SHT: Shtr2, Source Shutter, Powerstate 2, RelayMask 4, ManualChange 0
02:00:14.300 RSL: stat/tasmota_D9C6B8/RESULT = {"POWER3":"OFF"}
02:00:14.304 RSL: stat/tasmota_D9C6B8/POWER3 = OFF
02:00:14.306 SRC: Shutter
02:00:14.309 SHT: Switched relay 8 by Shutter
02:00:14.312 SHT: Shtr1, Source Shutter, Powerstate 0, RelayMask 8, ManualChange 0
02:00:14.315 SHT: Shtr2, Source Shutter, Powerstate 0, RelayMask 8, ManualChange 0
02:00:14.319 RSL: stat/tasmota_D9C6B8/RESULT = {"POWER4":"OFF"}
02:00:14.322 RSL: stat/tasmota_D9C6B8/POWER4 = OFF
02:00:14.325 SRC: Button
02:00:14.328 CMD: Group 0, Index 1, Command "PWM", Data "117"
02:00:14.342 RSL: stat/tasmota_D9C6B8/RESULT = {"PWM":{"PWM1":117,"PWM2":0}}
02:00:14.346 SHT: Shtr2 Real 415, Start 415, Stop 0, Dir 0, Delay 4, Rtc 0.65 [s], Freq 0, PWM 117
02:00:14.349 RSL: stat/tasmota_D9C6B8/SHUTTER2 = 4
02:00:14.353 RSL: stat/tasmota_D9C6B8/RESULT = {"Shutter2":{"Position":4,"Direction":0,"Target":0}}
02:00:14.817 CFG: Saved to flash at F7, Count 309, Bytes 4096
02:00:18.441 HTP: Console
02:00:20.595 CMD: pwm
02:00:20.597 SRC: WebConsole from 192.168.1.138
02:00:20.601 CMD: Group 0, Index 1, Command "PWM", Data ""
02:00:20.605 RSL: stat/tasmota_D9C6B8/RESULT = {"PWM":{"PWM1":117,"PWM2":0}}
```

Why is the shutter mask wrong?

mikep1998, 2021-02-07:
Maybe found an issue that is turning on relay 3 for shutter1 at line 1080 in the shutter code:

 if (ShutterGlobal.position_mode != SHT_TIME_UP_DOWN) ExecuteCommandPowerShutter(Settings.shutter_startrelay[index]+2, 1, SRC_SHUTTER);

adds +2 to the relay number. removed this and fixed my code with a servo. Now sure what this line is here for? May break someone else's shutter code.

Still looking at other issue with shutter1 moving when operating shutter2.

mikep1998, 2021-02-07:
I have also found that the line of code below is causing an Exception 0. When comment out this line the exception does not happen. When I put it back the Exception comes back.

line 386            Shutter[i].accelerator = - tmin(tmax( velocity_change_per_step_max*toBeAcc/100  , (velocity_change_per_step_max*9/10)), (velocity_change_per_step_max*11/10));

mikep1998, 2021-02-09:
Found another issue in the shutter code.

line 461     snprintf_P(scmnd, sizeof(scmnd), PSTR(D_CMND_PWM " %d" ),Shutter[i].pwm_value);

this command needs to include the shutter index. Other wise it is always working on shutter 1.

mikep1998, 2021-02-10:
@stefanbode Are you available to work on these issues? 

If so how to resolve this:

if (ShutterGlobal.position_mode != SHT_TIME_UP_DOWN) ExecuteCommandPowerShutter(Settings.shutter_startrelay[index]+2, 1, SRC_SHUTTER);

adds +2 to the relay number. removed this and fixed my code with a servo. Now sure what this line is here for? May break someone else's shutter code.

Change the +2 or just comment out the whole line?

Thanks
mike

stefanbode, 2021-02-14:
The idea is to have the servo power on relay 3. That's save energy when the servo do not move. It should be in the docu. Same for the stepper motor. Relay 3 is on the enable pin.

mikep1998, 2021-02-14:
@stefanbode Thank you for the reply. But you did not answer the question. 

How to resolve this bug in the code:

if (ShutterGlobal.position_mode != SHT_TIME_UP_DOWN) ExecuteCommandPowerShutter(Settings.shutter_startrelay[index]+2, 1, SRC_SHUTTER);

adds +2 to the relay number. removed this and fixed my code with a servo. Now sure what this line is here for? May break someone else's shutter code.

Change the +2 or just comment out the whole line?

Thanks
mike

stefanbode, 2021-02-18:
The reason is how TASMOTA handle PWM "relays". For these shutters the RELAY1 is ON/OFF, RELAY2 is DIRECTION and RELAY3 is the PWM. You can change this but with default configuration you get an additional relay as soon as you define one PWM. With this line I ensure the PWM is on. Otherwhise neither the stepper nor the servo will work. But I agree there is an issue, if you have 2 or 3 shutter because the PWM are always added to the END. Therefore the "+2" is not correct. I t must depend on the shutter number and the total shutters defined. uff... Maybe there is a better way to get the relay number, if you know it is PWM3.

stefanbode, 2021-02-18:
@arendst : Do we have a function where I can POWER ON the PWM2. My problem is: depending on the amount of "normal" relays I have no idea on which relay number the PWM2 is defined. I want to have a replacement for the bold one.
OLD:
ExecuteCommandPowerShutter(**Settings.shutter_startrelay[index]+2**, 1, SRC_SHUTTER)

arendst, 2021-02-18:
I will investigate. Give some time.

stefanbode, 2021-02-18:
For the stepper it is ok to untouch it, but for the servo it makes sense to switch it on/off so there is no PWM signal anymore on the servo. In the current commit I removed the line

mikep1998, 2021-02-19:
@stefanbode thank you for the relay fix. 

I still have 1 issue with this configuration. That is system reboot. 
I have narrowed it down to this section of code:

```void ShutterUpdateVelocity(uint8_t i)
{
  // No Logging allowed. Part of RTC Timer
  // will be calles through RTC every 50ms.
  Shutter[i].pwm_velocity += Shutter[i].accelerator;
  Shutter[i].pwm_velocity = tmax(0,tmin(Shutter[i].direction==1 ? ShutterGlobal.open_velocity_max : 
  Shutter[i].close_velocity_max,Shutter[i].pwm_velocity));
  if (ShutterGlobal.position_mode == SHT_COUNTER) {
  	Shutter[i].pwm_velocity = tmax(100,Shutter[i].pwm_velocity);
  }
}
```
When Shutter[i].accelerator is negative and Shutter[i].pwm_velocity goes to 0.
this causes a reboot some where else in the code. I have tried to narrow it down further without success. 
The code is very hard to understand with all the ? : ? :.

Have have found that if you Shutter[i].pwm_velocity for 0 and force to 100 then it all works. Although I do not believe this is a final solution. 

So changing the code to this seems to have resolved the problem.

```void ShutterUpdateVelocity(uint8_t i)
{
  // No Logging allowed. Part of RTC Timer
  // will be calles through RTC every 50ms.
  Shutter[i].pwm_velocity += Shutter[i].accelerator;
  Shutter[i].pwm_velocity = tmax(100,tmin(Shutter[i].direction==1 ? ShutterGlobal.open_velocity_max : 
  Shutter[i].close_velocity_max,Shutter[i].pwm_velocity));
}
```
Do you agree?

stefanbode, 2021-02-19:
@mikep1998 : yes I agree. The PWMfrequency is hardcoded in the Arduino SDK to 100Hz. You cannot get lower. I already submitted this yesterday as you also posted above in the case of SHT_COUNTER. For all other speed controlled shutter like the SERVO there is no limit of 100. This can and must go down to 0. I tested yesterday with the STEPPER motor and did not get any reboots. I also have a problem understanding why 0 is a problem. The only thing I can image is a WATCHDOG reboot. Please can you try to change the line from:
```
  Shutter[i].pwm_velocity = tmax(0,tmin(Shutter[i].direction==1 ? ShutterGlobal.open_velocity_max : 
  Shutter[i].close_velocity_max,Shutter[i].pwm_velocity));
```
to
```
  Shutter[i].pwm_velocity = tmax(1,tmin(Shutter[i].direction==1 ? ShutterGlobal.open_velocity_max : 
  Shutter[i].close_velocity_max,Shutter[i].pwm_velocity));
```
1 is the lowest value and it is fine. What kind of shutter mode do you have? where does the error occur?

mikep1998, 2021-02-19:
@stefanbode  Hello,

I am using the shutter code to tilt my blinds. I am controlling a MG995 servo.
I am using a PWMfrequency of 200.

I have tried pwm_velocity values 0, 1, 10, 50, 99 all of these fail with reboot. Only 100 or above work fine. I can not find the reason of the reboot. May be WATCHDOG  or DIVIDE by ZERO. If you know how to check which reboot error please tell me how.

The settings I use are:

SetOption80 1
Shuttermode 5
SetOption15 0
PWMfrequency 200
ShutterRelay1 1             // enable first shutter.
ShutterRelay2 3             // enable second shutter.
ShutterPwmRange1 100, 500
ShutterPwmRange2 100, 500
ShutterOpenDuration1 0.5
ShutterOpenDuration2 0.5
ShutterCloseDuration1 0.5
ShutterCloseDuration2 0.5
ShutterMotorDelay1 0.2
ShutterMotorDelay2 0.2
Restart 1

The error occurs when opening or closing the blinds. near the end of movement.

mikep1998, 2021-02-19:
@stefanbode Hello,

I just loaded the latest developement 9.3.0.1
This is the resboot reason when one of the shutters starts to move.
```
16:23:56.085 MQT: tele/blinds_livingroom/INFO3 = {"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["4000dce5","00000000","40000f68"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40215804","4024942e","4026eaa0","4024a0f6","4026eaad","4026eaf2","402465b0","40000f49","40000f49","40000e19","40105a71","40105a77","4010000d","4026753c","402674ed","401058b9","4010549e","40104ded","401053a1","401058b9","40103942","401037c3","4010549e","401058b9","40104ded","40103614","401053a1","40105679","40103942","401037c3","40100a40"]}}
```

When I change the code to this it works fine. 100 instead of 0.

```
void ShutterUpdateVelocity(uint8_t i)
{
  // No Logging allowed. Part of RTC Timer
  // will be calles through RTC every 50ms.
  Shutter[i].pwm_velocity += Shutter[i].accelerator;
  Shutter[i].pwm_velocity = tmax(100,tmin(Shutter[i].direction==1 ? ShutterGlobal.open_velocity_max : Shutter[i].close_velocity_max,Shutter[i].pwm_velocity));
  if (ShutterGlobal.position_mode == SHT_COUNTER) {
  	Shutter[i].pwm_velocity = tmax(100,Shutter[i].pwm_velocity);
  }
}
```

stefanbode, 2021-02-20:
Found it. Will fix today

mikep1998, 2021-02-20:
@stefanbode Hello,

I have tried the latest code with the 2 commits and the Exception 0 is still there.

```

15:28:49.130 MQT: tele/blinds_livingroom/INFO3 = {"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["4000dce5","00000000","40000f68"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["402124a8","4023b8de","40260864","4023c38e","40260871","402608b6","4023917c","40000f49","40000f49","40000f49","40000e19","4010552d","40105533","4010000d","40259304","402592b5","40100784","40100784","40105427","40102174","40105427","40102d0b","40102eec","40100784","401025a5","40102174","4010540f","401045ea","40104a5b","40100784","40104523"]}}

```
Let me know if you need anything else?

stefanbode, 2021-02-22:
Ok there is a DIV 0 exception then the velocity == 0. This causes the reboot. This is easy to fix. And maybe the 200 as a minimum might also be a solution. Currently, I do see some mistakes in the calculation of the stop-way and stop-time. This causes the velocity==0 before reaching the final position. I have now a testbench with a stepper and a servo because both use the accelerate/decelerate functions but in a very different way.

stefanbode, 2021-02-22:
OK, removed some fancy functions and streamline the decelerate process. The bug is removed for sure because this was part of the "adaption" procedure that is now completely removed. To Stop the servo or stepper this is now mathematically calculated and not "on the fly" corrected. Please can you give it a try and give feedback. I had some issues with my servo. Sometimes work, sometimes move nothing....

mikep1998, 2021-02-22:
@stefanbode Hello,

I have tried the latest code from your branch and it works without any errors. 

Thanks for the fix.

stefanbode, 2021-02-22:
Heureka. I was a little bit surprised about your fast servos. My did not work on that openduration.

mikep1998, 2021-02-22:
My servo are only turning the tilt operation of the blinds. 180 degrees from close to open to close. I have actually extended the time to 10 seconds and now they run quieter.

stefanbode, 2021-02-23:
hi mike. The deceleration is still part also of the servo. Just the calculation is simplified. I agree that there is no ramp required if the load on the servo is low. If you have heavy mass to move the ramp really helps to keep your servo healthy. 

last not least: please can you close the case. THX

closed 2021-02-23 by mikep1998

fabiorcosta, 2022-10-18:
@arendst I am trying to make 2 shutters working on a board with 4 relays
I did: 
SetOption80 1
ShutterRelay1 1
ShutterRelay2 1
Interlock 1,2 3,4

but relays 3 and 4 never gets interlocked. Can you advise?

stefanbode, 2022-10-19:
The shutterrelay2 must be 3. it is always the first relay to assign to the shuttergroup.