arendst/Tasmota #10867: DHT11/DHT22 Below freezing temp issue

alittletoolost, 2021-02-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
DHT11/DHT22 Temp Sensors when temp goes below freezing (32F or 0C) will give a false reading.  For instance if using the DHT11 the temp will jump back to 58 degrees F.  When using the am2301 for DHT22 the temp will go -5862.1 and then will continue reading the wrong number. The numbers will continue to change as temp changes however are obviously meaningless.    

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NODEMCU
- [ ] Tasmota binary firmware version number used: 9.2.0.6
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

14:15:26 RSL: tele/tasmota_E367A2/SENSOR = {"Time":"2021-02-07T14:15:26","AM2301":{"Temperature":0.1,"Humidity":55.6,"DewPoint":-7.8},"TempUnit":"C"}
14:17:56 RSL: tele/tasmota_E367A2/STATE = {"Time":"2021-02-07T14:17:56","Uptime":"0T00:12:40","UptimeSec":760,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"WIFI CORNHOLE","BSSId":"3C:84:6A:F0:9E:F6","Channel":9,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}
14:17:56 RSL: tele/tasmota_E367A2/SENSOR = {"Time":"2021-02-07T14:17:56","AM2301":{"Temperature":-3275.8,"Humidity":61.3,"DewPoint":-4879.2},"TempUnit":"C"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
load a dht11 or dht22 and put them in temps below 0 degrees C 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
expect temperature reading to be -0.1 ... 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
In chat the dht12 (#10632) suffered from a similar fate as the both the dht11 and dht22 and was fixed with coding edit however i was unable to reproduce as the code is different in the (xsns_06_dht.ino).  

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-02-07:
As said. This is duplicate but as I feel pity for all those users still fond of the crappy DHT11/DHT22 a quick update:

[image]

Your DHT11 is not supported below 0 degrees..... The calculation also does not support negative temperatures.

The DHT22 might support negative values

alittletoolost, 2021-02-07:
Unfortunately, I have a DHT22 and does not work either.  The attached log was copied when using the DHT22.  I know the DHT22 operates from -20C to 80C.  I have a bme280 ordered however it's in transit for a couple weeks.  I more curious if it was an error in the driver because the sensor is consistent.  It's always the same wrong temp number when it drops below 0. It's also oddly consistent with the DHT11 with the error number.  DHT11 goes to 58.6 after temp drops and dht22 goes to -5862.1 after it goes below freezing.  I guess I am nervous that the bme280 is going to have the same issue.

sfromis, 2021-02-07:
I'm getting negative temperatures from my outdoor BME280

alittletoolost, 2021-02-07:
Great I have other things to repurpose the DHT22.  Just needed to get sensor for automations outside.

closed 2021-02-07 by alittletoolost