arendst/Tasmota #10887: Tasmota client stability issues

cn-superman, 2021-02-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Wrong characters in "RESULT" subject

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ yes] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ yes] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ yes] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ yes] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ no] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ yes] Device used (e.g., Sonoff Basic): _8266-01____
- [ yes] Tasmota binary firmware version number used: _9.2.0.6____
  - [ no] Pre-compiled
  - [ yes] Self-compiled
- [yes ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
04:05:13.184 SRC: MQTT
04:05:13.187 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:05:13.194 MQT: NB/RESULT = {"ClientSend":"Done"}
04:05:13.235 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping."}
04:05:13.955 SRC: MQTT
04:05:13.959 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:05:13.966 MQT: NB/RESULT = {"ClientSend":"Done"}
04:05:13.973 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup."}
04:05:14.072 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping."}
04:05:14.520 SRC: MQTT
04:05:14.523 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:05:14.530 MQT: NB/RESULT = {"ClientSend":"Done"}
04:05:14.572 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping."}
04:05:15.075 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup.�} (The wrong character is the last one, invisible)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-02-09:
Well lucky for you you may solve this yourself.

The message is not part of Tasmota (Search code base and you won't find it) but is received from the attached Atmel device. You will need to fix the code on the Atmel device to supply valid responses.

ascillato2, 2021-02-09:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-09 by ascillato2

cn-superman, 2021-02-09:
Thank you~  Okay, let me try.