arendst/Tasmota #10897: Bear SSL Compile Error on ESP32 (env. webcam)

FotoFieber, 2021-02-09:
### PROBLEM DESCRIPTION
Target ESP32-webcam doesn't compile when USE_MQTT_TLS is defined.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): AI Thinker ESP32 CAM
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Enable TLS for MQTT according to https://tasmota.github.io/docs/TLS/

2. Enable target tasmota32-webcam.

3. Build tasmota32-webcam.

### EXPECTED BEHAVIOUR
Compile a binary.

### SCREENSHOTS
`
C:/Users/mm/Documents/PlatformIO/Tasmota-development-esp32/tasmota/tasmota.ino:45:25: fatal error: t_bearssl.h: No such file or directory

*******************************************************************
* Looking for t_bearssl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:t_bearssl.h"
* Web  > https://platformio.org/lib/search?query=header:t_bearssl.h
*
*******************************************************************

   #include <t_bearssl.h>                    // We need to include before "tasmota_globals.h" to take precedence over the BearSSL version in Arduino
                         ^
compilation terminated.
*** [.pio\build\tasmota32-webcam\src\WiFiClientSecureLightBearSSL.cpp.o] Error 1
*** [.pio\build\tasmota32-webcam\src\tasmota.ino.cpp.o] Error 1`

### ADDITIONAL CONTEXT
Compilation is then aborted.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-02-09:
As the docs already tell, the standard build setup does not allow adding stuff to the tasmota32-xxxxxx environments, only to the base "tasmota32". Also, as webcam is already a very heavy task for ESP32, adding the also rather heavy TLS would not be a good mix.

FotoFieber, 2021-02-09:
I followed the docs for the MQTT TLS setup and added 
#define USE_MQTT_TLS  
to user_config_override.h 

I didn't add stuff to tasmota32-xxxxxx environments.

sfromis, 2021-02-09:
You do. tasmota32-webcam is one of the tasmota-xxxxxx environments.

FotoFieber, 2021-02-09:
I did no modification to tasmota32-webcam.

I edited platformio_tasmota32.ini according to the description in the file:
```
; ***              BETA ESP32 Tasmota version                 ***
; ***  expect the unexpected. Some features not working!!!    ***

[platformio]

; *** Tasmota build variant selection
default_envs                =  ${build_envs.default_envs}
; *** **Uncomment by deleting** ";" in the line(s) below to select version(s)
;               tasmota32
                tasmota32-webcam
```

FotoFieber, 2021-02-09:
Patched platformio_tasmota_env32.ini and it is compiling now.

[env:tasmota32-webcam]
extends                 = env:tasmota32
board                   = esp32cam
board_build.f_cpu       = 240000000L
build_flags             = ${common32.build_flags} -DFIRMWARE_WEBCAM
lib_extra_dirs          = lib/libesp32, lib/lib_basic, **lib/lib_ssl**

sfromis, 2021-02-09:
Yes you did. Enabling TLS, will affect all builds you do thereafter, and you specifically selected tasmota32-webcam. "Add stuff to" does not mean changing the definition of tasmota32-webcam, only doing changes leading to that target being amended.

sfromis, 2021-02-09:
What I mean is that what you experienced is working as designed, not an issue. While it is not impossible to go outside the standard build setup, webcam and TLS is still a bad combo for the small processor.

FotoFieber, 2021-02-09:
Maybe we can add a warning message when the USE_MQTT_TLS is enabled on an esp32_webcam and let the user decide, if he wants to use it or not?

I will test the stability of this setup. Space doesn't seem to be a problem:
```
RAM:   [==        ]  22.2% (used 72736 bytes from 327680 bytes)
Flash: [=======   ]  67.5% (used 1370597 bytes from 2031616 bytes)
```

Using Tasmota without encryption is in my opinion a really bad practice unless you have a complete separate network. I have TLS enabled for MQTT on all my Tasmota devices. (And Web disabled.)

closed 2021-02-09 by ascillato2