arendst/Tasmota #10899: Restart every 20-25 minutes

mefisto22, 2021-02-10:
### PROBLEM DESCRIPTION
I have 20+ esp8266/esp32 devices with tasmota 9.2.0. My tasmota devices started to restart every 20-25 minutes.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff T1 1ch and Sonoff T1 2ch
- [x ] Tasmota binary firmware version number used: 9.2.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: NodeMCU-PyFlasher-4.0-x64.
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/nappali/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
stat/nappali/RESULT = {"Module":{"29":"Sonoff T1 2CH"}}
stat/nappali/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
`01:00:46 CMD: status 0
01:00:46 MQT: stat/nappali/STATUS = {"Status":{"Module":29,"DeviceName":"nappali","FriendlyName":["nappali","Tasmota2"],"Topic":"nappali","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:00:46 MQT: stat/nappali/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:04:12","StartupUTC":"2021-02-09T23:56:34","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2021-02-09T21:02:16","SaveCount":38,"SaveAddress":"F6000"}}
01:00:46 MQT: stat/nappali/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"409/699"}}
01:00:46 MQT: stat/nappali/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TorokHome_IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AA009","2805C8000100060000005A00000000000000","00000200","00006000","00000000"]}}
01:00:46 MQT: stat/nappali/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
01:00:46 MQT: stat/nappali/STATUS5 = {"StatusNET":{"Hostname":"nappali","IPAddress":"192.168.22.221","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.202","Mac":"D8:F1:5B:C8:F7:CB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:00:46 MQT: stat/nappali/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.22.202","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C8F7CB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:00:46 MQT: stat/nappali/STATUS7 = {"StatusTIM":{"UTC":"2021-02-10T00:00:46","Local":"2021-02-10T01:00:46","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:06","Sunset":"18:02"}}
01:00:46 MQT: stat/nappali/STATUS10 = {"StatusSNS":{"Time":"2021-02-10T01:00:46"}}
01:00:46 MQT: stat/nappali/STATUS11 = {"StatusSTS":{"Time":"2021-02-10T01:00:46","Uptime":"0T00:04:12","UptimeSec":252,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"TorokHome_IoT","BSSId":"E2:63:DA:71:95:C3","Channel":6,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:03"}}}
01:00:46 MQT: stat/nappali/STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40104a57","40101494","40260d7f","401037b3","40260d14","40260cbb","4025fe18","4025fe41","4025d8ac","40259c15","4025e5a3","4025e016","40267767","40267027","40245018","40000f49","40000f49","40000e19","40105a61","40105a67","4010000d","40265f10","40265ec1","4010595b","40100a40","4010595b","40106201","4010323f","40103420","40100a40","40106201"]}}``

```

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
https://imgur.com/a/OPu3iPO

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I use two unifi ap-ac-lr, with mikrotik crs328. Everything worker for half year, and without any change tasmota's started to restart every 20-25 minutes.
Mikrotik version: 6.47.1
Unifi controller version: 6.0.45 (Build: atag_6.0.45_14358)
Unifi ap version: 4.3.26.11358
I flashed sonoffs with usb to ttl, not OTA.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-02-10:
Hi,

This is a duplicated issue. You must allow NTP to sync.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-10 by ascillato2

mefisto22, 2021-02-10:
I set time 0,and local ntp server. But problem is not solved.