arendst/Tasmota #10906: Incorrect NTC resistance calculation in ADC_TEMP

FransOv, 2021-02-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The calculation of the resistor value of the NTC at line 362 in module xsns_02_analog.ino is incorrect and does not take into account the higher resolution of the esp32.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WEMOS D1Mini and Lolin ESP32_____
- [x] Tasmota binary firmware version number used: 9.1.0 problem also exists in 9.2.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PlatformIO and Tasmotizer 1.2_____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:32:27 MQT: stat/tasmota-c1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:32:27 MQT: stat/tasmota-c1/RESULT = {"Module":{"0":"Generic"}}
14:32:27 MQT: stat/tasmota-c1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"320":"Led_i1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:33:36 MQT: stat/tasmota-c1/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:33:37 MQT: stat/tasmota-c1/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:33:37 MQT: stat/tasmota-c1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:34:15 MQT: stat/tasmota-c1/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-c1","FriendlyName":["Sensor Caravan"],"Topic":"tasmota-c1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:34:15 MQT: stat/tasmota-c1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:46:19","StartupUTC":"2021-02-10T11:47:56","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-02-09T13:48:20","SaveCount":47,"SaveAddress":"F5000"}}
14:34:15 MQT: stat/tasmota-c1/STATUS2 = {"StatusFWR":{"Version":"9.1.0(tasmota)","BuildDateTime":"2021-02-08T16:42:57","Boot":31,"Core":"2_7_4_5","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"469/699"}}
14:34:15 MQT: stat/tasmota-c1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR61_EXT","Office24Automation"],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
14:34:15 MQT: stat/tasmota-c1/STATUS4 = {"StatusMEM":{"ProgramSize":608,"Free":392,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAE797","04368001","000000CF","010013C0","C000F981","00004004","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37","Sensors":"1,2,3,4,5,6"}}
14:34:15 MQT: stat/tasmota-c1/STATUS5 = {"StatusNET":{"Hostname":"tasmota-c1","IPAddress":"192.168.2.240","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:60:67:D3","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:34:15 MQT: stat/tasmota-c1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.33","MqttPort":1883,"MqttClientMask":"tasmota-c1","MqttClient":"tasmota-c1","MqttUser":"hamqtt","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:34:15 MQT: stat/tasmota-c1/STATUS7 = {"StatusTIM":{"UTC":"2021-02-10T13:34:15","Local":"2021-02-10T14:34:15","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:06","Sunset":"18:02"}}
14:34:15 MQT: stat/tasmota-c1/STATUS10 = {"StatusSNS":{"Time":"2021-02-10T14:34:15","ANALOG":{"Temperature":20.2},"TempUnit":"C"}}
14:34:15 MQT: stat/tasmota-c1/STATUS11 = {"StatusSTS":{"Time":"2021-02-10T14:34:15","Uptime":"0T01:46:19","UptimeSec":6379,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"NETGEAR61_EXT","BSSId":"44:A5:6E:A2:6C:6F","Channel":10,"RSSI":100,"Signal":-49,"LinkCount":2,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Not applicable

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
define ADC_TEMP
set adcparam with ntc parameters
14:49:22 MQT: stat/tasmota-c1/RESULT = {"AdcParam1":[2,16200,10000,3385]}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct calculation of the temperature based on the adcparam

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The statement to calculate the resistance of the NTC in module xsns_02_analog.ino at line nr. 362 is incorrect.
The statement currently is: double Rt = (adc * Adc[idx].param1) / (1024.0 * ANALOG_V33  - (double)adc);
In my opinion this statement has two errors:
- Multiplying the analog range by ANALOG_V33 (=3.3) is incorrect, multiplying 1024 by 3.3 does not give the correct resolution of the adc
- The statement should use ANALOG_RANGE instead of 1024 so it will use the correct analog range on an esp32
The correct statement then would be: double Rt = (adc * Adc[idx].param1) / (ANALOG_RANGE  - (double)adc);

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-02-10:
Thx.

Fixed for ESP32.

Remember Rt is not the NTC resistance but a calculated parameter in the Steinhart-Hart equation:

```
      int adc = AdcRead(Adc[idx].pin, 2);
      // Steinhart-Hart equation for thermistor as temperature sensor
      double Rt = (adc * Adc[idx].param1) / (ANALOG_RANGE * ANALOG_V33 - (double)adc);  // Shelly param1 = 32000 (ANALOG_NTC_BRIDGE_RESISTANCE)
      double BC = (double)Adc[idx].param3 / 10000;                                      // Shelly param3 = 3350 (ANALOG_NTC_B_COEFFICIENT)
      double T = BC / (BC / ANALOG_T0 + TaylorLog(Rt / (double)Adc[idx].param2));       // Shelly param2 = 10000 (ANALOG_NTC_RESISTANCE)
      Adc[idx].temperature = ConvertTemp(TO_CELSIUS(T));
```

As it stands it calculates the correct temperature for a shelly 2.5 using above parameters.

closed 2021-02-10 by ascillato2

refael6, 2021-11-14:
As FransOv say and I checked there is a mistake in the formula, no need to multiply by 3.3V.
double Rt = (adc * Adc[idx].param1) / (ANALOG_RANGE * ANALOG_V33 - (double)adc); 
Calculation of units
resistance != (bit*resistance)/(bit*Voltage-bit)
resistance = (bit*resistance)/(bit-bit)

the Right formula:
double Rt = (adc * Adc[idx].param1) / (ANALOG_RANGE - (double)adc); 

I'm new if github and tasmota hope I helped.

FransOv, 2021-11-14:
Just thinking. Theo stated that he got the correct temperature with the original formula on a Shelly 2.5. Could that be because the ADC in the Shelly does not use a 3.3V reference voltage, as most development boards do, but the plain1 volt for the Esp8266 ADC.

barbudor, 2021-11-14:
Yes, the formula is for the ESP ADC input, not for a specific development board.
If a development board has an additional voltage divider, then the formula is not valid
Cannot this be taken into consideration with the AdcParam command ?
If not you can suggest a way to improve the AdcParam command to cover more use cases

ascillato2, 2021-11-14:
https://github.com/arendst/Tasmota/issues/13668#issuecomment-968320441

refael6, 2021-11-15:
> Yes, the formula is for the ESP ADC input, not for a specific development board. If a development board has an additional voltage divider, then the formula is not valid Cannot this be taken into [...]

https://github.com/arendst/Tasmota/issues/13668#issuecomment-968925843