arendst/Tasmota #10910: esp32 and pn532 very slow

mefisto22, 2021-02-10:
PROBLEM DESCRIPTION
I tried new esp32 module with pn532 reader. Sometimes not read, but if read, then have 5s delay.

REQUESTED INFORMATION
Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!

[x ] Read the Contributing Guide and Policy and the Code of Conduct
[x ] Searched the problem in issues
[ x] Searched the problem in discussions
[ x] Searched the problem in the docs
[ x] Searched the problem in the chat
[ x] Device used (e.g., Sonoff Basic): ESP32-WROOM-32
[x ] Tasmota binary firmware version number used: 9.2.0.7
[ x] Pre-compiled
 Self-compiled
[ x] Flashing tools used: esptool
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/tasmota_0FD834/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
stat/tasmota_0FD834/RESULT = {"Module":{"1":"ESP32-DevKit"}}

18:45:06.653 RSL: stat/tasmota_0FD834/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"160":"Switch1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"2848":"PN532 Tx"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"2880":"PN532 Rx"},"GPIO28":{"0":"None"}}
18:45:06.662 RSL: stat/tasmota_0FD834/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:00:45.429 RSL: stat/tasmota_0FD834/STATUS = {"Status":{"Module":1,"DeviceName":"hades","FriendlyName":["hades"],"Topic":"tasmota_0FD834","ButtonTopic":"tasmota_0FD834","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[13,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:00:45.434 RSL: stat/tasmota_0FD834/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:32:13","StartupUTC":"2021-02-10T17:28:32","Sleep":50,"CfgHolder":4617,"BootCount":51,"BCResetTime":"2021-01-22T01:46:53","SaveCount":153}}
19:00:45.437 RSL: stat/tasmota_0FD834/STATUS2 = {"StatusFWR":{"Version":"9.2.0.7(sensors)","BuildDateTime":"2021-02-10T15:02:49","Core":"1_0_5_RC6","SDK":"v3.3.4-402-gcd59d107b","CpuFrequency":80,"Hardware":"ESP32","CR":"442/699"}}
19:00:45.440 RSL: stat/tasmota_0FD834/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TorokHome_IoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000001"]}}
19:00:45.450 RSL: stat/tasmota_0FD834/STATUS4 = {"StatusMEM":{"ProgramSize":1207,"Free":1984,"Heap":231,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","00152005","B7FFBFCF","01DA9BC4","60360CC7","00084052","20200000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,14,17,20,21,24,29,34,50","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,64,66,67,74"}}
19:00:45.455 RSL: stat/tasmota_0FD834/STATUS5 = {"StatusNET":{"Hostname":"tasmota_0FD834-6196","IPAddress":"192.168.22.239","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.202","Mac":"3C:61:05:0F:D8:34","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:00:45.458 RSL: stat/tasmota_0FD834/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0FD834","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:00:45.463 RSL: stat/tasmota_0FD834/STATUS7 = {"StatusTIM":{"UTC":"2021-02-10T18:00:45","Local":"2021-02-10T19:00:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:06","Sunset":"18:02"}}
19:00:45.468 RSL: stat/tasmota_0FD834/STATUS10 = {"StatusSNS":{"Time":"2021-02-10T19:00:45","Switch1":"ON"}}
19:00:45.474 RSL: stat/tasmota_0FD834/STATUS11 = {"StatusSTS":{"Time":"2021-02-10T19:00:45","Uptime":"0T00:32:13","UptimeSec":1933,"Heap":231,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"TorokHome_IoT","BSSId":"E2:63:DA:71:95:C3","Channel":6,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-02-10:
Hi,

Please, can you provide the console logs when you use the PN532?

Please, as requested in the template, set weblog to 4 BEFORE doing the tests so as to gather all debug information. Thanks

mefisto22, 2021-02-10:
``19:25:47.761 QPC: Reset
19:25:49.753 APP: Boot Count 52
19:25:50.169 HTP: Console
19:25:50.951 CFG: Saved, Count 155, Bytes 4096
19:25:51.011 RSL: tele/tasmota_0FD834/STATE = {"Time":"2021-02-10T19:25:51","Uptime":"0T00:00:11","UptimeSec":11,"Heap":235,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":76,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"TorokHome_IoT","BSSId":"E2:63:DA:71:95:C3","Channel":6,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:05"}}
19:25:51.034 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:25:51","Switch1":"ON"}
19:25:56.772 RSL: tasmota/discovery/3C61050FD834/config = {"ip":"192.168.22.239","dn":"hades","fn":["hades",null,null,null,null,null,null,null],"hn":"tasmota_0FD834-6196","mac":"3C61050FD834","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.2.0.7","t":"tasmota_0FD834","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[13,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":["Switch1",null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0, ...
19:25:56.775 RSL: tasmota/discovery/3C61050FD834/sensors = {"sn":{"Time":"2021-02-10T19:25:56","Switch1":"ON"},"ver":1}
19:25:59.173 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:25:59","PN532":{"UID":"E5739E23"}}
19:26:01.174 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:26:01","PN532":{"UID":"E5739E23"}}
19:26:03.191 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:26:03","PN532":{"UID":"E5739E23"}}
19:26:05.153 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:26:05","PN532":{"UID":"E5739E23"}}
19:26:06.900 WIF: Checking connection...
19:26:07.165 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:26:07","PN532":{"UID":"E5739E23"}}
19:26:09.165 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:26:09","PN532":{"UID":"E5739E23"}}
19:26:11.184 RSL: tele/tasmota_0FD834/SENSOR = {"Time":"2021-02-10T19:26:11","PN532":{"UID":"E5739E23"}}

TheReiner, 2021-02-25:
Same problem for me ...  With RFWF-02A.
Same Log ... WIF: Checking connection...
And 5-10s to turn off for exemple.
Thanks.

ascillato2, 2021-02-28:
Closing this issue as it is fixed by Theo. Please, test it from the development branch. Thanks.

closed 2021-02-28 by ascillato2

mefisto22, 2021-02-28:
It is working. But after scan rfid tag, the value not change back to null after few seconds. How can i set this? Because in Home Assistant can't trigger automation in second time, because the value is the same. If i set "expiration_after", sensor in Home Assistant goes unavailable, but with TelePeriod 10, every 10 seconds show the last scanned tag.

arendst, 2021-02-28:
As designed. The last scanned value will be returned.

mefisto22, 2021-02-28:
Is there any way to change this?

TheReiner, 2021-02-28:
> Same problem for me ... With RFWF-02A.
> Same Log ... WIF: Checking connection...
> And 5-10s to turn off for exemple.
> Thanks.

It's not fix for me... Any solution ?

arendst, 2021-02-28:
@TheReiner Stop polluting this issue. You have another problem.