arendst/Tasmota #10917: TuyaMCU temperature resolution broken

jaymemaurice, 2021-02-11:
DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Some TuyaMCU devices report temperatures as intger 1000% of actual = eg. 238 when it's really 23.8
Observed on Inkbird ITC-308

{"TuyaReceived":{"Data":"55AA000700086A020004000000EE6C","Cmnd":7,"CmndData":"6A020004000000EE","DpType2Id106":238,"106":{"DpId":106,"DpIdType":2,"DpIdData":"000000EE"}}}
03:55:03 TYA: fnId=72 is set for dpId=106
03:55:03 TYA: RX value 238 from dpId 106 

Adjusting TempRes from 0 to 3 will adjust from 238, to 238.0, 238.00, 238.000 accordingly

I believe the current relevant code for these sensors to be:
714: if ((fnId >= TUYA_MCU_FUNC_TEMP) && (fnId <= TUYA_MCU_FUNC_TIMER4)) {      // Sensors start from fnId 71
          if (packetValue != Tuya.Sensors[fnId-71]) {
            Tuya.SensorsValid[fnId-71] = true;
            Tuya.Sensors[fnId-71] = packetValue;
            SnsUpdate = true;
          }
        }

1292: if (sensor > 74) {
          res = 0;
        } else { res = Settings.flag2.temperature_resolution; }

        GetTextIndexed(sname, sizeof(sname), (sensor-71), kTuyaSensors);
        ResponseAppend_P(PSTR("\"%s\":%s"), sname,
                        (Tuya.SensorsValid[sensor-71] ? dtostrfd(Tuya.Sensors[sensor-71], res, tempval) : PSTR("null")));
        added = true;
      }

I think dtostrfd(str,Settings.flag2.temperature_resolution) is confused/pointless because Tuya.Sensors is uint16_t and nowhere is packetValue changed form int to float.
Either we are missing a separate scaling factor parameter or Settings.flag2.temperature_resolution should also be used to divide the packetValue with temperature_resolution as 0 to 3 by 1,10,100,1000 accordingly

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [✓] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [✓] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [✓] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [✓] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [✓] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [✓] Device used (e.g., Sonoff Basic): TuyaMCU
- [✓] Tasmota binary firmware version number used: 9.2.0
  - [✓] Pre-compiled
  - [ ] Self-compiled
- [✓] Flashing tools used: esptool.py
- [✓] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:26:01 SRC: Backlog
15:26:01 CMD: Group 0, Index 1, Command "MODULE", Data ""
15:26:01 RSL: stat/tasmota_6FE466/RESULT = {"Module":{"54":"Tuya MCU"}}
15:26:01 SRC: Backlog
15:26:01 CMD: Group 0, Index 1, Command "GPIO", Data "255"
15:26:01 RSL: stat/tasmota_6FE466/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}
15:26:04 {"TuyaReceived":{"Data":"55AA0007000868020004000000D753","Cmnd":7,"CmndData":"68020004000000D7","DpType2Id104":215,"104":{"DpId":104,"DpIdType":2,"DpIdData":"000000D7"}}}
15:26:04 TYA: fnId=71 is set for dpId=104
15:26:04 TYA: RX value 215 from dpId 104 
15:26:04 {"TuyaReceived":{"Data":"55AA0007000874020004000002C34D","Cmnd":7,"CmndData":"74020004000002C3","DpType2Id116":707,"116":{"DpId":116,"DpIdType":2,"DpIdData":"000002C3"}}}
15:26:04 TYA: fnId=0 is set for dpId=116
15:26:04 TYA: RX value 707 from dpId 116 
15:26:11 TYA: Send "55aa00000000ff"
15:26:11 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ✓] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:27:14 CMD: Status 0
15:27:14 SRC: WebConsole from 192.168.2.34
15:27:14 CMD: Group 0, Index 1, Command "STATUS", Data "0"
15:27:14 RSL: stat/tasmota_6FE466/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_6FE466","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T09:32:42","StartupUTC":"2021-02-11T04:54:32","Sleep":50,"CfgHolder":4617,"BootCount":35,"BCResetTime":"2021-02-06T06:05:42","SaveCount":164,"SaveAddress":"FA000"}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["923284-things",""],"TelePeriod":300,"Resolution":"D58180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","08000000","00006200","00000000"]}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS5 = {"StatusNET":{"Hostname":"tasmota_6FE466-1126","IPAddress":"192.168.254.216","Gateway":"192.168.254.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"BC:DD:C2:6F:E4:66","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6FE466","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS7 = {"StatusTIM":{"UTC":"2021-02-11T14:27:14","Local":"2021-02-11T15:27:14","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"08:04","Sunset":"18:03"}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS10 = {"StatusSNS":{"Time":"2021-02-11T15:27:14","TuyaSNS":{"Temperature":215.000},"TempUnit":"C"}}
15:27:14 RSL: stat/tasmota_6FE466/STATUS11 = {"StatusSTS":{"Time":"2021-02-11T15:27:14","Uptime":"0T09:32:42","UptimeSec":34362,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"923284-things","BSSId":"8E:DC:96:47:00:C4","Channel":6,"RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
XXX

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash Inkbird (note this had to be done manually, I unsoldered TW3YS completely hold EN and RST high and GPIO0 low while flashing, seems wiki is incomplete regarding requirement to hold EN high as when done in place on some devices VCC and EN may be connected?)
Apply template included above
TuyaMCU 72,102
Note temperature is off by 1000.0%
Adjust TempRes from TempRes 1 to TempRes 2, 
Note temperature is off by 1000.00% 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Temperature scaling should be adjusted by tempres or some other scaling factor

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Thanks!

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-02-11:
Hi,

The `TempRes` command is working fine. It is just how much decimals are going to be shown for temperature. This command has nothing to do with your issue.

Your issue is that your tuya device is sending the temperature multiplied by 10. You need to fix that on the tuya driver in Tasmota.

jaymemaurice, 2021-02-11:
@ascillato2, would it not make sense, in the TuyaDriver, to link TempRes with the divisor? If the purpose of TempRes is to show how many decimals, should those decimals not have some significance? 

Eg. if TuyaMCU device reports packetValue in integer steps eg. 22, 23, 24, why would we ever want to display/return 22.0, 23.0, 24.0?? If it reports integers like 222 223 231, we might want to display/return 22.2, 22.3, 23.1...

ascillato2, 2021-02-11:
`TempRes` command is for all temperature sensors. Tuya is just a special case in Tasmota.

jaymemaurice, 2021-02-11:
Can packetValue ever be a float as the TuyaDriver is currently programmed?
Is there ever a case where devices exist with temperature sensors connected to both TuyaMCU and the ESP directly?

ascillato2, 2021-02-11:
The TempRes command will not solve your issue due to is not what you expect. Your issue is just a need to divide by 10 the value is being sent by the TUYAMCU to Tasmota. That must be done in the Tuya driver.

ascillato2, 2021-02-11:
Checking the driver's code ( https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_16_tuyamcu.ino ), the temperature is sent to Tasmota as it is received (as long with all the rest of the sensors). So, you will need to add there a new check for the DPID of temperature and divide it by 10.

jaymemaurice, 2021-02-11:
This is my proposal to modify TuyaMCU driver, but in a way that can go upstream.
Currently TuyaMCU driver has no support for any device but an integer temperature sensor value...
'''lua
if ((fnId >= TUYA_MCU_FUNC_TEMP) && (fnId <= TUYA_MCU_FUNC_TIMER4)) {      // Sensors start from fnId 71
          if (packetValue != Tuya.Sensors[fnId-71]) {
            Tuya.SensorsValid[fnId-71] = true;
            Tuya.Sensors[fnId-71] = packetValue; **// struct has Tuya,Sensors this as integer **
            SnsUpdate = true;
          }
        }
        
I am thinking before places with dtostrfd(Tuya.Sensors[sensor-71] should be should go through a function which returns a float adjusted based on TempRes...

ascillato2, 2021-02-11:
> I am thinking before places with dtostrfd(Tuya.Sensors[sensor-71] should be should go through a function which returns a float adjusted based on TempRes...

Yes, that sound good. As you have the device to test, please, can you provide a PR for that?

Just a note. In the subroutine `void TuyaSensorsShow(bool json)` is the code that send the data to the rest of tasmota (by MQTT is `JSON` is 1 or to the webinterface). May be that could be the place for your fix. What do you think?

arendst, 2021-02-11:
There is no need to change to float. This acan always be done during the dtosrtfd process.

Before you make any change here you will need to check if every Tuya device reports it's temperature multiplied by 10. If so all you need to do is call dtostrfd like:

```
(Tuya.SensorsValid[sensor-71] ? dtostrfd((float)Tuya.Sensors[sensor-71] / 10, res, tempval) : PSTR("null")));
```
and
```
dtostrfd((float)Tuya.Sensors[1] / 10, Settings.flag2.temperature_resolution, tempval), TempUnit());
```
But again, only if you can guarantee that tuya always returns temps this way.
```

jaymemaurice, 2021-02-11:
I don't think all tuya devices are /10... I think that depends on the precision... I'm not a programmer and haven't set up a build environment but I am going to try building this in gitpod:

```lua
(Tuya.SensorsValid[sensor-71] ? dtostrfd((float)Tuya.Sensors[sensor-71] / pow(10, res), res, tempval) : PSTR("null")));
```

jaymemaurice, 2021-02-11:
I tried, it built but I am having trouble upgrading...

https://blush-silverfish-m0x63qz1.ws-us03.gitpod.io/files/download/?id=282e1655-f55b-4fd2-a0b3-613360c63e93

17:58:39 UPL: https://blush-silverfish-m0x63qz1.ws-us03.gitpod.io/files/download/?id=282e1655-f55b-4fd2-a0b3-613360c63e93
17:58:40 HTP: Web server active on tasmota_6FE466-1126 with IP address 192.168.254.216
17:58:41 HTP: Web server stopped
17:58:42 HTP: Web server active on tasmota_6FE466-1126 with IP address 192.168.254.216
17:58:43 TYA: Send "55aa00000000ff"
17:58:43 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
17:58:43 TYA: Heartbeat
17:58:44 RSL: stat/tasmota_6FE466/UPGRADE = {"Upgrade":"Failed Wrong HTTP Code"}

[image]

jaymemaurice, 2021-02-11:
I have been successful. Thank you.

I needed to change also:
          case 71:
            WSContentSend_Temp("", (float)Tuya.Sensors[0] / pow(10, Settings.flag2.temperature_resolution));
            break;

[image]

jaymemaurice, 2021-02-11:
Sorry for the mess. This is my first PR. Thanks.

closed 2021-02-11 by jaymemaurice

yvesdm3000, 2021-03-01:
By Beca BHT-002 needs the value to be divided by 2, therefore this 'hack' using TempRes doesn't work for me...

jaymemaurice, 2021-03-01:
@yvesdm3000 - I am not familiar with this device. Is it TuyaMCU?
It's not now broken, but still broken, correct??
I guess we need a divisor setting and then the more correct thing to do would be to make TempRes round a more precise temperature?

b0rder, 2021-04-16:
Hi, I faced related problem with AVATTO ME81AH during SET temperature.
To getting temperature I have a message:

```{"TuyaReceived":{"Data":"55AA0307000818020004000000E615","Cmnd":7,"CmndData":"18020004000000E6","DpType2Id24":230,"24":{"DpId":24,"DpIdType":2,"DpIdData":"000000E6"}}}```

That means temperature reported 23.0C.
But for Set Temperature, the value seems devided by 10. Below is the example where I send command to set temperature 27C (and it is successfully set to 27C on thermostat), but then reported back as 2.7C: 

```19:16:20.644 CMD: Group 0, Index 2, Command "TUYASEND", Data "16,27"
19:16:20.663 TYA: Send "55aa00060008100200040000001b3e"
19:16:20.667 RSL: stat/tasmota_C9937A/RESULT = {"TuyaSend":"Done"}
19:16:20.797 {"TuyaReceived":{"Data":"55AA03070008100200040000001B42","Cmnd":7,"CmndData":"100200040000001B","DpType2Id16":27,"16":{"DpId":16,"DpIdType":2,"DpIdData":"0000001B"}}}
19:16:20.801 TYA: fnId=72 is set for dpId=16
19:16:20.806 RSL: tele/tasmota_C9937A/SENSOR = {"TuyaSNS":{"TempSet":2.7}}```

grabnab, 2023-01-26:
> Hi, I faced related problem with AVATTO ME81AH during SET temperature. To getting temperature I have a message:
> 
> `{"TuyaReceived":{"Data":"55AA0307000818020004000000E615","Cmnd":7,"CmndData":"18020004000000E6","DpType2Id24":230,"24":{"DpId":24,"DpIdType":2,"DpIdData":"000000E6"}}}`
> 
> That means temperature reported 23.0C. But for Set Temperature, the value seems devided by 10. Below is the example where I send command to set temperature 27C (and it is successfully set to 27C on thermostat), but then reported back as 2.7C:
> 
> ```
> 19:16:20.663 TYA: Send "55aa00060008100200040000001b3e"
> 19:16:20.667 RSL: stat/tasmota_C9937A/RESULT = {"TuyaSend":"Done"}
> 19:16:20.797 {"TuyaReceived":{"Data":"55AA03070008100200040000001B42","Cmnd":7,"CmndData":"100200040000001B","DpType2Id16":27,"16":{"DpId":16,"DpIdType":2,"DpIdData":"0000001B"}}}
> 19:16:20.801 TYA: fnId=72 is set for dpId=16
> 19:16:20.806 RSL: tele/tasmota_C9937A/SENSOR = {"TuyaSNS":{"TempSet":2.7}}```
> ```

I have also the same Issue with the ME81H ... Sending: `Tuyasend2 16,10` reply: `SENSOR = {"Time":"2023-01-26T11:08:57","TuyaSNS":{"Temperature":21.0,"TempSet":1.0},"TempUnit":"C"}`

barbudor, 2023-01-26:
@grabnab Check https://tasmota.github.io/docs/Commands/#tuyatempsetres
Although it looks like this command is about the number of decimals, it is indeed applying a scaling factor on the response

From memory, it was a response of an long discussion as different devices were responding with different temperature scale

yvesdm3000, 2023-01-27:
> @grabnab Check https://tasmota.github.io/docs/Commands/#tuyatempsetres Although it looks like this command is about the number of decimals, it is indeed applying a scaling factor on the response [...]

And I think the choice was wrong , some devices need a scale factor of 2 and some of 5. Now we're stuck with only factors of 10. What do we do, add another command ? 
I have (had?) a branch with two command added but stopped finishing the feature because suddenly tempsetres appeared ...
You actually need two different scale factors , one for temperature measured and one for target temperature .

Yves

grabnab, 2023-01-27:
Maybe we can add an optional scale factor to the request. So everyone would be happy?

yvesdm3000, 2023-01-27:
> Maybe we can add an optional scale factor to the request. So everyone would be happy?

How do you propose to do that so that it works with tempsetres? It should work separately for Temperature and TempSet

PR #10922 (2021-02-12): TuyaMCU Precise Temperature
Sensor packetValue now interpreted as float based on configuration of  TempRes command.

## Description:

**Related issue (if applicable):** fixes #10917

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc6
  - [x] I accept the [CLA](https://github.c [... cut]