arendst/Tasmota #10984: Extra "TYA: Send ..." preceding and blocking intended "TYA: Send ..."

coby, 2021-02-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Minoston MS10W 3-Way Dimmer Switch
- [x] Tasmota binary firmware version number used: 9.2.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tuya-Convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:43:02 RSL: stat/tasmota_C844F2/RESULT = {"NAME":"MS10W","GPIO":[1,2272,1,2304,1,1,0,0,1,1,1,1,1,0],"FLAG":0,"BASE":54}
22:43:03 RSL: stat/tasmota_C844F2/RESULT = {"Module":{"0":"MS10W"}}
22:43:03 RSL: stat/tasmota_C844F2/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:47:05 RSL: stat/tasmota_C844F2/STATUS = {"Status":{"Module":0,"DeviceName":"Bedroom_2_Dimmer","FriendlyName":["Bedroom_2_Dimmer"],"Topic":"tasmota_C844F2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:01","StartupUTC":"2021-02-15T21:41:04","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-02-15T22:38:57","SaveCount":23,"SaveAddress":"F5000"}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"409/699"}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HouseWifiSSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS5 = {"StatusNET":{"Hostname":"Bedroom_2_Dimmer","IPAddress":"192.168.1.75","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.21","Mac":"98:F4:AB:**:**:**","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C844F2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS7 = {"StatusTIM":{"UTC":"2021-02-15T21:47:05","Local":"2021-02-15T22:47:05","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:57","Sunset":"18:10"}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS10 = {"StatusSNS":{"Time":"2021-02-15T22:47:05"}}
22:47:05 RSL: stat/tasmota_C844F2/STATUS11 = {"StatusSTS":{"Time":"2021-02-15T22:47:05","Uptime":"0T00:06:01","UptimeSec":361,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"HouseWifiSSID","BSSId":"B4:75:0E:43:BE:FB","Channel":2,"RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

21:31:23 CMD: Power On
--
21:31:23 SRC: WebConsole from 192.168.1.246
21:31:23 CMD: Group 0, Index 1, Command "POWER", Data "On"
21:31:23 TYA: Send "55aa00060008030200040000001a30"
21:31:23 TYA: Send "55aa0006000501010001010e"
21:31:23 MQT: stat/Bedroom_2_Dimmer/RESULT = {"POWER":"ON"}
21:31:23 MQT: stat/Bedroom_2_Dimmer/POWER = ON
21:31:23 CFG: Saved to flash at F5, Count 39, Bytes 4096
(Power does not turn on.)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I have two (in theory) identical dimmer light switches (Minoston MS10W). Both work as expected when using the buttons.

Switch 1 works with Console/UI commands (and HTTP, MQTT, etc.). It takes a "Power On" command and respond with "21:31:52 TYA: Send "55aa0006000501010001010e". Everything works as expected.

Switch 2 uses a preceding "TYA: Send "55aa00060008030200040000001a30"" command on the first attempt of sending "Power On" (or "Power Off"). The switch then ignores the following TYA: Send "55aa0006000501010001010e" (the correct one). Upon sending "Power On" again, it does NOT prepend the mystery "TYA: Send...", and it follows the correct (and only this time) "TYA: Send ..." command.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the response to entering "Power On" or "Power Off" in the console or toggling on the UI to respond by sending the same "TYA: Send ..." response, as it does in the same make/model switch elsewhere, instead of sometimes sending two commands on the first attempt to toggle the power.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

MattWestb, 2021-02-16:
@s-hadinger Is you sending one default responce to the device after revived commands / data from the device ?
If not its waiting for it and dont sending any more before its timing out.

All unicast made to / from ZCL compatible clusters is handled by the stack and broadcast is being sent 3 times without some kind of acks (cant knowing how many devices that is in the group).

All custom clusters is not "handled" by the stack as they is not one part of ZCL and shall being handled by the application in both ends (acked from the application in both ends) (application <-> application).

jaymemaurice, 2021-02-26:
@coby Can you send the output of "TuyaMCU" on both dimmers?
I'm interested to know the fnId/dpId mapping.

closed 2021-03-28 by github-actions[bot]