arendst/Tasmota #10985: Dimmer changes brightness on its own - TuyaMCU / Treatlife Dimmer.

jschwalbe, 2021-02-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x ] Device used (e.g., Sonoff Basic): Treatlife DS02S Dimmer
- [ x ] Tasmota binary firmware version number used: 9.2.0.7
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ Tamotizer ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:48:27.474 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
00:48:27.497 SRC: Backlog
00:48:27.500 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
00:48:27.507 MQT: stat/island_light/RESULT = {"NAME":"DS02S Dimmer","GPIO":[0,2272,0,2304,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":54}
00:48:27.713 SRC: Backlog
00:48:27.716 CMD: Group 0, Index 1, Command "MODULE", Data ""
00:48:27.720 MQT: stat/island_light/RESULT = {"Module":{"0":"DS02S Dimmer"}}
00:48:27.925 SRC: Backlog
00:48:27.928 CMD: Group 0, Index 1, Command "GPIO", Data "255"
00:48:27.936 MQT: stat/island_light/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
00:48:31.427 TYA: Send "55aa00000000ff"
00:48:31.470 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
none

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:48:51.601 CMD: Group 0, Index 1, Command "STATUS", Data "0"
00:48:51.607 MQT: stat/island_light/STATUS = {"Status":{"Module":0,"DeviceName":"Island Light","FriendlyName":["Island Light"],"Topic":"island_light","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:48:51.614 MQT: stat/island_light/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:55","StartupUTC":"2021-02-15T23:43:56","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"2020-11-03T17:52:11","SaveCount":2104,"SaveAddress":"FB000"}}
00:48:51.621 MQT: stat/island_light/STATUS2 = {"StatusFWR":{"Version":"9.2.0.7(tasmota)","BuildDateTime":"2021-02-15T21:26:10","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"422/699"}}
00:48:51.627 MQT: stat/island_light/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx-NoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00028009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
00:48:51.636 MQT: stat/island_light/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":0,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:48:51.644 MQT: stat/island_light/STATUS5 = {"StatusNET":{"Hostname":"island_light-6709","IPAddress":"192.168.40.146","Gateway":"192.168.40.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.40.1","Mac":"98:F4:AB:ED:DA:35","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:48:51.650 MQT: stat/island_light/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.25.81","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EDDA35","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:48:51.658 MQT: stat/island_light/STATUS7 = {"StatusTIM":{"UTC":"2021-02-15T23:48:51","Local":"2021-02-16T00:48:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:56","Sunset":"18:12"}}
00:48:51.664 MQT: stat/island_light/STATUS10 = {"StatusSNS":{"Time":"2021-02-16T00:48:51"}}
00:48:51.670 MQT: stat/island_light/STATUS11 = {"StatusSTS":{"Time":"2021-02-16T00:48:51","Uptime":"0T00:04:55","UptimeSec":295,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":7,"Fade":"OFF","Speed":40,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxx-NoT","BSSId":"FC:EC:DA:14:EA:83","Channel":1,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Here are two different logs. 
FIRST LOG:  pay attention to 00:45:09.641 and  00:45:59.906

00:44:15.465 HTP: Console
00:44:18.439 TYA: Send "55aa00000000ff"
00:44:18.481 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
00:44:18.484 TYA: Heartbeat
00:44:23.036 WIF: Checking connection...
00:44:29.446 TYA: Send "55aa00000000ff"
00:44:29.487 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
00:44:29.490 TYA: Heartbeat
00:44:40.675 CMD: so54 0
[... 167 lines cut ...]
_Steps to reproduce the behavior:_
Load console. 
```
power 0
dimmer 1
dimmer 2
dimmer 3
dimmer 4
dimmer 5
...
```
at some point it will go back to a different number dimmer setting. It happens on multiple switches but on this specific switch it seems to default to dimmer value of 7. 

### EXPECTED BEHAVIOUR
I wouldn't expect it to change dimmer values after I've changed them. Just stick to the one it should be.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
It seems at least partially related to https://github.com/arendst/Tasmota/issues/9864 so I'll just that there. Seems like @effelle had at least partially solved that issue. `fade` is OFF.

Also it seems that the "set point" (as I'm going to call it), in this case `DIMMER 7`, seems to be related to the `tuyasend2 3,xxx` command. As I was playing around with this more, I learned that for instance if I use `tuyasend2 3,400` then the "set point" is 40; `tuyasend2 3,200` goes to 20, etc. 

**(Please, remember to close the issue when the problem has been addressed)**

jschwalbe, 2021-02-16:
The more I play around with this, the more it seems like the problem revolves around this:```00:45:21.533 TYA: Send dim skipped value 10 for dpid 2
```it's not sending the dimmer command for some reason!  sending `tuyasend2 2,500` does work however

jaymemaurice, 2021-02-18:
I tried to reproduce on one of my dimmers. 
I could not reproduce on 9.1.0.2...

jaymemaurice, 2021-02-18:
Ahhh like this:

20:22:20.127 CMD: Dimmer 200
20:22:10.882 SRC: WebConsole from 192.168.2.34
20:22:10.885 CMD: Group 0, Index 1, Command "DIMMER", Data "200"
20:22:10.889 RSL: stat/tasmota_17A3A7/RESULT = {"Dimmer":20}
20:22:12.757 MQT: Attempting connection...

sirtow, 2021-02-22:
It seems the `TYA: Send dim skipped value`   is the reason(or symptom)  of my issue: #11078 .

jschwalbe, 2021-02-24:
> I tried to reproduce on one of my dimmers.
> I could not reproduce on 9.1.0.2...

FWIW, 9.1.0.2 is old. I'm using 9.2.0.7. 

But try this please:

`tuyasend2 2,1000` <-- should increase to full brightness, then type
`dimmer 50` (or try the slider from the UI)
do you see `Send dim skipped value 500 for dpid 2` and no change in actual brightness of the light?
`dimmer 50` a second time then works as expected.

The tuyasend2 command is what happens when you adjust the light manually (I think).

jaymemaurice, 2021-02-24:
Okay, can confirm I have reproduce on Costco dimmer in 9.2.0.7, 9.2.0, and 9.1.0, dimmer command after tuyasend =  Send dim skipped value <x>

9.1.0
23:16:08 CMD: TuyaSend2 22,1000
23:16:08 SRC: WebConsole from 192.168.2.50
23:16:08 CMD: Group 0, Index 2, Command "TUYASEND", Data "22,1000"
23:16:08 TYA: Send "55aa0006000816020004000003e814"
23:16:08 MQT: stat/tasmota_174DCD/RESULT = {"TuyaSend":"Done"}
23:16:08 {"TuyaReceived":{"Data":"55AA0007000816020004000003E815","Cmnd":7,"CmndData":"16020004000003E8","DpType2Id22":1000,"22":{"DpId":22,"DpIdType":2,"DpIdData":"000003E8"}}}
23:16:08 TYA: fnId=21 is set for dpId=22
23:16:08 TYA: RX value 1000 from dpId 22 
23:16:10 WIF: Checking connection...
23:16:10 TYA: Send "55aa00000000ff"
23:16:10 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:16:10 TYA: Heartbeat
23:16:13 CMD: dimmer 1
23:16:13 SRC: WebConsole from 192.168.2.50
23:16:13 CMD: Group 0, Index 1, Command "DIMMER", Data "1"
23:16:13 MQT: stat/tasmota_174DCD/RESULT = {"POWER":"ON","Dimmer":1}
23:16:13 TYA: Send dim skipped value 3 for dpid 22
23:16:14 CFG: Saved to flash at F4, Count 699, Bytes 4096
23:16:21 TYA: Send "55aa00000000ff"
23:16:21 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:16:21 TYA: Heartbeat
23:16:30 WIF: Checking connection...

9.2.7
23:08:04.778 CMD: TuyaSend2 22,1000
23:08:04.781 SRC: WebConsole from 192.168.2.50
23:08:04.784 CMD: Group 0, Index 2, Command "TUYASEND", Data "22,1000"
23:08:04.804 TYA: Send "55aa0006000816020004000003e814"
23:08:04.808 MQT: stat/tasmota_17A3A7/RESULT = {"TuyaSend":"Done"}
23:08:04.868 {"TuyaReceived":{"Data":"55AA0007000816020004000003E815","Cmnd":7,"CmndData":"16020004000003E8","DpType2Id22":1000,"22":{"DpId":22,"DpIdType":2,"DpIdData":"000003E8"}}}
23:08:04.871 TYA: fnId=21 is set for dpId=22
23:08:04.873 TYA: RX value 1000 from dpId 22 
23:08:04.876 SRC: Switch
23:08:04.879 CMD: Group 0, Index 3, Command "DIMMER", Data "100"
23:08:04.885 MQT: stat/tasmota_17A3A7/RESULT = {"POWER":"ON","Dimmer":100}
23:08:05.303 CFG: Saved to flash at F4, Count 120, Bytes 4096
23:08:10.968 CMD: dimmer 50
23:08:10.971 SRC: WebConsole from 192.168.2.50
23:08:10.974 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
23:08:10.980 MQT: stat/tasmota_17A3A7/RESULT = {"POWER":"ON","Dimmer":50}
23:08:11.037 TYA: Send dim skipped value 500 for dpid 22
23:08:11.303 CFG: Saved to flash at FB, Count 121, Bytes 4096
23:08:12.421 TYA: Send "55aa00000000ff"
23:08:12.431 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
23:08:12.433 TYA: Heartbeat
23:08:18.335 CMD: dimmer 50
23:08:18.338 SRC: WebConsole from 192.168.2.50
23:08:18.341 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
23:08:18.347 MQT: stat/tasmota_17A3A7/RESULT = {"POWER":"ON","Dimmer":50}
23:08:18.420 TYA: Send "55aa0006000816020004000001f41e"
23:08:18.437 {"TuyaReceived":{"Data":"55AA0007000816020004000001F41F","Cmnd":7,"CmndData":"16020004000001F4","DpType2Id22":500,"22":{"DpId":22,"DpIdType":2,"DpIdData":"000001F4"}}}
23:08:18.441 TYA: fnId=21 is set for dpId=22
23:08:18.444 TYA: RX value 500 from dpId 22

jaymemaurice, 2021-02-24:
LightSerialDuty near line 632 we set a timer to ignore the next dim commands
```lua
      if (duty < Settings.dimmer_hw_min) { duty = Settings.dimmer_hw_min; }  // dimming acts odd below 25(10%) - this mirrors the threshold set on the faceplate itself
        Tuya.ignore_dimmer_cmd_timeout = millis() + 250; // Ignore serial received dim commands for the next 250ms
```

TuyaProcessStatePacket near line 768 we have:
```lua
AddLog(LOG_LEVEL_DEBUG, PSTR("TYA: RX value %d from dpId %d "), packetValue, Tuya.buffer[dpidStart]);

...

          if (Tuya.ignore_dimmer_cmd_timeout < millis()) {
            if ((TasmotaGlobal.power || Settings.flag3.tuya_apply_o20) && ((Tuya.Levels[dimIndex] > 0) && (Tuya.Levels[dimIndex] != Tuya.Snapshot[dimIndex]))) { // SetOption54 - Apply SetOption20 settings to Tuya device
              Tuya.ignore_dim = true;
```            

When Tuya.ignore_dim is not false, we ignore the dim and reset the flag Tuya.ignore_dim to false...
I think we should be just checking Tuya.ignore_dimmer_cmd_timeout < millis() instead of Tuya.ignore_dim... unless I am missing something

jaymemaurice, 2021-02-24:
@jschwalbe can you test my pull request - #11121
I believe this fixes your issue.

closed 2021-02-25 by ascillato2

jschwalbe, 2021-03-10:
Is this in 9.3.1.1?

jaymemaurice, 2021-03-10:
@jschwalbe - I believe this went into 9.3.1.. Were you able to test? 
Did you reproduce something?

jschwalbe, 2021-03-10:
I think it is working! (Had an id-10T error for a sec but that’s fixed!)

jaymemaurice, 2021-03-11:
beeeaaauuttiful!