arendst/Tasmota #10990: ESP32 using integrated BT, beacon scan outputs wrong values

Elix-g, 2021-02-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I use some IBeacon for presence detection. One of the beacons outputs major / minor values of 0x000D / 0x0007. Scanning the beacon with an 8266 and HM-10 module results in MQTT messages with expected major / minor values of 000D / 0007. Scanning the same beacon with an ESP32 integrated BT module results in MQTT messages with values of 0D00 / 0700. Firmware for both 8266 and ESP32 has been compiled from the same source.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): JoyIt NodeMCU  ESP32_____
- [X] Tasmota binary firmware version number used: 9.2.0.3_____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: VSC_____
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
/stat/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"}}
/stat/RESULT = {"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
/stat/STATUS = {"Status":{"Module":1,"DeviceName":"TasmoESP32","FriendlyName":["TasmoESP32"],"Topic":"tasmoesp32","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"4T21:37:20","StartupUTC":"2021-02-11T17:06:56","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-01-20T01:15:11","SaveCount":52}}
/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(tasmota)","BuildDateTime":"2021.01.21 00:50:42","Core":"1_0_5_RC6","SDK":"v3.3.4-402-gcd59d107b","CpuFrequency":80,"Hardware":"ESP32","CR":"492/699"}}
/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"syslog.privnet.xmas","LogPort":514,"SSId":["fingerwech",""],"TelePeriod":180,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00206000","00000002"]}}
/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1514,"Free":1984,"Heap":22,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000407","8F8A45CE","00010001","00001C47","010080C0","00800805","80020000","00200000"],"Drivers":"1,2,3,4,5,8,9,10,17,24,26,29,32,41,50,52,81","Sensors":"1,2,3,5,9,10,37,52,62"}}
/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmo-esp32.privnet.xmas","IPAddress":"192.168.179.232","Gateway":"192.168.179.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.179.23","Mac":"24:6F:28:0B:10:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.privnet.xmas","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B1018","MqttUser":"tasmoesp32","MqttCount":64,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-16T14:44:17","Local":"2021-02-16T15:44:17","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:56","Sunset":"18:12"}}
/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-16T15:44:17","BLE":{"scans":21004,"adverts":304906,"devices":2,"resets":713}}}
/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-16T15:44:17","Uptime":"4T21:37:21","UptimeSec":423441,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":64,"Wifi":{"AP":1,"SSId":"fingerwech","BSSId":"44:4E:6D:40:CB:AD","Channel":6,"RSSI":100,"Signal":-27,"LinkCount":2,"Downtime":"0T00:02:01"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Use BT integrated in ESP32 and scan some Ibeacon

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

MQTT messages with correct major/minor values

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Elix-g, 2021-03-14:
Some more information: It's not a hardware beacon that I use but a software running on Android:

[(https://play.google.com/store/apps/details?id=net.alea.beaconsimulator&hl=de&gl=US)]

To exclude a software issue, I've also tested another one:

[https://play.google.com/store/apps/details?id=com.davidgyoungtech.beaconscanner&hl=de&gl=US]

The results are identical, so root cause is not the beacon.

Jason2866, 2021-04-02:
Can you try latest development version? There are some changes made.

Elix-g, 2021-04-02:
Thank you for taking care of this issue!

I've tried the latest dev. version compiled by myself since I need MQTT TLS. Recent attempt results in a firmware that constantly restarts the device right after an iBeacon has been detected. Unfortunately, hi and lo are still swapped for major/minor. Compiled with following parameters changed:

_platformio_tasmota32.ini_
```
tasmota32-DE
lib/libesp32_div
```

_my_user_config.h_
```
#define USE_MQTT_TLS
  #define USE_MQTT_TLS_FORCE_EC_CIPHER
  #define USE_MQTT_TLS_DROP_OLD_FINGERPRINT

  #define SUPPORT_MQTT_EVENT

  #define USE_BMP
  #define USE_BH1750

  #define USE_IBEACON_ESP32

```

Unrelated hint, got the following warning during compilation:

> C:\users\%username%\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
> C:\users\%username%\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
>                  uint8_t * last_out8 = &result[c_longs-1];
>                                        ^
> C:\users\%username%\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
>                  uint8_t * last_data8 = &last_data;
>                                         ^

Jason2866, 2021-04-02:
Ping @btsimonh can you take a look?

The compiler warning is in Arduino ESP32 Core and NOT from Tasmota

btsimonh, 2021-04-08:
@Elix-g - sorry for the delayed response.
It's very likely that we should byte-reverse the values. It seems that for everything else in BLE we have to....
Unfortunately, very few (1) person has commented in real iBeacon usage, until now.  And I don't happen to own a real one!.
I put a PR in today to ensure that iBeacons with identical UUIDs are seen as different iBeacons (basically, it will store by MAC, not by UUID, but only found this issue after submitting.

So... pls confirm...  you are convinced that the major and minor reported are byte reversed (both are 16 bit values)?
If so, I'll modify the code to byte reverse them at capture, then we would be golden?

br,
Simon

Elix-g, 2021-04-08:
@btsimonh - no problem, appreciate that you take care of this issue!

Indeed, byte-swapping major and minor would do the magic and resolve this issue.

About your latest PR, please see [#8693](https://github.com/arendst/Tasmota/issues/8693) and [#8745](https://github.com/arendst/Tasmota/issues/8745). Originally, the identification of IBeacons was based on the MAC address. Upon my request, gemu2015 kindly modified the code in order to identify tokens based on their UID. Background: Apple devices which send out IBeacons, can uniquely be identified with their MAC address. A software emulation of IBeacons running on an Android device depends on Android system functions and on most smartphones, the send out MAC is being changed frequently. In that moment, a unique identification by the MAC address isn't possible anymore. The common denominator is the UID which don't change on either platform.

Many thanks and BR,
Elix

btsimonh, 2021-04-08:
haha! 
see https://github.com/arendst/Tasmota/discussions/11601 and discuss!
s

btsimonh, 2021-04-08:
seems we need a scheme whereby we cope with both....  not sure how.  Since UUID is used for multiple iBeacon for the same 'company/use', major and minor identify the individual iBeacon itself.  So propose a solution that works for a common uuid for a bunch of beacons, yet a changing mac address (damn phones!).
s

btsimonh, 2021-04-08:
actually, reviewing both previous issues, you will now get a zero RSSI (once) with the old MAC? correct?  But apart from that, all is still OK.  You get both UUID and MAC (and major & minor) in each message - so if you ignore messages with RSSI=0, then it's pretty similar, except now you can have two iBeacons with the same UUID?
Pls confirm....

btsimonh, 2021-04-09:
@Elix-g - pls see comments in 
https://github.com/arendst/Tasmota/pull/11629
and test my new branch if you can & report.
I don't want to do a third PR without some testing! :)

Elix-g, 2021-04-09:
@btsimonh - read through the discussions now. Of course it's totally fine to consider just the official Apple definition. I've compiled your latest branch now and the format of the JSON message looks fine for me. I can pick they key/value pairs I want to consider and ignore what I want to ignore - absolutely fine for me.

About the original issue: Major and minor are still byte swapped unfortunately.

Many thanks,
Elix

btsimonh, 2021-04-09:
this was my change:
```
#define ENDIAN_CHANGE_U16(x) ((((x)&0xFF00) >> 8) + (((x)&0xFF) << 8))

//      uint16_t    Major = ENDIAN_CHANGE_U16(oBeacon.getMajor());
//      uint16_t    Minor = ENDIAN_CHANGE_U16(oBeacon.getMinor());
      uint16_t    Major = oBeacon.getMajor();
      uint16_t    Minor = oBeacon.getMinor();
      uint8_t     PWR   = oBeacon.getSignalPower();

      DumpHex((const unsigned char*)&UUID,16,ib.UID);
      DumpHex((const unsigned char*)&Major,2,ib.MAJOR);
      DumpHex((const unsigned char*)&Minor,2,ib.MINOR);
```

Basically, I removed the ENDIAN_CHANGE_U16.
I can't see what would go wrong....

Pls, to absolutely confirm you are using the new driver, not the old one.  Do you have "PERSEC":<number> in your MQTT messages? (drop a few here...)

btsimonh, 2021-04-09:
AHH...  also tripple check that you don't have TWO xsns_52_esp32_ibeacon_ble.ino and xsns_52_ESP32_ibeacon_BLE.ino in your repo that you are building - my VSCode seems to think I have two in a search!!!  The repo on github seems to have only the correct one though.  If you do, I'll need to rename to something else and commit to clear that up! damn....

btsimonh, 2021-04-09:
from my Android phone using 'Beacon Simulator' and 1/2:
`11:12:29.476 RSL: tele/tasmota_E89E98/SENSOR = {"Time":"2021-04-09T11:12:29","IBEACON":{"UID":"14155040A54045BFB369E4D5CBD3E627","MAJOR":"0001","MINOR":"0002","MAC":"41C474055151","RSSI":-64,"STATE":"ON","PERSEC":0}}`
Is this how you expect it?
s

Elix-g, 2021-04-09:
- Indeed I don't have PERSEC in the MQTT message. What did I miss?!

- For ESP32, there's just one ino in my repo: xsns_52_esp32_ibeacon_ble.ino. Besides, there's xsns_52_ibeacon.ino too but that shouldn't hurt I guess.

-And yeah, the output in your previous post looks pretty good and is something I can perfectly work with!

Thanks a lot for your patience!

btsimonh, 2021-04-09:
build the bluetooth version - vanilla it should just work.  Do NOT define any IBEACON defines....
You will need to 'ibeacon 1', and if using random addresses (phones), 'BLEAddrFilter 1' to enable them.
Mostly, we put these in autoexec.bat, as they are not saved through reboot.
The dev binary would also work - but I'd rather you tested my latest commit ahead of the PR, as it addresses the UUID/MAJOR/MINOR identification thing.

Elix-g, 2021-04-09:
I think you can PR:

[image]

major/minor are correct now!

Jason2866, 2021-04-09:
Thx, @all here! I think issue can be closed.

Elix-g, 2021-04-09:
No, thank YOU!

closed 2021-04-09 by Elix-g