arendst/Tasmota #10994: ESP32 tasmota can't show wifi AP

alexandre72, 2021-02-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I've tasmitize an ESP32 (Wemos D1 ESP32) using procedure available on tasmota website but when I'm trying to see WIFI AP from my smatphone(or other WIFI scanner tools) can't see it. Using serial console I see:

17:37:54.874 -> 00:00:00 CFG: Loaded, Count 16
17:37:54.874 -> 00:00:00 QPC: Count 1
17:37:54.874 -> 00:00:00 CFG: CR 344/699, Busy 0
17:37:54.907 -> 00:00:00 SRC: Restart
17:37:54.907 -> 00:00:00 Project tasmota Tasmota Version 9.2.0(tasmota)-1_0_4_2(2020-12-21T15:35:43)
17:37:55.040 -> 00:00:00 WIF: Checking connection...
17:37:55.040 -> 00:00:00 WIF: WifiManager active for 3 minutes
17:37:55.372 -> 00:00:00 WIF: Wifimanager set AccessPoint
17:37:55.903 -> 00:00:01 HTP: Web server active on tasmota_CB0A00-2560 with IP address 192.168.4.1
17:38:01.776 -> 00:00:06 QPC: Reset
17:38:03.767 -> 00:00:08 APP: Boot Count 10
17:38:04.199 -> 00:00:09 CFG: Saved, Count 17, Bytes 4096

 
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Wemos D1 ESP32
- [X ] Tasmota binary firmware version number used: 9.2
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: esptools.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
7:35:29.371 -> 00:02:35 SRC: Serial
17:35:29.371 -> 00:02:35 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
17:35:29.404 -> 00:02:35 SRC: Backlog
17:35:29.404 -> 00:02:35 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
17:35:29.404 -> 00:02:35 RSL: stat/tasmota_CB0A00/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
17:35:29.669 -> 00:02:35 SRC: Backlog
17:35:29.669 -> 00:02:35 CMD: Group 0, Index 1, Command "MODULE", Data ""
17:35:29.669 -> 00:02:35 RSL: stat/tasmota_CB0A00/RESULT = {"Module":{"1":"ESP32-DevKit"}}
17:35:29.902 -> 00:02:35 SRC: Backlog
17:35:29.902 -> 00:02:35 CMD: Group 0, Index 1, Command "GPIO", Data "255"
17:35:29.902 -> 00:02:35 RSL: stat/tasmota_CB0A00/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"}}}
17:35:29.968 -> 00:02:35 RSL: stat/tasmota_CB0A00/RESULT = {"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
7:36:06.943 -> 00:00:05 CMD: Group 0, Index 1, Command "STATUS", Data "0"
17:36:06.943 -> 00:00:05 RSL: stat/tasmota_CB0A00/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_CB0A00","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:36:06.976 -> 00:00:05 RSL: stat/tasmota_CB0A00/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:06","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"1970-01-01T00:00:00","SaveCount":14}}
17:36:07.042 -> 00:00:05 RSL: stat/tasmota_CB0A00/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:35:43","Core":"1_0_4_2","SDK":"v3.3.4-232-g44ec7972b","CpuFrequency":80,"Hardware":"ESP32","CR":"344/699"}}
17:36:07.042 -> 00:00:05 RSL: stat/tasmota_CB0A00/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
17:36:09.166 -> 00:00:07 RSL: stat/tasmota_CB0A00/STATUS4 = {"StatusMEM":{"ProgramSize":1359,"Free":1984,"Heap":111,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29","Sensors":"1,2,3,5,6,62"}}
17:36:09.166 -> 00:00:07 RSL: stat/tasmota_CB0A00/STATUS5 = {"StatusNET":{"Hostname":"tasmota_CB0A00-2560","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"9C:9C:1F:CB:0A:00","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
17:36:09.200 -> 00:00:07 RSL: stat/tasmota_CB0A00/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CB0A00","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:36:09.233 -> 00:00:07 RSL: stat/tasmota_CB0A00/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:00:07","Local":"1970-01-01T00:00:07","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
17:36:09.233 -> 00:00:07 RSL: stat/tasmota_CB0A00/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:07"}}
17:36:09.233 -> 00:00:07 RSL: stat/tasmota_CB0A00/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:07","Uptime":"0T00:00:06","UptimeSec":6,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Flash wemos D1 ESP32 with esptool following procedure from website. And try to scan wifi AP.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
See WIFI AP

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-02-24:
Please try with latest Tasmota release version. Erase the complete flash before (with esptool.py).

ascillato2, 2021-02-24:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-24 by ascillato2

Jason2866, 2021-02-24:
Tested no issue
```
Using 'COM4' as serial port.
Writing at 0x000b8000... (100 %)Wrote 1176096 bytes (701082 compressed) at 0x00010000 in 16.7 seconds (effective 563.2 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[13:29:14]
[13:29:14]lib/libesp32/LITTLEFS/src/lfs.c:1003:error: Corrupted dir pair at {0x0, 0x1}
[13:29:14]E (23) esp_littlefs: mount failed,  (-84)
[13:29:15]E (23) esp_littlefs: Failed to initialize LittleFS
[13:29:15]00:00:00.111 UFS: FlashFS mounted with 56 kB free
[13:29:15]00:00:00.115 CFG: Loaded from Nvm, Count 0
[13:29:15]00:00:00.117 CFG: Use defaults
[13:29:15]00:00:00.249 QPC: Reset
[13:29:15]00:00:00.257 Project tasmota Tasmota Version 9.3.1.1(tasmota)-1_0_5_RC6(2021-02-24T10:04:39)
[13:29:15]00:00:00.487 WIF: WifiManager active for 3 minutes
[13:29:16]00:00:01.195 HTP: Web server active on tasmota_362330-0816 with IP address 192.168.4.1
[13:29:56]00:00:41.402 WIF: Hostname %s-%04d, SSId1 xxxx, SSId2 , CORS 
[13:29:58]00:00:42.763 APP: Restarting
[13:29:58]ets Jun  8 2016 00:22:57
[13:29:58]
[13:29:58]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:29:58]configsip: 0, SPIWP:0xee
[13:29:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:29:58]mode:DOUT, clock div:2
[13:29:58]load:0x3fff0018,len:4
[13:29:58]load:0x3fff001c,len:1044
[13:29:58]load:0x40078000,len:10124
[13:29:58]load:0x40080400,len:5828
[13:29:58]entry 0x400806a8
[13:29:59]
[13:29:59]00:00:00.027 UFS: FlashFS mounted with 52 kB free
[13:29:59]00:00:00.084 CFG: Loaded from File, Count 5
[13:29:59]00:00:00.098 QPC: Count 1
[13:29:59]00:00:00.106 Project tasmota Tasmota Version 9.3.1.1(tasmota)-1_0_5_RC6(2021-02-24T10:04:39)
[13:29:59]00:00:00.798 WIF: Connecting to AP1 xxxx in mode 11n as tasmota_362330-0816...
[13:30:03]00:00:04.740 WIF: Connected
[13:30:04]13:30:04.096 HTP: Web server active on tasmota_362330-0816 with IP address 192.168.2.169
[13:30:05]13:30:05.035 RSL: tele/tasmota_362330/INFO1 = {"Module":"ESP32-DevKit","Version":"9.3.1.1(tasmota)","FallbackTopic":"cmnd/DVES_362330_fb/","GroupTopic":"cmnd/tasmotas/"}
[13:30:05]13:30:05.041 RSL: tele/tasmota_362330/INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_362330-0816","IPAddress":"192.168.2.169"}
[13:30:05]13:30:05.052 RSL: tele/tasmota_362330/INFO3 = {"RestartReason":"Software reset CPU"}
[13:30:05]13:30:05.842 QPC: Reset
[13:30:10]13:30:10.777 RSL: tele/tasmota_362330/STATE = {"Time":"2021-02-24T13:30:10","Uptime":"0T00:00:12","UptimeSec":12,"Heap":242,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":241,"MqttCount":0,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:05"}}
```