arendst/Tasmota #10998: zigbee2mqtt seems not to work

muebau, 2021-02-17:
### PROBLEM DESCRIPTION
It seems like the bridge does not work like I would expect it to do. NodeRED zigbee2mqtt
A command like:
```
publish "{"state": ""}" to topic zigbee2mqtt/bridge/info
```
does not give any reponse.

If I trigger some Zigbee sensor (switch some power plug) there is no response at zigbee2mqtt.
I found there is some output at topic:
```tele/tasmota_04DA1C/SENSOR```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:07:15.495 SRC: Backlog
12:07:15.497 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
12:07:15.503 MQT: stat/tasmota_04DA1C/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:07:15.723 SRC: Backlog
12:07:15.725 CMD: Group 0, Index 1, Command "MODULE", Data ""
12:07:15.729 MQT: stat/tasmota_04DA1C/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
12:07:15.972 SRC: Backlog
12:07:15.974 CMD: Group 0, Index 1, Command "GPIO", Data "255"
12:07:15.981 MQT: stat/tasmota_04DA1C/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
12:07:34.041 WIF: Checking connection...
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
11:56:35.204 MQT: stat/tasmota_04DA1C/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_04DA1C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:56:35.208 MQT: stat/tasmota_04DA1C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:13:25","StartupUTC":"2021-02-16T10:43:10","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2021-02-12T09:04:30","SaveCount":19,"SaveAddress":"FC000"}}
11:56:35.211 MQT: stat/tasmota_04DA1C/STATUS2 = {"StatusFWR":{"Version":"9.2.0.6(zbbridge)","BuildDateTime":"2021-02-09T14:49:13","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"417/699"}}
11:56:35.214 MQT: stat/tasmota_04DA1C/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hidden",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
11:56:35.218 MQT: stat/tasmota_04DA1C/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":1216,"Heap":24,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
11:56:35.223 MQT: stat/tasmota_04DA1C/STATUS5 = {"StatusNET":{"Hostname":"tasmota_04DA1C-6684","IPAddress":"192.168.0.54","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"C8:2B:96:04:DA:1C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:56:35.228 MQT: stat/tasmota_04DA1C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.55","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_04DA1C","MqttUser":"muebau","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:56:35.233 MQT: stat/tasmota_04DA1C/STATUS7 = {"StatusTIM":{"UTC":"2021-02-16T10:56:35","Local":"2021-02-16T11:56:35","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:56","Sunset":"18:12"}}
11:56:35.236 MQT: stat/tasmota_04DA1C/STATUS10 = {"StatusSNS":{"Time":"2021-02-16T11:56:35"}}
11:56:35.242 MQT: stat/tasmota_04DA1C/STATUS11 = {"StatusSTS":{"Time":"2021-02-16T11:56:35","Uptime":"0T00:13:25","UptimeSec":805,"Vcc":3.504,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"hidden","BSSId":"C4:6E:1F:FF:13:28","Channel":9,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
A behaviour similar to zigbee2mqtt bridge, NodeRED compatibility.
But I may have misunderstood the way this implementation was intended to work. 😉

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-02-17:
Zigbee implementation in zbbridge is in no way related to zigbbe2mqtt nor compatible to.

muebau, 2021-02-17:
Ok thank you. I was not aware of that.

closed 2021-02-17 by muebau