arendst/Tasmota #11038: TUYA Zigbee SOS button registering but not receiving button press

sim0njo, 2021-02-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I received a TUYA Zigbee SOS button (https://www.aliexpress.com/item/1005001971589088.html?spm=a2g0s.9042311.0.0.7bdc4c4dfKMej5), I am able to register it with Sonoff ZIgbee Bridge running Tasmota 9.2.0.3, but when pressing the SOS button nothing is received by Tasmota.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____Sonoff Zigbee Bridge + TUYA Zigbee SOS button
- [x] Tasmota binary firmware version number used: _____9.2.0.3
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:33:47.954 CMD: Group 0, Index 1, Command "STATUS", Data "0"
23:33:47.958 MQT: avdln2/z2m/tzb/stat/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tzb","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:33:47.962 MQT: avdln2/z2m/tzb/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:30:40","StartupUTC":"2021-02-18T22:03:07","Sleep":50,"CfgHolder":4617,"BootCount":47,"BCResetTime":"2021-01-20T19:41:11","SaveCount":81,"SaveAddress":"FA000"}}
23:33:47.964 MQT: avdln2/z2m/tzb/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0.3(zbbridge)","BuildDateTime":"2021-01-20T16:49:11","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"360/699"}}
23:33:47.966 MQT: avdln2/z2m/tzb/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dlink-1F40",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000080","00006000","00000000"]}}
23:33:47.971 MQT: avdln2/z2m/tzb/stat/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":1216,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":2,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
23:33:47.975 MQT: avdln2/z2m/tzb/stat/STATUS5 = {"StatusNET":{"Hostname":"tzb-5797","IPAddress":"192.168.0.96","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"98:F4:AB:E2:B6:A5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:33:47.979 MQT: avdln2/z2m/tzb/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E2B6A5","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:33:47.984 MQT: avdln2/z2m/tzb/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-18T22:33:47","Local":"2021-02-18T23:33:47","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:52","Sunset":"18:15"}}
23:33:47.987 MQT: avdln2/z2m/tzb/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-18T23:33:47"}}
23:33:47.990 MQT: avdln2/z2m/tzb/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-18T23:33:47","Uptime":"0T00:30:40","UptimeSec":1840,"Vcc":3.466,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"dlink-1F40","BSSId":"18:0F:76:3B:1F:40","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

23:24:25.176 CMD: Group 0, Index 1, Command "ZBPERMITJOIN", Data "2"
23:24:25.178 ZIG: ZbEZSPSend 22003C
23:24:25.179 ZIG: ZbEZSPSend 3600FCFF000036000000000000000B1E01030B3C01
23:24:25.182 MQT: avdln2/z2m/tzb/tele/RESULT = {"ZbState":{"Status":21,"Message":"Pairing mode enabled"}}
23:24:25.185 MQT: avdln2/z2m/tzb/stat/RESULT = {"ZbPermitJoin":"Done"}
23:24:25.369 ZIG: {"ZbEZSPReceived":"220000"}
23:24:25.378 ZIG: {"ZbEZSPReceived":"36000086"}
23:24:25.383 ZIG: {"ZbEZSPReceived":"4500050000360000000000000086FF000000FFFF030B3C01"}
23:24:26.343 ZIG: {"ZbEZSPReceived":"3F0006FCFF0000360000000000000086010000"}
23:24:32.463 ZIG: {"ZbEZSPReceived":"2300010166FD784305FEFF27718404"}
23:24:32.668 ZIG: {"ZbEZSPReceived":"240066FD784305FEFF27718401000000"}
23:24:32.671 MQT: avdln2/z2m/tzb/tele/RESULT = {"ZbState":{"Status":34,"IEEEAddr":"0x847127FFFE054378","ShortAddr":"0xFD66","ParentNetwork":"0x0000","JoinStatus":1,"Decision":0}}
23:24:33.027 ZIG: {"ZbEZSPReceived":"6200784305FEFF277184"}
23:24:33.031 ZIG: {"ZbEZSPReceived":"45000400001300000000040000FEECD766FDFFFF0C8D66FD784305FEFF2771848002"}
23:24:33.035 MQT: avdln2/z2m/tzb/tele/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"0x847127FFFE054378","ShortAddr":"0xFD66","PowerSource":false,"ReceiveWhenIdle":false,"Security":false}}
23:24:33.037 ZIG: ZbEZSPSend 34000066FD000005000000400100000C01030C66FD
23:24:33.047 ZIG: {"ZbEZSPReceived":"34000089"}
23:24:33.375 ZIG: {"ZbEZSPReceived":"3F0006FDFF00001300000000040000FE000000"}
23:24:33.659 ZIG: {"ZbEZSPReceived":"45000000000580000040010000FFECD766FDFFFF060C0066FD010102"}
23:24:33.663 MQT: avdln2/z2m/tzb/tele/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
23:24:33.665 ZIG: ZbEZSPSend 34000066FD040100000101400100000D0107000D0004000500
23:24:33.670 ZIG: {"ZbEZSPReceived":"3F000066FD0000050000004001000089010000"}
23:24:33.675 ZIG: {"ZbEZSPReceived":"3400008A"}
[... 39 lines cut ...]
23:36:36.240 CMD: Group 0, Index 1, Command "ZBINFO", Data ""
23:36:36.246 MQT: avdln2/z2m/tzb/tele/SENSOR = {"ZbInfo":{"0xE2EB":{"Device":"0xE2EB","Name":"living/id1","IEEEAddr":"0x04CF8CDF3C79814A","ModelId":"lumi.remote.b686opcn01","Manufacturer":"LUMI","Endpoints":[1,2,3,4,5,6],"Config":[],"Reachable":true,"LastSeen":1122,"LastSeenEpoch":1613686674,"LinkQuality":63}}}
23:36:36.251 MQT: avdln2/z2m/tzb/tele/SENSOR = {"ZbInfo":{"0x00C4":{"Device":"0x00C4","Name":"ikea/id2","IEEEAddr":"0x000D6FFFFE19F00D","ModelId":"TRADFRI remote control","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":false,"LastSeen":31528,"LastSeenEpoch":1613656268,"LinkQuality":60}}}
23:36:36.256 MQT: avdln2/z2m/tzb/tele/SENSOR = {"ZbInfo":{"0x3DD6":{"Device":"0x3DD6","Name":"ikea/id3","IEEEAddr":"0xEC1BBDFFFE9AC0EF","ModelId":"TRADFRI on/off switch","Manufacturer":"IKEA of Sweden","Endpoints":[1],"Config":[],"Reachable":false,"LastSeen":156091,"LastSeenEpoch":1613531705,"LinkQuality":105}}}
23:36:36.262 MQT: avdln2/z2m/tzb/tele/SENSOR = {"ZbInfo":{"0xFE68":{"Device":"0xFE68","Name":"tuya/od1","IEEEAddr":"0x60A423FFFE69C41F","ModelId":"TS0121","Manufacturer":"_TZ3000_rdtixbnu","Endpoints":[1],"Config":["O01"],"Power":1,"Reachable":true,"LastSeen":231,"LastSeenEpoch":1613687565,"LinkQuality":63}}}
23:36:36.269 MQT: avdln2/z2m/tzb/tele/SENSOR = {"ZbInfo":{"0xFD66":{"Device":"0xFD66","Name":"SosButton","IEEEAddr":"0x847127FFFE054378","ModelId":"TS0215A","Manufacturer":"_TZ3000_4fsgukof","Endpoints":[1],"Config":["A01.8"],"ZoneType":271,"ZoneStatus":0,"Reachable":true,"BatteryPercentage":100,"LastSeen":706,"LastSeenEpoch":1613687090,"LinkQuality":126}}}
23:36:36.274 MQT: avdln2/z2m/tzb/stat/RESULT = {"ZbInfo":"Done"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Register the TUYA SOS button and press the red button

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The button press should be received and result in an outgoing MQTT msg

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

nagyrobi, 2021-03-03:
Similar problem with an IKEA SYMFONISK turnknob. Can pair but don't get events.

ascillato, 2021-03-03:
> Similar problem with an IKEA SYMFONISK turnknob. Can pair but don't get events.

Please, in order to help you, open a new issue and post the logs. Thanks.

nagyrobi, 2021-03-05:
No need to open a new issue: the IKEA SYMFONISK turnknob behaving similar was faulty.

Most probably because I also tried to pair it with ZHA and upgrade firmware. Now it was either loaded with wrong firmware or remained stuck in firmware update mode.

Ikea replaced it. Lesson learnt: dont 'use IKEA SYMFONISK turnknob in ZHA as it breaks it.

sim0njo, 2021-03-30:
There is no solution, still waiting for a fix or a request for additional info to fix the issue. 

Pls keep this issue open

Thx, Jo

ascillato2, 2021-04-01:
Moving to Discussions as it is a Feature Request

closed 2021-04-01 by ascillato2