arendst/Tasmota #11040: TasmotaClient problem (Follow-up)

cn-superman, 2021-02-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

hi, still me.
Last time, I tested the tasmotaClient module. According to your tips, I tested different baud rates and other different operations and found some rules.
When I use other IO of 8266 MCU (for example: GPIO4 GPIO5) to communicate with 328p MCU, the garbled problem is improved.
And when I use the hard serial port (GPIO1 GPIO3) of 8266 MCU, the garbled problem reappeared.
English is not my native language and I am not a professional developer either. I hope to give you some inspiration.
Thank you for developing such a great software.

- [ ] Device used (e.g., Sonoff Basic): __ESP826601___
- [ ] Tasmota binary firmware version number used: __9.2.0.7___
  - [ ] Pre-compiled
  - [yes ] Self-compiled
- [ ] Flashing tools used: _OTA____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
      {"NAME":"TasDuino","GPIO":[32,4160,4224,4192,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
```lua
  Configuration output here:

```

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

04:43:47.546 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:43:47.553 MQT: NB/RESULT = {"ClientSend":"Done"}
04:43:47.594 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 100 ms"}
04:43:47.795 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}
04:43:47.893 SRC: MQTT
04:43:47.896 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:43:47.903 MQT: NB/RESULT = {"ClientSend":"Done"}
04:43:47.995 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 100 ms"}
04:43:48.066 SRC: MQTT
04:43:48.069 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:43:48.076 MQT: NB/RESULT = {"ClientSend":"Done"}
04:43:48.095 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}
04:43:48.195 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 100 ms"}
04:43:48.266 SRC: MQTT
04:43:48.269 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:43:48.276 MQT: NB/RESULT = {"ClientSend":"Done"}
04:43:48.295 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}
04:43:48.395 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 100 ms"}
04:43:48.495 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}
04:43:49.644 SRC: MQTT
04:43:49.647 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:43:49.654 MQT: NB/RESULT = {"ClientSend":"Done"}
04:43:49.695 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 100 ms"}
04:43:49.821 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup }
04:43:50.043 SRC: MQTT
04:43:50.046 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,100"
04:43:50.053 MQT: NB/RESULT = {"ClientSend":"Done"}
04:43:50.095 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 100 ms"}
04:43:50.295 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}
04:43:54.284 WIF: 检查连接...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

cn-superman, 2021-02-20:
Maybe I didn’t express it clearly,
After testing, I found that when using GPIO1 and GPIO3 of 8266 MCU, the problem of garbled characters appeared again.
Thanks~

meingraham, 2021-02-20:
1/3 are hardware serial whereas using 4/5 uses software serial. The Baud rate for hardware & software serial is set separately. See `Baudrate` and `SBaudrate` commands.

cn-superman, 2021-02-20:
Thank you,
It may be because I set up the hard serial port (1/3),
I can only get the speed of the hard serial port in the console, it is normal
When I use Sbaudrate, it shows unknown commands.

The following is the web log content:

00:00:00.050 CFG: 从 flash 载入: FB, 数量: 17
00:00:00.055 QPC: Count 1
00:00:00.057 CFG: CR 339/699, Busy 0
00:00:00.060 SRC: Restart
00:00:00.061 项目: tasmota Tasmota 版本 9.3.0.1(tasmota)-2_7_4_9(2021-02-20T09:29:57)
00:00:00.212 WIF: 检查连接...
00:00:00.213 WIF: 尝试连接...
00:00:00.549 WIF: 连接到 AP1 wxsoft Channel 13 BSSId 48:0E:EC:75:4C:FA 模式: 11n 名称: NB-7668...
00:00:01.753 WIF: 检查连接...
00:00:01.754 WIF: 已连接
00:00:02.003 HTP: Web 服务器地址: NB-7668 IP 地址: 192.168.100.163
00:00:02.697 RTC: UTC 2021-02-20T02:04:22, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
03:04:23.002 MQT: 尝试连接...
03:04:23.037 MQT: 已连接
03:04:23.039 MQT: NB/LWT = Online (已保留)
03:04:23.041 MQT: NB/cmnd/POWER = 
03:04:23.042 MQT: 订阅 NB/cmnd/#
03:04:23.043 MQT: 退订 homeassistant/status
03:04:23.046 MQT: NB/INFO1 = {"Module":"TasDuino","Version":"9.3.0.1(tasmota)","FallbackTopic":"cmnd/DVES_C7FDF4_fb/","GroupTopic":"NB/cmnd/"}
03:04:23.048 MQT: NB/INFO2 = {"WebServerMode":"Admin","Hostname":"NB-7668","IPAddress":"192.168.100.163"}
03:04:23.050 MQT: NB/INFO3 = {"RestartReason":"Software/System restart"}
03:04:26.315 HTP: 主菜单
03:04:26.670 QPC: Reset
03:04:27.398 MQT: NB/STATE = {"Time":"2021-02-20T03:04:27","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":49,"MqttCount":1,"Wifi":{"AP":1,"SSId":"wxsoft","BSSId":"48:0E:EC:75:4C:FA","Channel":13,"RSSI":100,"Signal":-28,"LinkCount":1,"Downtime":"0T00:00:03"}}
03:04:27.450 HTP: 控制台
03:04:28.398 APP: 启动次数 8
03:04:28.682 CFG: 保存到 flash: FA, 数量: 18, 大小: 4096
03:04:29.390 SNS: Hardware Serial
03:04:29.398 TCL: Enabled
03:04:31.387 TCL: Version 20191129
03:04:32.403 MQT: tasmota/discovery/A4CF12C7FDF4/config = {"ip":"192.168.100.163","dn":"Tasmota","fn":["Tasmota",null,null,null,null,null,null,null],"hn":"NB-7668","mac":"A4CF12C7FDF4","md":"TasDuino","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.0.1","t":"NB","ft":"NB","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (已保留)
03:04:32.408 MQT: tasmota/discovery/A4CF12C7FDF4/sensors = {"sn":{"Time":"2021-02-20T03:04:32"},"ver":1} (已保留)

03:04:39.296 SRC: MQTT
03:04:39.298 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,200"
03:04:39.305 MQT: NB/RESULT = {"ClientSend":"Done"}
03:04:39.390 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 200 ms"}
03:04:39.590 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}

03:05:16.193 SRC: MQTT
03:05:16.196 CMD: 组: 0, 索引: 1, 命令: "CLIENTSEND", 数据: "slp,200"
03:05:16.204 MQT: NB/RESULT = {"ClientSend":"Done"}
03:05:16.294 MQT: NB/RESULT = {"TasmotaClient":"System is sleeping to 200 ms"}
03:05:16.494 MQT: NB/RESULT = {"TasmotaClient":"System is wakeup"}

03:06:28.717 CMD: baudrate
03:06:28.723 CMD: 组: 0, 索引: 1, 命令: "BAUDRATE", 数据: ""
03:06:28.729 MQT: NB/RESULT = {"Baudrate":57600}
03:06:36.810 CMD: sbaudrate
03:06:36.813 SRC: WebConsole from 192.168.100.100
03:06:36.816 CMD: 组: 0, 索引: 1, 命令: "SBAUDRATE", 数据: ""
03:06:36.823 MQT: NB/RESULT = {"Command":"Unknown"}

cn-superman, 2021-02-20:
It seems that without using hardware serial ports (GPIO1 GPIO3), TasmotaClient seems to work well, but I prefer to use it so that I don't have to give up using 8266-01 MCU (8 pins).

ascillato2, 2021-03-10:
Moving to discussions as it is a hardware related issue and not a Tasmota software issue.

closed 2021-03-10 by ascillato2