arendst/Tasmota #11041: WS2812 Scheme to Solid Color Problem

Beormund, 2021-02-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When using schemes for a WS2812 addressable LED strip, going from a solid color, e.g. blue to a scheme, e.g. rainbow scheme 11 and back to the same solid color doesnt work.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): D1 Mini v.3
- [ ] Tasmota binary firmware version number used: 9.2.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
 
{"NAME":"Wemos D1 Mini V3","GPIO":[0,0,1376,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Wemos D1 Mini V3"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1376":"WS2812"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Cinema Screen","FriendlyName":["Cinema Screen"],"Topic":"wifi2mqtt/Cinema-Screen/Tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T19:22:28","StartupUTC":"2021-02-18T13:27:38","Sleep":50,"CfgHolder":4617,"BootCount":17,"BCResetTime":"2021-02-13T17:53:15","SaveCount":139,"SaveAddress":"F8000"}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS2 = {"StatusFWR":{"Version":"9.2.0(tasmota)","BuildDateTime":"2020-12-21T15:03:40","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"469/699"}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VodafoneConnect47505423","Recovery"],"TelePeriod":900,"Resolution":"55C180C0","SetOption":["00008009","2805C8000107060000005A00000000000000","00000000","00006000","00000000"]}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS4 = {"StatusMEM":{"ProgramSize":586,"Free":416,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS5 = {"StatusNET":{"Hostname":"wifi2mqtt/Cinema-Screen/Tasmota","IPAddress":"192.168.1.20","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"F4:CF:A2:ED:BA:CB","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.37","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EDBACB","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-19T08:50:06","Local":"2021-02-19T09:50:06","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:50","Sunset":"18:17"}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-19T09:50:06"}}
09:50:06 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-19T09:50:06","Uptime":"0T19:22:28","UptimeSec":69748,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Dimmer":60,"Color":"000099","HSBColor":"240,100,60","Channel":[0,0,60],"Scheme":0,"Width":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"VodafoneConnect47505423","BSSId":"84:47:65:DF:9A:8F","Channel":1,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

09:59:26 CMD: color 3
09:59:26 SRC: WebConsole from 192.168.1.41
09:59:26 CMD: Group 0, Index 1, Command "COLOR", Data "3"
09:59:26 SRC: Light
09:59:26 MQT: wifi2mqtt/Cinema-Screen/Tasmota/stat/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0000FF","HSBColor":"240,100,100","Channel":[0,0,100]}
09:59:28 CFG: Saved to flash at F6, Count 141, Bytes 4096
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
09:59:28 UDP: Packet (119)
[... 127 lines cut ...]
color 3
scheme 11
color 3
color 9

Or 

color 3
scheme 11
scheme 0
color 3
color 9

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

After any scheme 7...12 returning to previous solid color doesnt work. Strip freezes until a different solid color is chosen.  I would expect to return to previous solid color.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

iChUdY, 2021-02-26:
I have to confirm this bug. However, this bug did not occur on the older Tasmota 6.1

ascillato2, 2021-02-28:
Thanks for reporting. Fixed by Theo. Please, try latest development build. Thanks.

closed 2021-02-28 by ascillato2

iChUdY, 2021-03-02:
Works fine 👍🏻

Beormund, 2021-03-02:
> Thanks for reporting. Fixed by Theo. Please, try latest development build. Thanks.

Works perfectly. Thank you!