arendst/Tasmota #11047: Wiegand 26/34 missed some key press if they are press at normal speed

Fires04, 2021-02-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Weigand 26/34 - if the keypress is fast(normal speed) there are some keypress skipped. If they are slow for example just 1 keypress/second there are ok. Also, tag scan is ok.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff SV
- [x ] Tasmota binary firmware version number used: 9.2.0.7
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:58:32.183 CMD: Backlog Template; Module; GPIO 255
13:58:32.186 SRC: WebConsole from 192.168.50.190
13:58:32.190 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
13:58:32.258 SRC: Backlog
13:58:32.261 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
13:58:32.267 RSL: stat/tasmota_E35B75/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
13:58:32.272 WIE: ScanForTag().
13:58:32.476 SRC: Backlog
13:58:32.479 CMD: Group 0, Index 1, Command "MODULE", Data ""
13:58:32.484 RSL: stat/tasmota_E35B75/RESULT = {"Module":{"3":"Sonoff SV"}}
13:58:32.725 SRC: Backlog
13:58:32.728 CMD: Group 0, Index 1, Command "GPIO", Data "255"
13:58:32.738 RSL: stat/tasmota_E35B75/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6464":"Wiegand D1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"6432":"Wiegand D0"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:59:13.126 RSL: stat/tasmota_E35B75/STATUS = {"Status":{"Module":3,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E35B75","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:59:13.133 RSL: stat/tasmota_E35B75/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:19","StartupUTC":"2021-02-19T12:41:54","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2021-02-19T13:27:48","SaveCount":15,"SaveAddress":"F5000"}}
13:59:13.139 RSL: stat/tasmota_E35B75/STATUS2 = {"StatusFWR":{"Version":"9.2.0.7(tasmota)","BuildDateTime":"2021-02-19T13:24:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"352/699"}}
13:59:13.144 RSL: stat/tasmota_E35B75/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["steinovi_iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
13:59:13.155 RSL: stat/tasmota_E35B75/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","04001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,82"}}
13:59:13.164 RSL: stat/tasmota_E35B75/STATUS5 = {"StatusNET":{"Hostname":"tasmota_E35B75-7029","IPAddress":"192.168.200.11","Gateway":"192.168.200.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"98:F4:AB:E3:5B:75","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:59:13.170 RSL: stat/tasmota_E35B75/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E35B75","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:59:13.178 RSL: stat/tasmota_E35B75/STATUS7 = {"StatusTIM":{"UTC":"2021-02-19T12:59:13","Local":"2021-02-19T13:59:13","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:50","Sunset":"18:17"}}
13:59:13.185 RSL: stat/tasmota_E35B75/STATUS10 = {"StatusSNS":{"Time":"2021-02-19T13:59:13"}}
13:59:13.191 RSL: stat/tasmota_E35B75/STATUS11 = {"StatusSTS":{"Time":"2021-02-19T13:59:13","Uptime":"0T00:17:19","UptimeSec":1039,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"steinovi_iot","BSSId":"CE:2D:E0:31:C6:22","Channel":11,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Take Sonoff SV connect the RFID reader with Weigand 26/34 interface and try press keypad in "normal" speed for example 2-3 press per second. 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It should print all pressed characters to console.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I try keypad in 26 also in 34 mode.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-02-19:
The Tasmota driver is written and supports the Wiegand RFID reader https://github.com/arendst/Tasmota/commit/2dbd1ef97387fa20242a9f58ace80592ed8abca3
Funny that the Keypad is working a bit

Fires04, 2021-02-19:
Yes, the keypad uses the Wiegand protocol to send read data also to read keypress. (keypress is the same as reading RFID token just with 4-bit id..  If are keypress slow there is no problem. So there should be just some timing issue.

ascillato2, 2021-02-19:
> FUNC_EVERY_100_MSECOND now it looks like it working mutch better. with tags also with keys. Sometimes it miss some keys if it really fast.

Can you provide a PR please? Thanks.

Fires04, 2021-02-19:
Sorry, but what is PR ? :)

ascillato, 2021-02-19:
Pull Request. As you have already find and fix your issue, you can go to the file here in github, click on the pencil, do the editings and then click on commit. Then, after that, github will show you to perform a pull request. The pull request is a request to review and accept your code changes and merge them into Tasmota.

Fires04, 2021-02-19:
Ok, but I do not think this is THE right solution, this should check someone who created that driver and exactly know what that change could affect.

nagyrobi, 2021-03-03:
> the keypad uses the Wiegand protocol to send read data also to read keypress.

What model of keypad is this?

iot-sle, 2021-03-06:
As written in my source code comments,I had some timing issue with readers I bought for tests in case of reading too fast. I guess it was related to count of log outputs. But I don't have a reader with a key pad. We can change the code to 100ms reading and use a define for delay_cycle in case of slower devices ( I did the test on a ESP-01S and ESP32. )  Any other idea?

iot-sle, 2021-03-08:
So first look: the problem of fast typing... the buffer will contain another bitcount value than 4,8,24,... something in between or for a 4 bit key pad  2 key strokes = 8 bit, which is not a valid 8 bit key :-). To shorten the time looking at this buffer makes the situation "a little bit" better, but as already mentioned above it is not a final solution. The same situation may happen less times, but it will happen. I need to think about it and hopefully find a solution during next days.

Fires04, 2021-03-08:
https://www.aliexpress.com/item/4001029318011.html?spm=a2g0s.9042311.0.0.27424c4dmzNIec

PR #11284 (2021-03-11): fix #11047Wiegand 26/34 missed some key press if they are press at normal speed 
## Description:

**Related issue (if applicable):** fixes #11047 Wiegand 26/34 missed some key press if they are press at normal speed 
## Checklist:
  - [x ] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x ] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ x] The code change is tested and works with core ESP32 V.1.0.5
  - [ x] I acce [... cut]
commit 07f641c0 (2021-03-11): fix #11047 Wiegand 26/34 missed some key press
closed 2021-03-11 by arendst