arendst/Tasmota #11053: Smart Meter first tele full of zeros

nagyrobi, 2021-02-19:
### PROBLEM DESCRIPTION
After reboot, the first tele message on MQTT is sent with all the values at zero. Supposedly because Smart Meter interface didn't receive values from the meter yet. P1 serial meters usually send values at every 10 seconds. Tasmota boots and sends the first tele faster, and the zeros disturb the accuracy on the database.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 9.2.0.7(tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:13:37.079 MQT: stat/bal/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:13:37.307 MQT: stat/bal/RESULT = {"Module":{"18":"Generic"}}
00:13:37.512 MQT: stat/bal/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
00:14:05.064 CMD: STATUS 0
00:14:05.072 MQT: stat/bal/STATUS = {"Status":{"Module":18,"DeviceName":"Bal (Normál)","FriendlyName":["Bal (Normál)"],"Topic":"bal","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:14:05.081 MQT: stat/bal/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:00","StartupUTC":"2021-02-19T23:09:05","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-02-19T23:31:47","SaveCount":11,"SaveAddress":"3FB000"}}
00:14:05.087 MQT: stat/bal/STATUS2 = {"StatusFWR":{"Version":"9.2.0.7(tasmota)","BuildDateTime":"2021-02-19T22:50:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"461/699"}}
00:14:05.094 MQT: stat/bal/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AA",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","000000C0","00006000","00000000"]}}
00:14:05.105 MQT: stat/bal/STATUS4 = {"StatusMEM":{"ProgramSize":654,"Free":1392,"Heap":11,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F989","00024004","00201000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,41,45,50","Sensors":"1,2,3,4,5,6,53"}}
00:14:05.115 MQT: stat/bal/STATUS5 = {"StatusNET":{"Hostname":"bal-3514","IPAddress":"192.168.1.48","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"8C:AA:B5:7C:CD:BA","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:14:05.121 MQT: stat/bal/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7CCDBA","MqttUser":"hamqt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:14:05.129 MQT: stat/bal/STATUS7 = {"StatusTIM":{"UTC":"2021-02-19T23:14:05","Local":"2021-02-20T00:14:05","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:39","Sunset":"17:13"}}
00:14:05.138 MQT: stat/bal/STATUS10 = {"StatusSNS":{"Time":"2021-02-20T00:14:05","N":{"enrg_imp":41.157,"enrg_exp":3.762,"telj_imp":1.241,"telj_exp":0.000,"fesz_l1":228.4,"fesz_l2":230.2,"fesz_l3":232.8,"freq":49.94,"enrg_imp_med":0.568,"enrg_imp_med":19.463,"havi":"210201000000W"}}}
00:14:05.146 MQT: stat/bal/STATUS11 = {"StatusSTS":{"Time":"2021-02-20T00:14:05","Uptime":"0T00:05:00","UptimeSec":300,"Heap":11,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"AA","BSSId":"F0:9F:C2:A4:3B:1C","Channel":3,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.023 UFS: FlashFS mounted with 1984 kB free
00:00:00.088 CFG: Loaded from File, Count 12
00:00:00.093 QPC: Count 1
00:00:00.095 CFG: CR 461/699, Busy 0
00:00:00.098 SRC: Restart
00:00:00.099 Project tasmota Bal (Normál) Version 9.2.0.7(tasmota)-2_7_4_9(2021-02-19T22:50:42)
00:00:00.100 UFILESYSTEM OK!
00:00:00.115 Script: nv=0, tv=0, vns=0, ram=16
00:00:00.118 SNS: Hardware Serial
00:00:00.220 WIF: Checking connection...
00:00:00.221 WIF: Attempting connection...
00:00:01.125 WIF: Network (re)scan started...
00:00:01.126 WIF: Checking connection...
00:00:01.127 WIF: Attempting connection...
00:00:02.103 WIF: Checking connection...
00:00:02.104 WIF: Attempting connection...
00:00:03.126 WIF: Checking connection...
00:00:03.127 WIF: Attempting connection...
00:00:04.129 WIF: Network 0, AP1, SSId AA, Channel 3, BSSId F0:9F:C2:A4:3B:1C, RSSI -61, Encryption 1
00:00:04.478 WIF: Connecting to AP1 AA Channel 3 BSSId F0:9F:C2:A4:3B:1C in mode 11n as bal-3514...
00:00:04.479 WIF: Checking connection...
00:00:04.480 WIF: Attempting connection...
00:00:05.802 WIF: Checking connection...
00:00:05.803 WIF: Connected
00:00:06.007 HTP: Web server active on bal-3514 with IP address 192.168.1.48
00:00:06.503 QPC: Reset
00:00:06.524 RTC: UTC 2021-02-19T23:15:11, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
00:15:11.058 HTP: Main Menu
00:15:12.038 MQT: Attempting connection...
00:15:12.240 MQT: Connected
00:15:12.244 MQT: tele/bal/LWT = Online (retained)
00:15:12.249 MQT: cmnd/bal/POWER = 
00:15:12.251 MQT: Subscribe to cmnd/bal/#
00:15:12.255 MQT: Subscribe to cmnd/tasmotas/#
00:15:12.259 MQT: Subscribe to cmnd/DVES_7CCDBA_fb/#
00:15:12.262 MQT: Unsubscribe from homeassistant/status
00:15:12.269 MQT: tele/bal/INFO1 = {"Module":"Generic","Version":"9.2.0.7(tasmota)","FallbackTopic":"cmnd/DVES_7CCDBA_fb/","GroupTopic":"cmnd/tasmotas/"}
00:15:12.274 MQT: tele/bal/INFO2 = {"WebServerMode":"Admin","Hostname":"bal-3514","IPAddress":"192.168.1.48"}
00:15:12.279 MQT: tele/bal/INFO3 = {"RestartReason":"Software/System restart"}
00:15:12.286 TFS: File not found
00:15:13.150 HTP: Console
00:15:13.412 APP: Boot Count 6
00:15:13.701 CFG: Saved to flash at 3FB, Count 13, Bytes 4096
00:15:16.436 MQT: tele/bal/STATE = {"Time":"2021-02-20T00:15:16","Uptime":"0T00:00:13","UptimeSec":13,"Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"AA","BSSId":"F0:9F:C2:A4:3B:1C","Channel":3,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:15:16.446 MQT: tele/bal/SENSOR = {"Time":"2021-02-20T00:15:16","N":{"enrg_imp":0.000,"enrg_exp":0.000,"telj_imp":0.000,"telj_exp":0.000,"fesz_l1":0.0,"fesz_l2":0.0,"fesz_l3":0.0,"freq":0.00,"enrg_imp_med":0.000,"enrg_imp_med":0.000,"havi":"210201000000W"}}
```

### TO REPRODUCE
Reboot the module.

### EXPECTED BEHAVIOUR
Do not send a tele until values are filled. Maybe it would be enough to just have a delay command in Smart Meter to wait for the next data cycle, and only after that start teleposting.

### SCREENSHOTS
None.

### ADDITIONAL CONTEXT
The script:
>D  
>B  
->sensor53 r
>M 1  
```
+1,3,o,16,115200,N,1
1,1-0:1.8.0(@1,Energia import,kWh,enrg_imp,3
1,1-0:2.8.1(@1,Energia export,kWh,enrg_exp,3
1,0:1.7.0(@1,Teljesítmény import,kW,telj_imp,3
1,0:2.7.0(@1,Teljesítmény export,kW,telj_exp,3
1,32.7.0(@1,L1 Feszültség,V,fesz_l1,1
1,52.7.0(@1,L2 Feszültség,V,fesz_l2,1
1,72.7.0(@1,L3 Feszültség,V,fesz_l3,1
1,0:14.7.0(@1,Frekvencia,Hz,freq,2
1,0:3.8.0(@1,Energia meddő import,kvarh,enrg_imp_med,3
1,0:4.8.0(@1,Energia meddő export,kvarh,enrg_imp_med,3
1,0:98.1.0(@#),Havi adat,,havi,0
```

**(Please, remember to close the issue when the problem has been addressed)**

nagyrobi, 2021-02-19:
I tried various workarounds like setting `"tele_period": 0` from config.sys at boot, and add delay to the script:
```

>BS
delay(10000)
tper=30
```
But no success. A tele is sent immediately after boot with lots of zeros.

arendst, 2021-02-20:
The smart meter code has to be written in that it not responds to the teleperiod request if it has not yet received valid data.

Over to the smartmeter code writer @gemu2015  ....

closed 2021-02-20 by arendst

nagyrobi, 2021-02-20:
Thanks for this.
It's not perfect yet, it only works once. So after I freshly upgraded the firmware to a build containing this patch, at the first boot it indeed skipped the first tele. But after subsequent restarts from the web gui, a null mqtt is sent after boot...
:(

gemu2015, 2021-02-20:
meanwhile I rewired my sml test system and if no serial data comes in no MQTT is generated.
i made several restarts of sml and rebooting and could not observe a null mqtt.
please test in console with sensor53 r and check if a null mqtt is generated.
the valid flags are always reset on restart and once a serial datagram is received the valid flag is set and only then mqtt is sent.

i also did implement a global sml mqtt disable where you can disable mqtt from scripts.
the script variable is smlj
you may read and write it.
a 1 enables mqtt from script, a zero disables it.

i also implemented the bracket skipping. here is a new version. now that i can check with my simulator i found a bug in the previous version.

please test

[Archiv.zip](https://github.com/arendst/Tasmota/files/6016131/Archiv.zip)

nagyrobi, 2021-02-20:
Unfortunately still not good:
```
0:00:00.008 UFS: FlashFS mounted with 1984 kB free
00:00:00.056 CFG: Loaded from File, Count 6
00:00:00.062 QPC: Count 1
00:00:00.068 Project tasmota Bal (Normál) Version 9.3.0.1(tasmota)-2_7_4_9(2021-02-20T20:41:15)
00:00:00.069 UFILESYSTEM OK!
00:00:00.075 Script: nv=0, tv=0, vns=0, ram=16
00:00:00.085 SNS: Hardware Serial
00:00:04.475 WIF: Connecting to AP1 ZACUT Channel 6 BSSId F0:9F:C2:A4:1A:28 in mode 11n as elmu_bal-3514...
00:00:05.801 WIF: Connected
00:00:06.004 HTP: Web server active on elmu_bal-3514 with IP address XXXXXXXX
00:00:06.503 QPC: Reset
20:45:16.013 MQT: Attempting connection...
20:45:16.069 MQT: Connected
20:45:16.072 MQT: tele/elmu_bal/LWT = Online (retained)
20:45:16.075 MQT: cmnd/elmu_bal/POWER = 
20:45:16.083 MQT: tele/elmu_bal/INFO1 = {"Module":"Generic","Version":"9.3.0.1(tasmota)","FallbackTopic":"cmnd/DVES_7CCDBA_fb/","GroupTopic":"cmnd/tasmotas/"}
20:45:16.086 MQT: tele/elmu_bal/INFO2 = {"WebServerMode":"Admin","Hostname":"bal-3514","IPAddress":"XXXXXXXX"}
20:45:16.089 MQT: tele/elmu_bal/INFO3 = {"RestartReason":"Software/System restart"}
20:45:20.487 MQT: tele/elmu_bal/STATE = {"Time":"2021-02-20T20:45:20","Uptime":"0T00:00:13","UptimeSec":13,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"AAA","BSSId":"F0:9F:C2:A4:1A:28","Channel":6,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:07"}}
20:45:20.494 MQT: tele/elmu_bal/SENSOR = {"Time":"2021-02-20T20:45:20","N":{"enrg_imp":0.000,"enrg_exp":0.000,"telj_imp":0.000,"telj_exp":0.000,"fesz_l1":0.0,"fesz_l2":0.0,"fesz_l3":0.0,"freq":0.00,"enrg_imp_med":0.000,"enrg_imp_med":0.000}}
```
Maybe it's not too good to rely solely on serial dataflow detection. I assume in my case that if reboot happens in the middle of the data being pushed, only part of the information will be decoded, rest of it will remain zero. For instance I noticed several times that the last values were populated, the first ones were still at zero.

nagyrobi, 2021-02-20:
Can you suggest how to use `smlj` in my script above?
I imagine something like set it to 0 at boot.
Then wait 2 subsequent serial data pushes (because the first one may be incomplete) and then set `smlj` to 1.

nagyrobi, 2021-02-20:
Solution: https://github.com/arendst/Tasmota/discussions/11052#discussioncomment-389123
(for those directed here by a search engine)

PR #11056 (2021-02-20): send mqtt only on valid data
## Description:

**Related issue (if applicable):** fixes #11053

mqqt only send if at least one signal received 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CON [... cut]