arendst/Tasmota #11057: Not able to use SSD1306 on version 9.3.0(display)-2_7_4_9(2021-02-19T11:09:31)

gerardovf, 2021-02-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After updating to 9.3.0 release version or develop 9.3.0.1 the command "DisplayModel 2" reverts to command "DisplayModel 15" after several seconds of being executed and the subsequent reset.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _Wemos D1____
- [x ] Tasmota binary firmware version number used: _9.3.0____
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: __OTA___
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:28:22.740 MQT: gvf/cega/stat/salonBlinds/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:28:22.965 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Module":{"18":"Generic"}}
19:28:23.217 MQT: gvf/cega/stat/salonBlinds/RESULT = {"GPIO0":{"225":"Relay2"},"GPIO1":{"0":"None"},"GPIO2":{"224":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"227":"Relay4"},"GPIO13":{"164":"Switch5"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"226":"Relay3"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
19:29:04.851 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":302,"Free":209,"Rules":"on tele-si7021#Temperature do DisplayText [s1l1c1t][s1l1c8T] [s1l3c1]Temp.: %value%C endon       on tele-si7021#Humidity do DisplayText [s1l4c1]Hum.: %value%% endon       on tele-si7021#DewPoint do DisplayText [s1l5c1]Dew Pt.: %value%C endon       on system#boot do DisplayText [s1l1c1t][s1l1c8T] endon"}
19:29:05.080 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":97,"Free":414,"Rules":"ON switch5#state DO Backlog POWER5 ON; ruletimer1 30 ENDON   ON Rules#Timer=1 DO POWER5 OFF ENDON"}
19:29:05.279 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:29:38.658 CMD: Status 0
19:29:38.668 MQT: gvf/cega/stat/salonBlinds/STATUS = {"Status":{"Module":18,"DeviceName":"TasmotaSalon","FriendlyName":["Tasmota1","Tasmota2","Tasmota3","Tasmota4","Tasmota5"],"Topic":"salonBlinds","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":6,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
19:29:38.674 MQT: gvf/cega/stat/salonBlinds/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-display.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:08:50","StartupUTC":"2021-02-20T18:20:48","Sleep":50,"CfgHolder":4617,"BootCount":512,"BCResetTime":"2020-08-12T16:27:22","SaveCount":32500,"SaveAddress":"F9000"}}
19:29:38.679 MQT: gvf/cega/stat/salonBlinds/STATUS2 = {"StatusFWR":{"Version":"9.3.0.1(display)","BuildDateTime":"2021-02-20T16:47:33","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"480/699"}}
19:29:38.684 MQT: gvf/cega/stat/salonBlinds/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["homeGVF","indebuurt1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C209","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
19:29:38.693 MQT: gvf/cega/stat/salonBlinds/STATUS4 = {"StatusMEM":{"ProgramSize":579,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":16384,"FlashChipId":"1840EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","401F8000","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
19:29:38.699 MQT: gvf/cega/stat/salonBlinds/STATUS5 = {"StatusNET":{"Hostname":"salonBlinds-4361","IPAddress":"192.168.1.193","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.155","Mac":"5C:CF:7F:57:71:09","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:29:38.706 MQT: gvf/cega/stat/salonBlinds/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.151","MqttPort":1883,"MqttClientMask":"TasmotaSalon","MqttClient":"TasmotaSalon","MqttUser":"petingvf","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
19:29:38.714 MQT: gvf/cega/stat/salonBlinds/STATUS7 = {"StatusTIM":{"UTC":"2021-02-20T18:29:38","Local":"2021-02-20T19:29:38","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"08:06","Sunset":"18:57"}}
19:29:38.721 MQT: gvf/cega/stat/salonBlinds/STATUS10 = {"StatusSNS":{"Time":"2021-02-20T19:29:38","Switch5":"ON","SI7021":{"Temperature":18.8,"Humidity":46.3,"DewPoint":7.0},"TempUnit":"C"}}
19:29:38.728 MQT: gvf/cega/stat/salonBlinds/STATUS11 = {"StatusSTS":{"Time":"2021-02-20T19:29:38","Uptime":"0T00:08:50","UptimeSec":530,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"homeGVF","BSSId":"90:5C:44:B1:CE:2E","Channel":8,"RSSI":100,"Signal":-27,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
19:30:31.143 CMD: weblog 4
19:30:31.150 MQT: gvf/cega/stat/salonBlinds/RESULT = {"WebLog":4}
19:30:31.591 CFG: Saved to flash at F8, Count 32501, Bytes 4096
19:30:31.799 WIF: Checking connection...
19:30:42.471 DHT: Timeout waiting for start signal low pulse
19:30:51.805 WIF: Checking connection...
19:30:57.507 MQT: gvf/cega/tele/salonBlinds/STATE = {"Time":"2021-02-20T19:30:57","Uptime":"0T00:10:09","UptimeSec":609,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"homeGVF","BSSId":"90:5C:44:B1:CE:2E","Channel":8,"RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:30:57.520 MQT: gvf/cega/tele/salonBlinds/SENSOR = {"Time":"2021-02-20T19:30:57","Switch5":"ON","SI7021":{"Temperature":18.8,"Humidity":46.3,"DewPoint":7.0},"TempUnit":"C"} (retained)
19:30:57.532 RUL: TELE-SI7021#TEMPERATURE performs "DisplayText [s1l1c1t][s1l1c8T] [s1l3c1]Temp.: 18.8C"
19:30:57.534 SRC: Rule
19:30:57.537 CMD: Group 0, Index 1, Command "DISPLAYTEXT", Data "[s1l1c1t][s1l1c8T] [s1l3c1]Temp.: 18.8C"
19:30:57.541 LOG: TM1637: sString=[s1l1c1t][s1l1c8T] [s1l3c1]Temp.: 18.8C
19:30:57.543 LOG: TM1637: position=0
19:30:57.545 LOG: TM1637: length=0
19:30:57.547 Clearing digit 4
19:30:57.554 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1t][s1l1c8T] [s1l3c1]Temp.: 18.8C"}
19:30:57.567 RUL: TELE-SI7021#HUMIDITY performs "DisplayText [s1l4c1]Hum.: 46.3%"
19:30:57.570 SRC: Rule
[... 87 lines cut ...]
19:32:51.431 LOG: TM1637: position=0
19:32:51.434 LOG: TM1637: length=0
19:32:51.436 Clearing digit 4
19:32:51.444 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1t][s1l1c8T] [s1l3c1]Temp.: 18.8C"}
19:32:51.456 RUL: TELE-SI7021#HUMIDITY performs "DisplayText [s1l4c1]Hum.: 46.3%"
19:32:51.459 SRC: Rule
19:32:51.463 CMD: Group 0, Index 1, Command "DISPLAYTEXT", Data "[s1l4c1]Hum.: 46.3%"
19:32:51.467 LOG: TM1637: sString=[s1l4c1]Hum.: 46.3%
19:32:51.470 LOG: TM1637: position=0
19:32:51.473 LOG: TM1637: length=0
19:32:51.475 Clearing digit 4
19:32:51.483 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l4c1]Hum.: 46.3%"}
19:32:51.499 RUL: TELE-SI7021#DEWPOINT performs "DisplayText [s1l5c1]Dew Pt.: 7.0C"
19:32:51.501 SRC: Rule
[... cut ...] w Pt.: 7.0C"
19:32:51.509 LOG: TM1637: sString=[s1l5c1]Dew Pt.: 7.0C
19:32:51.511 LOG: TM1637: position=0
19:32:51.514 LOG: TM1637: length=0
19:32:51.517 Clearing digit 4
19:32:51.525 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l5c1]Dew Pt.: 7.0C"}
19:32:51.532 DHT: Timeout waiting for start signal low pulse
19:32:52.386 APP: Boot Count 513
19:32:52.592 CFG: Saved to flash at F5, Count 32504, Bytes 4096

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-02-20:
Me too. Looks like the new TM1637 driver is "taking over", thereby disabling other displays. Thin testing.... Looks like the driver is not even trying to detect if that display type is there or not. Not sure if there are easier ways than compiling without USE_DISPLAY_TM1637.

closed 2021-02-21 by ascillato2

sfromis, 2021-02-21:
Does not work for me on `Version 9.3.0.1(display)-1_0_5_RC6(2021-02-21T11:22:10)` 

The SSD1306 does show up as 0x3c on I2CScan, DisplayAddress correct at 60, and the on/off toggle does appear in the web front page. This also appears with no display on the I2C bus which I did not expect. At boot time no messages about either SSD1306 or TM1637. Both with and without the SSD1306 connected, the DisplayText causes:

```
13:59:29.849 CMD: i2cscan
13:59:29.910 MQT: t183es32/stat/I2CSCAN = {"I2CScan":"Device(s) found at 0x3c"}
13:59:34.799 CMD: displayaddress
13:59:34.816 MQT: t183es32/stat/DISPLAYADDRESS = {"DisplayAddress1":60}
13:59:40.094 CMD: displaytext [f2y20x25]34567
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ff4f2  PS      : 0x00060a30  A0      : 0x800e586e  A1      : 0x3ffb19e0  
A2      : 0x00000000  A3      : 0x3ffb1a1f  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ffb192e  A7      : 0x3ffc32ec  A8      : 0x4fa57b4f  A9      : 0x3ffb18c0  
A10     : 0x4fa57b4f  A11     : 0x4fa57b4f  A12     : 0x3ffc82f0  A13     : 0x00000038  
A14     : 0x3ffb18fe  A15     : 0x3ffc32ec  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000a  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

ELF file SHA256: 0000000000000000

Backtrace: 0x400ff4f2:0x3ffb19e0 0x400e586b:0x3ffb1a10 0x400e60af:0x3ffb1a40 0x400e63c9:0x3ffb1b10 0x400e647f:0x3ffb1b30 0x400e64ba:0x3ffb1b50 0x400f3981:0x3ffb1b70 0x400ecb0a:0x3ffb1b90 0x400f3c2c:0x3ffb1bc0 0x400e2b9e:0x3ffb1be0 0x400ed049:0x3ffb1c00 0x400ed229:0x3ffb1d00 0x400ef865:0x3ffb1d60 0x400ef979:0x3ffb1d80 0x400f0bc9:0x3ffb1dd0 0x4018e511:0x3ffb1e00 0x400fb576:0x3ffb1e20 0x400fb865:0x3ffb1e40 0x400fb5be:0x3ffb1e70 0x400fb73d:0x3ffb1ec0 0x400d9d97:0x3ffb1f10 0x400ece4d:0x3ffb1f30 0x400e2b9e:0x3ffb1f50 0x400f5821:0x3ffb1f70 0x400f58cb:0x3ffb1f90 0x40114b35:0x3ffb1fb0 0x4008acee:0x3ffb1fd0

Rebooting...
```

arendst, 2021-02-21:
But what reports command ``display``?

sfromis, 2021-02-21:
`{"Display":{"Model":15,"Width":128,"Height":64,"Mode":0,"Dimmer":1,"Size":4,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}`

arendst, 2021-02-21:
The display driver only supports one type of display.

In the case both a SSD1306 AND the TM1637 are connected and configured (which a user should not do) it selects the TM1637 as the current driver. I suspect in that case the command ``displaytext [f2y20x25]34567`` is not supported on that driver considering the amount of effort the programmer has put in adding an abundance of commands to control the display.

I don't like the TM1637 implementation as it deviates from the common way display command are addressed.

I will add an extra test for allowance to the TM1637 driver.

sfromis, 2021-02-21:
I'm having no TM1637 connected or attempted any config (I own none of that)

sfromis, 2021-02-21:
Fails as bad on ESP8266....
```
Exception (28):
epc1=0x4022d958 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000a depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff900 end: 3fffffc0 offset: 0190
3ffffa90:  00000000 3ffffdbc 3ffffb40 4024a1bc  
.....
```

arendst, 2021-02-21:
Pls retry with latest change.

sfromis, 2021-02-21:
Now works with SSD1306-only on both ESP8266 and ESP32, but still crashed until I manually selected DisplayModel 2. Did not try that last time.

gerardovf, 2021-02-21:
It works now. Thanks a lot for the quick response (it's Sunday!). Before updating I disabled both Rule1 and Rule2 just in case...