arendst/Tasmota #11059: ESP32 BLE Mija Tasmota 9.3.0 setoption115 1

Rushmed, 2021-02-20:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Hello,

what happend to setoption115 1 on Tasmota 9.3.0-sensors.
It was working fine on 9.2.0-sensors to read BLE Sensors on ESP32 using built-in Bluetooth.
But on 9.3.0-sensors setoption115 1 is answered with:
22:39:56 MQT: stat/tasmota/RESULT = {"SetOption115":"ON"}
but no sensor Mi sensor is detected.
Downgrading to 9.2.0-sensors solves the issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____ESP32 Wroom
- [x] Tasmota binary firmware version number used: _____9.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: _____OTA and esphome Flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
`22:51:07 CMD: Backlog Template; Module; GPIO 255
22:51:07 MQT: stat/tasmota/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
22:51:08 MQT: stat/tasmota/RESULT = {"Module":{"1":"ESP32-DevKit"}}
22:51:08 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"1152":"RFrecv"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"1120":"RFSend"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"160":"Switch1"},"GPIO28":{"0":"None"}}}
22:51:08 MQT: stat/tasmota/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}}`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Downgrading to 9.2.0-sensors

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Recognizing Mija sensors.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bshm, 2021-02-22:
I experienced the exactly the same issue. 
git bisect revealed that it is introduced by commit 55ba994956e39ea4fbba703cd3286b942255e733

arendst, 2021-02-22:
> git bisect revealed that it is introduced by commit 55ba994

In that case use the correct binary ``tasmota-bluetooth.bin`` as that is the only version with the defines enabled due to instabilities.

ascillato2, 2021-02-22:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-02-22 by ascillato2

Rushmed, 2021-02-25:
Will there be further development to the BLE funtionality to bring it back to a precompiled binary which supports RF tasceiver functionality?