arendst/Tasmota #11074: Setting the sensitivity for Aqara Vibration Sensor doesn't work

Mific78, 2021-02-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Bridge
- [ ] Tasmota binary firmware version number used: 9.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:
try to set the sensitivity with the command : ZbSend {"device":"0xD995","Endpoint":1,"Cluster":0,"Manuf":"0x115F","Write":{"0000/FF0D%20":"0x01"}}

The console output is :
18:37:58.761 CMD: ZbSend {"device":"0xD995","Endpoint":1,"Cluster":0,"Manuf":"0x115F","Write":{"0000/FF0D%20":"0x01"}}
18:37:58.777 MQT: stat/tasmota_04BD92/RESULT = {"ZbSend":"Done"}
18:38:06.468 ZIG: {"ZbEZSPReceived":"80004295D9"}
18:38:06.472 MQT: tele/tasmota_04BD92/RESULT = {"ZbRouteError":{"ShortAddr":"0xD995","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
18:38:15.711 ZIG: {"ZbEZSPReceived":"80004295D9"}
18:38:15.715 MQT: tele/tasmota_04BD92/RESULT = {"ZbRouteError":{"ShortAddr":"0xD995","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
18:38:24.955 ZIG: {"ZbEZSPReceived":"80004295D9"}
18:38:24.959 MQT: tele/tasmota_04BD92/RESULT = {"ZbRouteError":{"ShortAddr":"0xD995","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen.
Change of the sensitivity

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-02-21:
You need to wake up the device first. Sometimes these settings only work right after pairing, that would be an alternative

Mific78, 2021-02-21:
I try to wake up the device first but it doesn't work.

I try to pairing again (ZbForget 0xD995 + ZbPermitJoin 1 + pairing + ZbSend {"device":"0xD995","Endpoint":1,"Cluster":0,"Manuf":"0x115F","Write":{"0000/FF0D%20":"0x01"}}) but it still doesn't work.

s-hadinger, 2021-02-21:
Are you sure the address is the same after repairing?

Mific78, 2021-02-21:
Same result with the long address (after a wake up).

Mific78, 2021-02-21:
I try to wake AFTER the ZbSend command and it works !

s-hadinger, 2021-02-21:
Good to know. 

Can you copy the weblog 3 of the read command?

Mific78, 2021-02-21:
22:44:50.874 MQT: stat/tasmota_04BD92/RESULT = {"ZbSend":"Done"}
22:44:50.901 ZIG: {"ZbEZSPReceived":"340000D4"}
22:44:53.067 ZIG: {"ZbEZSPReceived":"450000040100000101000100001AFFE295D9FFFF4D1C5F11120A050042126C756D692E766962726174696F6E2E61713101FF422E0121670C0328200421A8130521450006240000010000082108030A21000098212800992100009A250A00F1FF9904"}
22:44:53.069 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0000","srcaddr":"0xD995","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":149,"securityuse":0,"seqnumber":26,"fc":"0x1C","frametype":0,"direction":1,"disableresp":1,"manuf":"0x115F","transact":18,"cmdid":"0x0A","payload":"050042126C756D692E766962726174696F6E2E61713101FF422E0121670C0328200421A8130521450006240000010000082108030A21000098212800992100009A250A00F1FF9904"}}
22:44:53.072 ZIG: ZbZCLRawReceived: {"0xD995":{"0000/0005":"lumi.vibration.aq1","0000/FF01":"0121670C0328200421A8130521450006240000010000082108030A21000098212800992100009A250A00F1FF9904","Endpoint":1,"LinkQuality":149}}
22:44:53.080 MQT: tele/tasmota_04BD92/SENSOR = {"ZbReceived":{"0xD995":{"Device":"0xD995","ModelId":"lumi.vibration.aq1","BatteryVoltage":3.17,"BatteryPercentage":100,"Xiaomi_98":40,"Xiaomi_99":0,"Xiaomi_9A":"0x0499FFF1000A","Endpoint":1,"LinkQuality":149}}}
22:44:53.927 ZIG: {"ZbEZSPReceived":"3F000095D904010000010140010000D4010000"}
22:44:53.934 ZIG: {"ZbEZSPReceived":"450000040100000101400100001BFFE395D9FFFF0A0C5F110A010DFF002000"}
22:44:53.936 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0000","srcaddr":"0xD995","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":152,"securityuse":0,"seqnumber":27,"fc":"0x0C","frametype":0,"direction":1,"disableresp":0,"manuf":"0x115F","transact":10,"cmdid":"0x01","payload":"0DFF002000"}}
22:44:53.938 ZIG: ZbZCLRawReceived: {"0xD995":{"0000/FF0D":0,"Endpoint":1,"LinkQuality":152}}
22:44:53.944 MQT: tele/tasmota_04BD92/SENSOR = {"ZbReceived":{"0xD995":{"Device":"0xD995","0000/FF0D":0,"Endpoint":1,"LinkQuality":152}}}

s-hadinger, 2021-02-22:
Thanks. I confirm from the raw Hex that the device responds 00 for this attribute

Jason2866, 2021-02-24:
Please close since your question has been answered.
Thx!

closed 2021-02-24 by ascillato2