arendst/Tasmota #11078: TuyaMCU: Unable to control dimmer  with consecutive MQTT commands

sirtow, 2021-02-22:
### PROBLEM DESCRIPTION
I have Zemismart blind controller flashed with 9.3.0. Every MQTT command , to set the Dimmer, after last successful command is ignored. In order to make it work, i need to send the old value and the new value again. For example, if Dimmer was set to "2"  sending Dimmer=95 will cause `Send dim skipped value 95 for dpid 2`  message in console. The only way to set Dimmer to new value is to send again Dimmer=2 MQTT command and then Dimmer=95 again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ]X Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Zemismart smart blinds controller (TYWES3)  - https://www.aliexpress.com/item/4000983297439.html
- [X ] Tasmota binary firmware version number used: 9.3.0
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X ] Flashing tools used: Tazmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:37:41.979 CMD: Backlog Template; Module; GPIO 255
00:37:41.982 SRC: WebConsole from 172.16.220.2
00:37:41.985 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"

```
- [X ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
00:39:16.957 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
00:39:17.008 SRC: Backlog
00:39:17.011 CMD: Group 0, Index 1, Command "RULE", Data ""
00:39:17.018 MQT: tasmota/blinds1/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":242,"Free":269,"Rules":"ON TuyaReceived#DpType2Id2 DO Publish tasmota/%topic%/stat/POSITION %value% ENDON ON TuyaReceived#10#DpIdData DO Publish tasmota/%topic%/stat/PROBLEM %value% ENDON ON TuyaReceived#DpType1Id5 DO Publish tasmota/%topic%/stat/MOTOR %value% ENDON"}
00:39:17.222 SRC: Backlog
00:39:17.226 CMD: Group 0, Index 2, Command "RULE", Data ""
00:39:17.236 MQT: tasmota/blinds1/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
00:39:17.437 SRC: Backlog
00:39:17.441 CMD: Group 0, Index 3, Command "RULE", Data ""
00:39:17.451 MQT: tasmota/blinds1/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ X] Provide the output of this command: `Status 0`:
```lua
 00:40:13.746 CMD: Group 0, Index 1, Command "STATUS", Data "0"
00:40:13.753 MQT: tasmota/blinds1/stat/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"blinds1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:40:13.762 MQT: tasmota/blinds1/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:59:00","StartupUTC":"2021-02-21T20:41:13","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-02-21T19:19:03","SaveCount":942,"SaveAddress":"F6000"}}
00:40:13.770 MQT: tasmota/blinds1/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.0(tasmota)","BuildDateTime":"2021-02-19T11:07:30","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
00:40:13.778 MQT: tasmota/blinds1/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["milky-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","08010080","00006000","00000000"]}}
00:40:13.791 MQT: tasmota/blinds1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:40:13.800 MQT: tasmota/blinds1/stat/STATUS5 = {"StatusNET":{"Hostname":"blinds1","IPAddress":"192.168.1.12","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"172.16.4.204","Mac":"24:A1:60:27:5A:F0","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:40:13.807 MQT: tasmota/blinds1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.milky","MqttPort":1883,"MqttClientMask":"blinds1","MqttClient":"blinds1","MqttUser":"blinds1","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:40:13.815 MQT: tasmota/blinds1/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-21T23:40:13","Local":"2021-02-22T00:40:13","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:45","Sunset":"18:21"}}
00:40:13.823 MQT: tasmota/blinds1/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-22T00:40:13"}}
00:40:13.832 MQT: tasmota/blinds1/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-22T00:40:13","Uptime":"0T02:59:00","UptimeSec":10740,"Heap":25,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":2,"POWER":"ON","Dimmer":95,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"milky-iot","BSSId":"16:80:63:06:E6:10","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":2,"Downtime":"0T00:00:05"}}}
00:40:20.143 TYA: Send "55aa00000000ff"

```
- [ X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:50:36.173 TYA: Heartbeat
00:50:36.605 MQT: tasmota/blinds1/tele/RESULT = {"TuyaReceived":{"Data":"55AA0307000802020004000000021B","Cmnd":7,"CmndData":"0202000400000002","DpType2Id2":2,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000002"}}}
00:50:36.619 RUL: TUYARECEIVED#DPTYPE2ID2 performs "Publish tasmota/blinds1/stat/POSITION 2"
00:50:36.622 SRC: Rule
00:50:36.625 CMD: Group 0, Index 1, Command "PUBLISH", Data "tasmota/blinds1/stat/POSITION 2"
00:50:36.632 MQT: tasmota/blinds1/stat/POSITION = 2
00:50:36.639 TYA: fnId=21 is set for dpId=2
00:50:36.642 TYA: RX value 2 from dpId 2 
00:50:37.105 MQT: tasmota/blinds1/tele/RESULT = {"TuyaReceived":{"Data":"55AA0307000802020004000000021B","Cmnd":7,"CmndData":"0202000400000002","DpType2Id2":2,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000002"}}}
00:50:37.117 RUL: TUYARECEIVED#DPTYPE2ID2 performs "Publish tasmota/blinds1/stat/POSITION 2"
00:50:37.120 SRC: Rule
00:50:37.123 CMD: Group 0, Index 1, Command "PUBLISH", Data "tasmota/blinds1/stat/POSITION 2"
[... 754 lines cut ...]
00:54:20.632 TYA: fnId=21 is set for dpId=2
00:54:20.635 TYA: RX value 95 from dpId 2 
00:54:20.637 SRC: Switch
00:54:20.640 CMD: Group 0, Index 3, Command "DIMMER", Data "95"
00:54:20.647 MQT: tasmota/blinds1/stat/RESULT = {"POWER":"ON","Dimmer":95}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Send Dimmer=96 to MQTT (it must work and move the blinds motor)
Send Dimmer=1 to MQTT ( "Send dim skipped" log message appears). Motor does not turn `00:54:09.601 SRC: MQTT` entry in the log
Send Dimmer=96 to MQTT.  Motor does not turn `00:54:13.458 SRC: MQTT` entry in the log
Send DImmer=1 to MQTT . Motor start spinning `00:54:19.874 SRC: MQTT` entry log

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Sending Dimmer updates over MQTT should be passed to the TuyaMCU and not be skipped

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jschwalbe, 2021-02-23:
Just curious if it behaves like my issue, if you send the same dimmer command a second time, does it work?

sirtow, 2021-02-23:
@jschwalbe Sending the same (new value) results in the skip message. I tried that for couple of minutes.

jaymemaurice, 2021-02-24:
@sirtow - can you test my pull request -  https://github.com/arendst/Tasmota/pull/11121
I believe it fixes this issue.

PR #11121 (2021-02-25): Change TuyaMCU dimmer timeout
…xpired

## Description:
Change TuyaMCU dimmer timeout timer 

**Related issue (if applicable):** fixes #11078, #10985

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [ ] The code change is tested and works on Tasmota core ESP32 V.1.0.5-rc6
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/developme [... cut]
closed 2021-02-25 by ascillato2