arendst/Tasmota #11081: Tasmota Zigbee should use a random network key.

chriscn, 2021-02-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Under the [Zigbee Documentation](https://tasmota.github.io/docs/Zigbee/); the encryption key is a default key `0x0D0C0A08060402000F0D0B0907050301`; this allows a remote third party to easily gain access to any Zigbee Tasmota network that hasn't been updated.

Tasmota should generate a random key on installation.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Bridge
- [x] Tasmota binary firmware version number used: 9.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ x Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I currently haven't started flashing but by reading the documentation this is the clear wrong behaviour.

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-02-22:
Sorry but documentation is outdated. Tasmota will choose a random key if no key is present. You can also force a random key by setting a zero key. Use ZbConfig

chriscn, 2021-02-22:
So @s-hadinger if I just flash the bridge normally and don't change anything it should use a random key?

s-hadinger, 2021-02-22:
Correct

closed 2021-02-22 by ascillato2