arendst/Tasmota #11093: Smart Meter not able to read ISKRA eHZ-EDL+ Typ MT681-D4A51-K0p

christi-ko, 2021-02-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am not able to get the data of my new Smart Meter Device interpreted. Data seems to be not according to standard or maybe corrupt. I tried to use 2 different Sensors, one with BTW40 Diode and PullUp Resistor and an other one based on TEKT (refer to https://forum.iobroker.net/topic/25224/volksz%C3%A4hler-ttl-schreib-lesekopf-f%C3%BCr-digitale-stromz%C3%A4hler/72?lang=de), both shown to work with my 2nd Smart Meter Device. 
According to Internet research there are contradicting messages which protocol to used (i tried all of them, o, s, m, M) without any success showing sense full data (even not partly corrupted etc.). 

By using the expected correct setting the following results is obtained:

CMD: sensor53 d1
20:57:04.443 MQT: stat/tasmota_678555/RESULT = {"Time":"2021-02-22T20:57:04","SML":{"CMD":"dump: 1"}}
20:57:08.152 nK8'nEX
20:57:08.203 : {
20:57:08.252 : {
20:57:10.506 : {
20:57:10.607 : {
20:57:14.516 : {
20:57:14.616 : {
20:57:16.847 : {
20:57:16.900 : {
20:57:20.798 : {
20:57:20.898 : {
20:57:23.059 : {
20:57:23.158 : {

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R2
- [x] Tasmota binary firmware version number used: ed7c06a ; 9.3.0.1(tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Atom
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:39:37.751 CMD: Backlog Template; Module; GPIO 255
21:39:37.827 MQT: stat/tasmota_678555/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:39:38.039 MQT: stat/tasmota_678555/RESULT = {"Module":{"18":"Generic"}}
21:39:38.292 MQT: stat/tasmota_678555/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
21:39:47.383 MQT: tele/tasmota_678555/STATE = {"Time":"2021-02-22T21:39:47","Uptime":"0T00:40:10","UptimeSec":2410,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"StaSi Zentralnetz","BSSId":"44:4E:6D:2C:5F:2A","Channel":6,"RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}

```
- [] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:41:21.252 MQT: stat/tasmota_678555/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_678555","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:41:21.260 MQT: stat/tasmota_678555/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:41:44","StartupUTC":"2021-02-22T19:59:37","Sleep":50,"CfgHolder":4617,"BootCount":20,"BCResetTime":"2021-02-22T18:24:51","SaveCount":47,"SaveAddress":"F5000"}}
21:41:21.269 MQT: stat/tasmota_678555/STATUS2 = {"StatusFWR":{"Version":"9.3.0.1(tasmota)","BuildDateTime":"2021.02.22 18:27:08","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"382/699"}}
21:41:21.275 MQT: stat/tasmota_678555/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["StaSi Zentralnetz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
21:41:21.286 MQT: stat/tasmota_678555/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000407","87DAC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,53"}}
21:41:21.294 MQT: stat/tasmota_678555/STATUS5 = {"StatusNET":{"Hostname":"tasmota_678555-1365","IPAddress":"192.168.178.62","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"84:0D:8E:67:85:55","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:41:21.300 MQT: stat/tasmota_678555/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.34","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_678555","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
21:41:21.309 MQT: stat/tasmota_678555/STATUS7 = {"StatusTIM":{"UTC":"2021-02-22T20:41:21","Local":"2021-02-22T21:41:21","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:45","Sunset":"18:21"}}
21:41:21.316 MQT: stat/tasmota_678555/STATUS10 = {"StatusSNS":{"Time":"2021-02-22T21:41:21"}}}
21:41:21.324 MQT: stat/tasmota_678555/STATUS11 = {"StatusSTS":{"Time":"2021-02-22T21:41:21","Uptime":"0T00:41:44","UptimeSec":2504,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"StaSi Zentralnetz","BSSId":"44:4E:6D:2C:5F:2A","Channel":6,"RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
22:02:01.706 CMD: sensor53 d1
[... 96 lines cut ...]
21:00:09.192 : 77 07 01 00 01 08 00 ff 65 00 00 01 80 01 62 1e 52 03 69 00 00 00 00 00 00 00 bd 01 
21:00:09.242 : 77 07 01 00 01 08 01 ff 01 01 62 1e 52 03 69 00 00 00 00 00 00 00 41 01 
21:00:09.292 : 77 07 01 00 01 08 02 ff 01 01 62 1e 52 03 69 00 00 00 00 00 00 00 7b 01 
21:00:09.381 : 77 07 81 81 c7 82 05 ff 01 01 01 01 83 02 87 9d 0c b2 9e 15 cb 9e 03 c9 75 25 16 34 14 f1 a0 e4 61 99 82 de 1f 79 9e 5a cc 02 a3 9d 4b ac e1 20 f4 72 27 26 6e 4b 86 b8 27 99 ee 91 c5 d8 01 01 01 63 67 a7 00 76 05 00 17 7a f8 62 00 62 00 72 63 02 01 71 01 63 ac 4b 00 1b 1b 1b 1b 1a 00 86 
21:00:11.194 : f5 1b 1b 1b 1b 01 01 01 01 76 05 00 17 7a f9 62 00 62 00 72 63 01 01 76 01 
21:00:11.233 : 01 05 00 07 d3 a9 0b 09 01 49 53 4b 00 04 8a fb 02 01 01 63 4a e6 00 76 05 00 17 7a fa 62 00 62 00 72 63 07 01 
21:00:11.263 : 77 01 0b 09 01 49 53 4b 00 04 8a fb 02 07 01 00 62 0a ff ff 72 62 01 65 00 0e e5 d0 76 
21:00:11.304 : 77 07 81 81 c7 82 03 ff 01 01 01 01 04 49 53 4b 01 
21:00:11.355 : 77 07 01 00 00 00 09 ff 01 01 01 01 0b 09 01 49 53 4b 00 04 8a fb 02 01 
21:00:11.406 : 77 07 01 00 01 08 00 ff 65 00 00 01 82 01 62 1e 52 03 69 00 00 00 00 00 00 00 bd 01 
21:00:11.455 : 77 07 01 00 01 08 01 ff 01 01 62 1e 52 03 69 00 00 00 00 00 00 00 41 01 
21:00:11.506 : 77 07 01 00 01 08 02 ff 01 01 62 1e 52 03 69 00 00 00 00 00 00 00 7b 01 
21:00:11.622 : 77 07 81 81 c7 82 05 ff 01 01 01 01 83 02 87 9d 0c b2 9e 15 cb 9e 03 c9 75 25 16 34 14 f1 a0 e4 61 99 82 de 1f 79 9e 5a cc 02 a3 9d 4b ac e1 20 f4 72 27 26 6e 4b 86 b8 27 99 ee 91 c5 d8 01 01 01 63 a2 9b 00 76 05 00 17 7a fb 62 00 62 00 72 63 02 01 71 01 63 1f b5 00 1b 1b 1b 1b 1a 00 0d 56 
21:00:15.197 : 1b 1b 1b 1b 01 01 01 01 76 05 00 17 7a fc 62 00 62 00 72 63 01 01 76 01 01 05 
21:00:15.233 : 00 07 d3 aa 0b 09 01 49 53 4b 00 04 8a fb 02 01 01 63 4b c6 00 76 05 00 17 7a fd 62 00 62 00 72 63 07 01 
21:00:15.263 : 77 01 0b 09 01 49 53 4b 00 04 8a fb 02 07 01 00 62 0a ff ff 72 62 01 65 00 0e e5 d4 76 
21:00:15.307 : 77 07 81 81 c7 82 03 ff 01 01 01 01 04 49 53 4b 01 
[... cut ...]
21:00:15.537 : 77 07 01 00 01 08 02 ff 01 01 62 1e 52 03 69 00 00 00 00 00 00 00 7b 01 
21:00:15.626 : 77 07 81 81 c7 82 05 ff 01 01 01 01 83 02 87 9d 0c b2 9e 15 cb 9e 03 c9 75 25 16 34 14 f1 a0 e4 61 99 82 de 1f 79 9e 5a cc 02 a3 9d 4b ac e1 20 f4 72 27 26 6e 4b 86 b8 27 99 ee 91 c5 d8 01 01 01 63 8e 34 00 76 05 00 17 7a fe 62 00 62 00 72 63 02 01 71 01 63 db be 00 1b 1b 1b 1b 1a 00 17 10 
21:00:17.443 : 1b 1b 1b 1b 01 01 01 01 76 05 00 17 7a ff 62 00 62 00 72 63 01 01 76 01 01 05 00 07 d3 

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-02-23:
this IS SML not OBIS!  ( sml starts with 77)
use an SML descriptor and it should work.

christi-ko, 2021-02-23:
Sorry, my fault. I was too fixed on OBIS from my previous meters.
I will close the topic

closed 2021-02-23 by christi-ko