arendst/Tasmota #11094: Support Zigbee TRV Moes H36530

avk999, 2021-02-22:
### PROBLEM DESCRIPTION
TRV Moes H36530 (Tuya-based) is successfully paired to ZigbeeBridge but the incoming commands are not deciphered.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _Sonoff ZigbeeBridgg___
- [x ] Tasmota binary firmware version number used: 9.3.0.1(zbbridge)
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: OTA
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
23:05:09.906 MQT: stat/zbbridge/RESULT = {"NAME":"Sonoff ZHABridge","GPIO":[320,5088,0,5120,323,322,0,0,0,0,0,0,32,0],"FLAG":0,"BASE":18}
23:05:10.113 MQT: stat/zbbridge/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
23:05:10.366 MQT: stat/zbbridge/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
23:06:46.741 MQT: stat/zbbridge/STATUS = {"Status":{"Module":75,"DeviceName":"zbbridge","FriendlyName":["Tasmota"],"Topic":"zbbridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
22:45:17.343 MQT: stat/zbbridge/RESULT = {"ZbProbe":"Done"}
22:45:17.516 MQT: tele/zbbridge/RESULT = {"ZbPing":{"Device":"0x79FC","IEEEAddr":"0x5C0272FFFEC57C51"}}
22:45:17.802 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":822,"Endpoint":1,"LinkQuality":0}}}
22:45:17.915 MQT: tele/zbbridge/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01"]}}
22:45:18.048 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","ModelId":"TS0601","Manufacturer":"_TZE200_zion52ef","Endpoint":1,"LinkQuality":0}}}
22:45:18.869 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":821,"Endpoint":1,"LinkQuality":0}}}
22:45:19.800 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":820,"Endpoint":1,"LinkQuality":3}}}
22:45:19.981 MQT: tele/zbbridge/RESULT = {"ZbState":{"Status":33,"Device":"0x79FC","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0051","DeviceVersion":1,"InClusters":["0x0000","0x000A","0x0004","0x0005","0xEF00"],"OutClusters":["0x0019"]}}
22:45:20.805 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":819,"Endpoint":1,"LinkQuality":3}}}
22:45:20.967 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00<24":"0000","Endpoint":1,"LinkQuality":0}}}
22:45:21.032 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00<24":"0000","Endpoint":1,"LinkQuality":0}}}
22:45:21.080 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00<24":"0000","Endpoint":1,"LinkQuality":0}}}
22:45:21.812 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":818,"Endpoint":1,"LinkQuality":0}}}
22:45:22.165 ZIG: Zigbee Devices Data saved in EEPROM (401 bytes)
22:45:23.660 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":817,"Endpoint":1,"LinkQuality":0}}}
22:45:33.282 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":816,"Endpoint":1,"LinkQuality":0}}}
22:45:33.398 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":815,"Endpoint":1,"LinkQuality":0}}}
22:45:33.524 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00/0276":814,"Endpoint":1,"LinkQuality":0}}}
22:45:33.647 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00<24":"0000","Endpoint":1,"LinkQuality":0}}}
22:45:38.889 MQT: tele/zbbridge/79FC/SENSOR = {"ZbReceived":{"0x79FC":{"Device":"0x79FC","EF00<24":"0000","Endpoint":1,"LinkQuality":0}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Pair the TRV with zbbridge

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Ability to send and receive HVAC commands, interpretation of reports.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-02-23:
Can you please attach here logs of undeciphered messages using `weblog 3`

ascillato2, 2021-02-23:
Moving to discussions as it is a feature request.

closed 2021-02-23 by ascillato2