arendst/Tasmota #11098: Shelly dimmer 2 / rotary encoder with push button

janramonstaal, 2021-02-23:
Hi
I have a Shelly dimmer 2 working with Tasmota v9.3.0. I also flashed the latest .bin for the stm32 v51.5
And I applied the Template.
Iam trying to get a "normal" setup. So I want every one to be able to use the dimmer with a rotary encoder with a push button for on/off.
I already have a Zigbee ECO-DIM.07 dimmer that works really great but Iam hoping to get the same functionality out of the Shelly because I need 4 dimmers extra....

1 -A normal dimmer with a push button for on/off only needs a Minimum setting.
For Tasmota this is done with : Dimmerrange 10,100 for example.
But this does not apply for the rotary encoder, the value will be lower than 10 if turned down.....

2- If you choose a "button" for GPIO0 than I think ONLY this pin should turn the dimmer on and off (manual)
there is a "Disable Auto Power On" "SetOption20" and this works but there should also be a "Disable Auto Power Off" I think.
So when you turn the dial at its lowest setting it should go to the lowest value that has been set with Dimmerrange

Thanks for this great project.
If you need more info or pics please let me know

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2 dimmer
- [X] Tasmota binary firmware version number used: 9.3.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-02-23:
Have you tried if `SetOption77 1` is working here too?

janramonstaal, 2021-02-23:
I just tried SetOption77 1  and this works !!
Very happy.
So when using a rotary encoder with push: apply  SetOption77 1 and "SetOption20 1
Thanks

closed 2021-02-23 by ascillato2