arendst/Tasmota #11102: Shelly 2.5 upgrade from 8.3.0 to 9.3.0 => migration path must include 9.2.0

emgecgn, 2021-02-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

When I tried to upgrade the Shelly 2.5 from 8.3.0 to 9.3.0 I have followed the migration path which is found in the Tasmota documentation:
=> v8.5.1 => v9.1 => v9.3
For each version I first uploaded the minimal.bin and then the respective full binary (DE version, downloaded to local drive).
When upgrading from 9.1 to 9.3.0 the upload failed with "...not compatible".
Even the minimal.bin did not work.

Solution: in my case (two devices) it only worked well when I upgraded from v9.1 to v9.2 and only then to v9.3.0
v9.2 was necessary as intermediate step to be successful.
It seems that at least for my Shellys the migration path needs to be including the v9.2 version!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 2.5
- [x] Tasmota binary firmware version number used: 9.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
14:14:05.109 MQT: Shelly-001/stat/RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,34,0,224,129,0,0,640,128,608,225,3456,4736],"FLAG":0,"BASE":18}
14:14:05.352 MQT: Shelly-001/stat/RESULT = {"Module":{"0":"Shelly 2.5"}}
14:14:05.563 MQT: Shelly-001/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"34":"Button3"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"129":"Button_in2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"128":"Button_in1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ"},"GPIO17":{"4736":"ADC Temp"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:15:13.955 MQT: Shelly-001/stat/RESULT = {"Rule1":"OFF","Once":"ON","StopOnError":"OFF","Length":124,"Free":387,"Rules":"on energy#current[2]>0.600 do backlog power1 0;power2 0; endon on energy#current[1]>0.600 do backlog power1 0;power2 0 endon"}
14:15:14.159 MQT: Shelly-001/stat/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":274,"Free":237,"Rules":"on button1#state=14 do Restart 1 endon on button1#state=10 do ShutterStopOpen endon on button1#state=11 do backlog ShutterStopOpen; publish Rollobutton/%topic%/btn1 \"ROOMUP\" endon on button1#state=12 do backlog ShutterStopOpen; publish Rollobutton/%topic%/btn1 \"ALLUP\" endon"}
14:15:14.367 MQT: Shelly-001/stat/RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Length":277,"Free":234,"Rules":"on button2#state=14 do Restart 1 endon on button2#state=10 do ShutterStopClose endon on button2#state=11 do backlog ShutterStopClose; publish Rollobutton/%topic%/btn2 \"ROOMDN\" endon on button2#state=12 do backlog ShutterStopClose; publish Rollobutton/%topic%/btn2 \"ALLDN\" endon"}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:15:57.707 MQT: Shelly-001/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Shelly-001","FriendlyName":["Shelly-001","Shelly-001b"],"Topic":"Shelly-001","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:15:57.714 MQT: Shelly-001/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T01:34:57","StartupUTC":"2021-02-23T11:41:00","Sleep":50,"CfgHolder":4617,"BootCount":115,"BCResetTime":"2020-05-14T13:26:25","SaveCount":4594,"SaveAddress":"F5000"}}
14:15:57.719 MQT: Shelly-001/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.0(tasmota)","BuildDateTime":"2021.02.19 11:15:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"425/699"}}
14:15:57.723 MQT: Shelly-001/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SyMa_SH-1","RaReSH-1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C00B","2805C8000100060000005A0A000000000000","40800081","00006000","00000000"]}}
14:15:57.731 MQT: Shelly-001/stat/STATUS4 = {"StatusMEM":{"ProgramSize":594,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
14:15:57.738 MQT: Shelly-001/stat/STATUS5 = {"StatusNET":{"Hostname":"Shelly-001","IPAddress":"192.168.178.35","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"BC:DD:C2:77:4F:77","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
14:15:57.743 MQT: Shelly-001/stat/STATUS6 = {"StatusMQT":{"MqttHost":"SHRaspi1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_774F77","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:15:57.752 MQT: Shelly-001/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-23T13:15:57","Local":"2021-02-23T14:15:57","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:27","Sunset":"18:01"}}
14:15:57.758 MQT: Shelly-001/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:15:57.768 MQT: Shelly-001/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-23T14:15:57","ANALOG":{"Temperature":41.6},"ENERGY":{"TotalStartTime":"2020-05-14T13:26:25","Total":0.230,"Yesterday":0.050,"Today":0.060,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"Direction":0,"Target":0},"TempUnit":"C"}}
14:15:57.777 MQT: Shelly-001/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-23T14:15:57","Uptime":"0T01:34:57","UptimeSec":5697,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SyMa_SH-1","BSSId":"98:9B:CB:1C:4D:59","Channel":1,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}}
14:15:57.783 MQT: Shelly-001/stat/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":80,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Tried with two devices, same behaviour.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

emgecgn, 2021-02-23:
Note:
I just tried using the .gz binaries for upgrading instead of the .bin files.
With the .gz the upgrade from 9.1 to 9.3 went smoothly. No issues there.
Only the .bin files seem to cause the problem in my case.

ascillato, 2021-02-23:
Thanks for expanding your explanations. Very appreciated. So, in that case your issue wasn't the upgrade path. Was just that the free space on the device wasn't enough for the incoming firmware.

emgecgn, 2021-02-23:
Well, I don't think so, because even the minimal.bin failed with "not compatible".

schrej-zz, 2021-02-23:
@emgecgn,
it's possible, that even the minimal.bin is too large for an update. I already had that issue.
I just upgraded from 9.1 to 9.3 via zipped files.
I suggest to close the issue.
Joerg

sfromis, 2021-02-23:
When discussed in Discord, the conclusion was that the only "issue" was the assumption that upgrading OTA via .bin instead of .bin.gz was a good plan. Also, with a previous compatibility check being removed, newer binries (without compression) are likely to fail the compatibility check in older binaries (only checked for uncompressed files).

emgecgn, 2021-02-23:
The intermediate step as well as using .gz files did solve the problem for me.
It is not really a serious matter, so I do agree that we can close that issue.
I just happened to realize that on Discord were a few other users reporting the same/similar problem.
So I suggest to adress that issue as a note, perhaps in the migration path section.
It could be highlighted there that either the use of the .gz files or the intermediate upgrade seems to be mandatory when upgrading from 9.1 to 9.3 on a number of devices.

closed 2021-02-23 by emgecgn