arendst/Tasmota #11103: sensor is overwritten by other values when retained which makes values disappear from retained message

jeroenst, 2021-02-23:
### PROBLEM DESCRIPTION

When using a sonoff temperature/humidity sensor (SNZB-02 - ZigBee, https://www.aliexpress.com/item/4001309350069.html) icm with the Tasmota Sonoff zbBridge I get separate messages for humidity and temperature on the same topic. When I set sensor retain, the temperature overwrites the humidity and visa versa making retain unusable because it is never complete.

Related discussion: https://github.com/arendst/Tasmota/discussions/10942

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff zbbridge
- [x] Tasmota binary firmware version number used: 9.3.0.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
14:13:30.386 CMD: Backlog Template; Module; GPIO 255
14:13:30.406 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"NAME":"Zigbee","GPIO":[320,3552,0,3584,5312,0,0,0,640,544,608,0,32,0],"FLAG":0,"BASE":75}
14:13:30.640 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
14:13:30.893 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:15:28.268 CMD: Backlog Rule1; Rule2; Rule3
14:15:28.288 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:15:28.522 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:15:28.773 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:15:51.943 CMD: status 0
14:15:51.948 MQT: home/TASMOTA_ZIGBEE/stat/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"TASMOTA_ZIGBEE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1,"InfoRetain":1,"StateRetain":1}}
14:15:51.954 MQT: home/TASMOTA_ZIGBEE/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/blob/development/tools/fw_SonoffZigbeeBridge_ezsp/ncp-uart-sw_6.7.8_115200.ota?raw=true","RestartReason":"Software/System restart","Uptime":"0T14:57:50","StartupUTC":"2021-02-22T22:18:01","Sleep":50,"CfgHolder":4617,"BootCount":208,"BCResetTime":"2021-02-21T12:36:05","SaveCount":291,"SaveAddress":"FB000"}}
14:15:51.958 MQT: home/TASMOTA_ZIGBEE/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.0.1(zbbridge)","BuildDateTime":"2021-02-22T19:37:25","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"483/699"}}
14:15:51.963 MQT: home/TASMOTA_ZIGBEE/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["J&M IOT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008269","2805C8000100060000005A00000000000000","00000080","00006080","00000380"]}}
14:15:51.970 MQT: home/TASMOTA_ZIGBEE/stat/STATUS4 = {"StatusMEM":{"ProgramSize":573,"Free":1216,"Heap":26,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
14:15:51.976 MQT: home/TASMOTA_ZIGBEE/stat/STATUS5 = {"StatusNET":{"Hostname":"TASMOTA_ZIGBEE","IPAddress":"172.30.31.54","Gateway":"172.30.30.254","Subnetmask":"255.255.254.0","DNSServer":"172.30.30.254","Mac":"84:CC:A8:9B:A2:74","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:15:51.980 MQT: home/TASMOTA_ZIGBEE/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BA274","MqttUser":"xxxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:15:51.987 MQT: home/TASMOTA_ZIGBEE/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-23T13:15:51","Local":"2021-02-23T14:15:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:43","Sunset":"18:23"}}
14:15:51.992 MQT: home/TASMOTA_ZIGBEE/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-23T14:15:51"}}
14:15:51.999 MQT: home/TASMOTA_ZIGBEE/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-23T14:15:51","Uptime":"0T14:57:50","UptimeSec":53870,"Vcc":3.438,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxxx","BSSId":"0E:EC:DA:8D:48:02","Channel":5,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:34:32.160 ZIG: {"ZbEZSPReceived2":"551C9001450000040105040101000100004388BE7CAAFFFF08189C0A0000212914"}
14:34:32.161 ZIG: ZbEZSPSentRaw 86
14:34:32.163 ZIG: {"ZbEZSPReceived":"450000040105040101000100004388BE7CAAFFFF08189C0A0000212914"}
14:34:32.164 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0405","srcaddr":"0xAA7C","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":55,"securityuse":0,"seqnumber":67,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":156,"cmdid":"0x0A","payload":"0000212914"}}
14:34:32.168 ZIG: ZbZCLRawReceived: {"0xAA7C":{"0405/0000":5161,"Endpoint":1,"LinkQuality":55}}
14:34:32.567 MQT: home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR = {"ZbReceived":{"0xAA7C":{"Device":"0xAA7C","Name":"Kantoor","Humidity":51.61,"Endpoint":1,"LinkQuality":55}}} (retained)
14:34:37.765 WIF: Checking connection...
14:34:42.195 ZIG: {"ZbEZSPReceived2":"651C9001450000040102040101000100004578BA7CAAFFFF08189D0A0000295B08"}
14:34:42.197 ZIG: ZbEZSPSentRaw 87
14:34:42.198 ZIG: {"ZbEZSPReceived":"450000040102040101000100004578BA7CAAFFFF08189D0A0000295B08"}
14:34:42.200 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0402","srcaddr":"0xAA7C","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":45,"securityuse":0,"seqnumber":69,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":157,"cmdid":"0x0A","payload":"0000295B08"}}
14:34:42.202 ZIG: ZbZCLRawReceived: {"0xAA7C":{"0402/0000":2139,"Endpoint":1,"LinkQuality":45}}
14:34:42.602 MQT: home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR = {"ZbReceived":{"0xAA7C":{"Device":"0xAA7C","Name":"Kantoor","Temperature":21.39,"Endpoint":1,"LinkQuality":45}}} (retained)

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Flash and configure a Sonoff zbBridge device
2. Set sensorretain true
3. See how temperature and humidity overwrite eachother

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Add the value that's already in memory for temperature or humidity when it's not present
OR Send SENSOR1 and SENSOR2 topics additional to sensor.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I will try to solve this issue myself but the question is what's the preferred option sensor1 and sensor2 or expanding the json with values available in memory.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-02-23:
Hi, how are you parsing the json string? Because in the json strings you posted, it has the valid keys. It is a valid JSON that is being parsed correctly in most common home automation softwares. It say temperature: value, then humidity: value.

jeroenst, 2021-02-23:
I use PHP to connect to my mqtt broker, subscribe to this topic, parse the received string in php with json_decode and display the values on a webpage. So when starting the webpage I only see temperature or humidity and after a while (can take up to an hour) I see the other value. Of course I can write a server that sends out the values in different retained topics but imo it would be better if the issue is solved at the zbbridge without addional software.

Also with nodered dashboard the same issue occurs. I have to save the value's to disk and load them at startup otherwise I have to wait until all values are available.

So the retain for temperature and humidity is not working in this usecase.

jeroenst, 2021-02-23:
My discussion https://github.com/arendst/Tasmota/discussions/11069 I allready foreseen this kind a issues with retaining values.
Sending temperature and humidity in different topics would solve this problem:
home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR/Temperature = "21.39" (retained)
home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR/Humidity = "51.61" (retained)

ascillato, 2021-02-23:
Ok. If you have an overwritten value is a problem in your custom software implementation. In your logs:

```lua
14:34:32.567 MQT: home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR = {"ZbReceived":{"0xAA7C":{"Device":"0xAA7C","Name":"Kantoor","Humidity":51.61,"Endpoint":1,"LinkQuality":55}}} (retained)

14:34:42.602 MQT: home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR = {"ZbReceived":{"0xAA7C":{"Device":"0xAA7C","Name":"Kantoor","Temperature":21.39,"Endpoint":1,"LinkQuality":45}}} (retained)
```

Your JSON messages are valid and correct. They are not overwritten. So, please, re check you own implementation. Common Home Automation softwares don't have your issue. 

Your request of splitting the JSON is not going to be added. Sorry. Splitting will increase Network usage, CPU usage, Energy usage and just for your own usecase. You should re check your PHP and javascript parsing software. Sorry.

ascillato2, 2021-02-23:
Closing as a discussion for this is already opened. Thanks.

closed 2021-02-23 by ascillato2

jeroenst, 2021-02-23:
Well it is still an issue (for me). And I still don't know which way the Tasmota team likes me to solve this.

jeroenst, 2021-02-23:
> e an overwritten value is a problem in your custom software implementation. In your logs:

The json fields temperature and humidity don't retain, they are removed in favour of each other.

This makes the retain option for temperature and humidity useless.

jeroenst, 2021-02-23:
> Ok. If you have an overwritten value is a problem in your custom software implementation. In your logs:
>
> ```lua
> 14:34:32.567 MQT: home/TASMOTA_ZIGBEE/AA7C/tele/SENSOR = [...]

The json fields temperature and humidity don't retain, they are removed in favour of each other.

This makes the retain option for temperature and humidity useless.

jeroenst, 2021-02-23:
Here you can see the effect after a deploy in nodered of the humidity value being removed in favor of humidity, only the temperature can be displayed because only temperature is retained:

[image]

ascillato, 2021-02-23:
For your use case, you don't need to modify Tasmota. Just use a simple rule with publish2 command (retained mqtt) for temperature to publish to the topic you want, and another rule for humidity. There are examples for that in the docs at rules.

barbudor, 2021-02-23:
If Tasmota sends 2 sensor messages (1 for T and 1 for HR) it is beause your Zigbee devices does the same.
When using other sensors such as I2C attached sensors like BME280, Tasmota send both T, HR, P, ... in the same SENSOR message.
But with Zigbee, Tasmota don't know how many different message (and different measurments) your Zigbee device can send so it can't properly group them into a single message. So it acts roughly as a Zigbee to MQTT gateway

I agree with @ascillato that the solution to your problem is some rules to either :
- Group both T & HR in a single custom SENSOR message
- Send separated custom SENSOR messages
In that case, as you can use Publish2, you don't need anymore the SensorRetain command

jeroenst, 2021-02-23:
Effect for nodered dashboard:
[image]

jeroenst, 2021-02-23:
> If Tasmota sends 2 sensor messages (1 for T and 1 for HR) it is beause your Zigbee devices does the same.
> When using other sensors such as I2C attached sensors like BME280, Tasmota send both T, HR, P, ... in the same SENSOR message.
> But with Zigbee, Tasmota don't know how many different message (and different measurments) your Zigbee device can send so it can't properly group them into a single message. So it acts roughly as a Zigbee to MQTT gateway
> 
> I agree with @ascillato that the solution to your problem is some rules to either :
> 
> * Group both T & HR in a single custom SENSOR message
> * Send separated custom SENSOR messages
>   In that case, as you can use Publish2, you don't need anymore the SensorRetain command
> 
> Rules and Scripting are precisely here to enable custom behavior without changing Tasmota

Ok, I will have a look into publish2, but still I think this has to be an effect to be aware of.

barbudor, 2021-02-23:
Personally I never use SensorRetain because I prefer to have no-value than an outdated value which age is unknown.
Seeing no value on a dashboard is a clue that something went wrong and that the backend may have restarted without notice and must be debugged.
After all, the updated value will be there in a few minutes after the restart.

jeroenst, 2021-02-23:
Not true, the updated value from a battery driven zigbee device can take up to an hour

ascillato, 2021-02-23:
> but still I think this has to be an effect to be aware of.

If you move to OpenHab or HomeAssistant you won't have this issue. Sorry but your issue is not in Tasmota. Is an issue of how your custom dashboard works. Sorry.

jeroenst, 2021-02-23:
Well, in my opinion the sensorretain option is working incorrectly this way.

The retain option in MQTT is used for clients that connect later than the values where send, so i this case they will miss temperature or humidity for up to an hour.

ascillato, 2021-02-23:
Agree, in that case, you should use a rule to customize the output (mqtt retained messages) as you prefer.

jeroenst, 2021-02-23:
Do I have to make a rule for every zigbee (10+) device in this case?

ascillato, 2021-02-23:
Custom softwares requires custom solutions. So yes. You have to do that for your use case.

jeroenst, 2021-02-23:
Ok, that's a big disappointment for me because I really think this is behaviour that breaks the retain option of MQTT.

barbudor, 2021-02-23:
I'm not using HA nor OpenHab, same as you do, I have but Node-Red as well as InfluxDB (and Grafana)
My suggestion would be to build a node-red flow to send your sensor data to InfluxDB which would enable you to not only get the last value back for the dashboard immediate value, but also to build nice historical graph. By default Node-Red graph loose history on each deploy or restart which makes them not very attractive without a database behind.
So instead of a complex set of rules on Tasmota to handle 10+ devices, a generic solution on the back-end would be much more valuable. 
InfluxDB would be a great enhancement to your platform

jeroenst, 2021-02-23:
@barbudor I use influxdb already with nodered, but are you proposing to save the values directly to influx and read them continuously to detect value changes (polling influx)?

barbudor, 2021-02-23:
Not continuously polling. 
I believe your problem with NR dashboard is when you deploys or restart NR.
So you could pull the last value on restart but keep feeding the updates directly.

jeroenst, 2021-02-23:
@barbudor  I can also make a function node that saves all topics to a file with context.set(msg.topic,msg.payload,"file");
and read and send them out on  deploy or restart to a link out or mqtt out.

But still IMO that's still a workaround, but so be it.

jeroenst, 2021-02-23:
For everybody who has the same problem here is the Rule I use now and will be expanded:
`Rule1
  on zbreceived#0xAA7C#humidity do publish2 home/zigbee/kantoor/humidity %value% endon
  on zbreceived#0xAA7C#temperature do publish2 home/zigbee/kantoor/humidity %value% endon
Rule1 1`

sfromis, 2021-02-23:
To condense the same rules for multiple similar sensors, you could use wildcards like 
`zbreceived#?#temperature`

jeroenst, 2021-02-23:
That's a good idea, but how do I distinguish the separate zigbee devices in this case?

0xAA7C is office in my case but another id can be a different location