arendst/Tasmota #11105: ledlink setting in template configuration of sonoff zbbridge (75) is not working

jeroenst, 2021-02-23:
### PROBLEM DESCRIPTION

I want to set the blue led on my sonoff zbbridge on to show that the device is powered on. At the it looks like the device has no power. When I change gpio13 from LedLink_i to LedLink this setting is saved but not working. The led stays off.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff zbbridge
- [x] Tasmota binary firmware version number used: 9.3.0.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
15:25:41.780 CMD: Backlog Template; Module; GPIO 255
15:25:41.798 RSL: home/TASMOTA_ZIGBEE/stat/RESULT = {"NAME":"Zigbee","GPIO":[320,3552,0,3584,5312,0,0,0,640,544,608,0,32,0],"FLAG":0,"BASE":75}
15:25:42.029 RSL: home/TASMOTA_ZIGBEE/stat/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
15:25:42.233 RSL: home/TASMOTA_ZIGBEE/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:15:28.268 CMD: Backlog Rule1; Rule2; Rule3
14:15:28.288 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:15:28.522 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
14:15:28.773 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:27:57.170 CMD: status 0
15:27:57.171 SRC: WebConsole from 172.30.31.8
15:27:57.172 CMD: Group 0, Index 1, Command "STATUS", Data "0"
15:27:57.176 MQT: home/TASMOTA_ZIGBEE/stat/STATUS = {"Status":{"Module":75,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"TASMOTA_ZIGBEE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1,"InfoRetain":1,"StateRetain":1,"LedLinkHeartbeat":0}}
15:27:57.179 MQT: home/TASMOTA_ZIGBEE/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/arendst/Tasmota/blob/development/tools/fw_SonoffZigbeeBridge_ezsp/ncp-uart-sw_6.7.8_115200.ota?raw=true","RestartReason":"Software/System restart","Uptime":"0T00:00:26","StartupUTC":"2021-02-23T14:27:31","Sleep":50,"CfgHolder":4617,"BootCount":212,"BCResetTime":"2021-02-21T12:36:05","SaveCount":306,"SaveAddress":"FC000"}}
15:27:57.183 MQT: home/TASMOTA_ZIGBEE/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.0.1(zbbridge)","BuildDateTime":"2021-02-22T19:37:25","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"483/699"}}
15:27:57.187 MQT: home/TASMOTA_ZIGBEE/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["J&M IOT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008269","2805C8000100060000005A00000000000000","00000080","00006080","00000180"]}}
15:27:57.194 MQT: home/TASMOTA_ZIGBEE/stat/STATUS4 = {"StatusMEM":{"ProgramSize":573,"Free":1216,"Heap":26,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
15:27:57.199 MQT: home/TASMOTA_ZIGBEE/stat/STATUS5 = {"StatusNET":{"Hostname":"TASMOTA_ZIGBEE","IPAddress":"172.30.31.54","Gateway":"172.30.30.254","Subnetmask":"255.255.254.0","DNSServer":"172.30.30.254","Mac":"84:CC:A8:9B:A2:74","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:27:57.203 MQT: home/TASMOTA_ZIGBEE/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BA274","MqttUser":"xxxxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:27:57.210 MQT: home/TASMOTA_ZIGBEE/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-02-23T14:27:57","Local":"2021-02-23T15:27:57","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:43","Sunset":"18:23"}}
15:27:57.215 MQT: home/TASMOTA_ZIGBEE/stat/STATUS10 = {"StatusSNS":{"Time":"2021-02-23T15:27:57"}}
15:27:57.222 MQT: home/TASMOTA_ZIGBEE/stat/STATUS11 = {"StatusSTS":{"Time":"2021-02-23T15:27:57","Uptime":"0T00:00:26","UptimeSec":26,"Vcc":3.438,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxx","BSSId":"0E:EC:DA:8D:48:02","Channel":5,"RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:31:53.517 HTP: Configuration
15:31:54.415 APP: Boot Count 217
15:31:54.576 CFG: Saved to flash at F9, Count 317, Bytes 4096
15:31:56.288 HTP: Configure Template
15:32:00.765 SRC: WebGui from 172.30.31.8
15:32:00.766 CMD: Group 0, Index 1, Command "TEMPLATE", Data "{"NAME":"Zigbee","GPIO":[320,3552,0,3584,5312,0,0,0,640,544,608,0,32,0],"FLAG":...
15:32:00.768 CFG: CR 483/699, Busy 0
15:32:00.772 MQT: home/TASMOTA_ZIGBEE/stat/RESULT = {"NAME":"Zigbee","GPIO":[320,3552,0,3584,5312,0,0,0,640,544,608,0,32,0],"FLAG":0,"BASE":75}
15:32:00.773 HTP: Restart
15:32:00.802 SRC: WebGui from 0.0.0.0
15:32:01.058 ZIG: Resetting EZSP device

### HERE THE DEVICE REBOOTS ###

00:00:00.032 CFG: Loaded from flash at F7, Count 319
00:00:00.035 QPC: Count 1
00:00:00.038 ZIG: ZBBridge EEPROM found at address 0x50
00:00:00.039 ZIG: GPIOs Rx:3 Tx:1
00:00:00.039 SNS: Hardware Serial
00:00:00.040 SRC: Restart
00:00:00.041 Project tasmota Tasmota Version 9.3.0.1(zbbridge)-2_7_4_9(2021-02-22T19:37:25)
00:00:00.042 ZIG: Resetting EZSP device
00:00:00.192 WIF: Checking connection...
00:00:00.192 WIF: Attempting connection...
00:00:00.529 WIF: Connecting to AP1 J&M IOT Channel 5 BSSId 0E:EC:DA:8D:48:02 in mode 11n as TASMOTA_ZIGBEE...
00:00:01.753 WIF: Checking connection...
00:00:01.753 WIF: Connected
00:00:02.003 HTP: Web server active on TASMOTA_ZIGBEE with IP address 172.30.31.54
00:00:02.157 HTP: Console
00:00:02.508 RTC: UTC 2021-02-23T14:32:41, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
15:32:42.011 MQT: Attempting connection...
15:32:42.025 MQT: Connected
15:32:42.027 MQT: home/TASMOTA_ZIGBEE/tele/LWT = Online (retained)
15:32:42.030 MQT: home/TASMOTA_ZIGBEE/cmnd/POWER = 
15:32:42.031 MQT: Subscribe to home/TASMOTA_ZIGBEE/cmnd/#
15:32:42.033 MQT: Subscribe to home/tasmotas/cmnd/#
15:32:42.035 MQT: Subscribe to cmnd/DVES_9BA274_fb/#
15:32:42.037 MQT: home/TASMOTA_ZIGBEE/tele/INFO1 = {"Module":"Sonoff ZbBridge","Version":"9.3.0.1(zbbridge)","FallbackTopic":"cmnd/DVES_9BA274_fb/","GroupTopic":"home/tasmotas/cmnd/"} (retained)
15:32:42.038 MQT: home/TASMOTA_ZIGBEE/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"TASMOTA_ZIGBEE","IPAddress":"172.30.31.54"} (retained)
15:32:42.039 MQT: home/TASMOTA_ZIGBEE/tele/INFO3 = {"RestartReason":"Software/System restart"} (retained)
15:32:45.460 QPC: Reset

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Flash and configure a Sonoff zbBridge device
2.Change gpio 13 in template configuration to ledlink instead of ledlink_i
3. See that the blue led stays of

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The blue led is on

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-02-23:
Did you use `Module 0`?

According to the logs you are still using `module 75` which is pre-defined and cannot be changed

jeroenst, 2021-02-23:
No I use module 75. If it is predefined and can't be overridden it shouldn't be able to change the values because now (me as) a user thinks it's not working while it seems configured correctly.

jeroenst, 2021-02-23:
When configuring module 0 it works.

What's the additional value of module 75, for me it's only confusing when settings are not honored.

ascillato2, 2021-02-23:
> When configuring module 0 it works.

Great to know that now it is working for you.

closed 2021-02-23 by ascillato2

jeroenst, 2021-02-23:
Working for me, but still confusing, maybe update code to prevent other users to experience this issue? Or change documentation? Or just wait for the next person to create this issue to again?

s-hadinger, 2021-02-23:
I'll submit to the dev team if we can make the gui more explicit

jeroenst, 2021-02-23:
I updated documentation: https://github.com/blakadder/zigbee/compare/master...jeroenst:patch-2