arendst/Tasmota #11117: DS18B20 sensor not working after FW 9.3.1

tiger42, 2021-02-24:
### PROBLEM DESCRIPTION
After updating an Sonoff S20 with connected DS18B20 sensor (GPIO1) from FW 9.3.0 to 9.3.1 (I've also tried 9.3.1.1), the sensor is not recognized any more. Both "tasmota" release, not "sensors"!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S20
- [x] Tasmota binary firmware version number used: 9.3.1 and 9.3.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
14:27:36.411 MQT: stat/steckdose-erker/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
14:27:36.621 MQT: stat/steckdose-erker/RESULT = {"Module":{"8":"Sonoff S2X"}}
14:27:36.874 MQT: stat/steckdose-erker/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"1312":"DS18x20"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```

- [ ] Provide the output of this command: `Status 0`:
```lua
13:46:31.502 MQT: stat/steckdose-erker/STATUS = {"Status":{"Module":8,"DeviceName":"Sonoff Erker","FriendlyName":["Sonoff Erker"],"Topic":"steckdose-erker","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:46:31.509 MQT: stat/steckdose-erker/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:00:16","StartupUTC":"2021-02-24T12:46:15","Sleep":50,"CfgHolder":4617,"BootCount":64,"BCResetTime":"2020-03-21T23:13:41","SaveCount":333,"SaveAddress":"FA000"}}
13:46:31.515 MQT: stat/steckdose-erker/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021.02.23 12:08:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"465/699"}}
13:46:31.522 MQT: stat/steckdose-erker/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Muffel.Netz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AA00B","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
13:46:31.534 MQT: stat/steckdose-erker/STATUS4 = {"StatusMEM":{"ProgramSize":594,"Free":408,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:46:31.543 MQT: stat/steckdose-erker/STATUS5 = {"StatusNET":{"Hostname":"steckdose-erker-0497","IPAddress":"192.168.0.35","Gateway":"192.168.0.100","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.100","Mac":"80:7D:3A:67:E1:F1","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:46:31.551 MQT: stat/steckdose-erker/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.1","MqttPort":1883,"MqttClientMask":"steckdose-erker_%06X","MqttClient":"steckdose-erker_67E1F1","MqttUser":"hagenau","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:46:31.560 MQT: stat/steckdose-erker/STATUS7 = {"StatusTIM":{"UTC":"2021-02-24T12:46:31","Local":"2021-02-24T13:46:31","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"07:41","Sunset":"18:25"}}
13:46:31.567 MQT: stat/steckdose-erker/STATUS10 = {"StatusSNS":{"Time":"2021-02-24T13:46:31"}}
13:46:31.573 MQT: stat/steckdose-erker/STATUS11 = {"StatusSTS":{"Time":"2021-02-24T13:46:31","Uptime":"0T00:00:16","UptimeSec":16,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Muffel.Netz","BSSId":"14:59:C0:98:7F:5A","Channel":1,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

### TO REPRODUCE
Flash the device with FW 9.3.1 tasmota release. Look in the Web-UI or MQTT output.

### EXPECTED BEHAVIOUR
The output contains the sensor readings.

Jason2866, 2021-02-24:
There is no general issue. Mine work fine. Please test with a device that has not a power supply being on the edge. For example with a NodeMCU

sfromis, 2021-02-24:
Discord chat ended with it for some weird reason working with tasmota-sensors 9.3.1, including warnings about the possible power supply issues with devices not designed for expansion.

Nenodema, 2021-02-24:
Sorry, i see that it is closed but i have exactly the same problem with my Sonoff TH16, i just flashed back to 9.1 and working fine as expected.

kristofejro, 2021-02-24:
For me the same. All of my DS18B20 on Sonoff Basic stopped working on 9.3.1.
After downgrade to 9.3.0 all is working back again.

nhsqr, 2021-02-25:
> Sorry, i see that the issue is closed but i have exactly the same problem with my Sonoff TH16, i just flashed back to 9.1 and working fine as expected.

Same here

fpacardoso, 2021-03-01:
Same here.
Sonoff TH with DS18B20 don't show data.

sfromis, 2021-03-01:
Did you try the workaround of using tasmota-sensors.bin.gz?

fpacardoso, 2021-03-01:
> 
> 
> Did you try the workaround of using tasmota-sensors.bin.gz?

Yes, the same issue.

kristofejro, 2021-03-01:
I can confirm that tasmota-sensors.bin.gz solves the issue (at least for me)
All of my DS18B20 on Sonoff Basic work on 9.3.1 tasmota-sensors.bin.gz

Nenodema, 2021-03-01:
Thanks, is this "works as designed"? *Can we re-open this issue?

fpacardoso, 2021-03-01:
> 
> 
> Thanks, is this "works as designed"? *Can we re-open this issue?

I think that isn't work as designed...
I have Sonoff TH with tasmota.bin and don't have this problem until 9.3.1
Please reopen this issue to solve this

ascillato2, 2021-03-01:
> Please reopen this issue to solve this

Reopening as requested.

andre1210, 2021-03-02:
I was using 9.2.0 for some time on an ESP8266 with my DS18B20.
After OTA update to 9.3.1 the DS18B20 stopped working.

MarcoCLA, 2021-03-02:
If I build tag v9.3.1 https://github.com/arendst/Tasmota/commit/3f7071c6b9caaf3ace325c096f2e201a0a1f11df myself, the bin differs from the released one somehow. And the temps are shown.

ascillato, 2021-03-02:
> If I build tag v9.3.1 3f7071c myself, the bin differs from the released one somehow. And the temps are shown.

@everyone

Please, can you try latest development version: http://ota.tasmota.com/tasmota/tasmota.bin.gz

Thanks.

MarcoCLA, 2021-03-02:
@ascillato I tried, it works.

tiger42, 2021-03-02:
I can also confirm that this version works. I've tried the German version (tasmota-DE.bin.gz).

fpacardoso, 2021-03-02:
@ascillato, Tested and it works!

andre1210, 2021-03-02:
I confirm too, it works

nagyrobi, 2021-03-02:
What was the issue?

ascillato2, 2021-03-02:
Thanks everyone for reporting and tested 👍

ascillato2, 2021-03-02:
> What was the issue?

Just a compilation issue.

closed 2021-03-02 by ascillato2

Nenodema, 2021-03-02:
@ascillato2 

Thanks!

arendst, 2021-03-09:
See #11270 

New binaries have been released.