arendst/Tasmota #11118: PZEM-004 Energy Monitoring not working after update to 9.3.1

valekrus, 2021-02-24:
### PROBLEM DESCRIPTION
After updating my custom ESP8266 board with connected PZEM-004 (GPIO1 - PZEM0XX Tx, GPIO3 - PZEM004 Rx) from FW 9.2.0 to 9.3.1 (also tried 9.3.1.1), there are no measurements from PZEM any more. Only zero values. Tested on "tasmota-sensors" fw.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): custom ESP8266 board
- [x] Tasmota binary firmware version number used: 9.3.1 and dev build 9.3.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
17:38:54.984 CMD: Backlog Template; Module; GPIO 255
17:38:55.104 MQT: stat/PowerMeter/RESULT = {"NAME":"","GPIO":[608,1472,0,1504,640,1312,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
17:38:55.155 MQT: tele/PowerMeter/STATE = {"Time":"2021-02-24T17:38:55","Uptime":"0T00:11:52","UptimeSec":712,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"V@LEK.NET","BSSId":"58:8B:F3:66:AD:E8","Channel":2,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:04"}}
17:38:55.173 MQT: tele/PowerMeter/SENSOR = {"Time":"2021-02-24T17:38:55","ENERGY":{"TotalStartTime":"2021-02-19T00:00:00","Total":82.217,"Yesterday":16.077,"Today":0.166,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"DS18B20":{"Id":"0516823628FF","Temperature":9.3},"BME280":{"Temperature":-8.7,"Humidity":45.1,"DewPoint":-18.5,"Pressure":775.6},"PressureUnit":"mmHg","TempUnit":"C"}
17:38:55.305 MQT: stat/PowerMeter/RESULT = {"Module":{"0":""}}
17:38:55.511 MQT: stat/PowerMeter/RESULT = {"GPIO0":{"608":"I2C SCL"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"0":"None"},"GPIO3":{"1504":"PZEM004 Rx"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"1312":"DS18x20"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
17:40:49.682 CMD: Backlog Rule1; Rule2; Rule3
17:40:49.777 MQT: stat/PowerMeter/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":439,"Free":72,"Rules":"on DS18B20#Temperature!=%var1% do var1 %value% endon on BME280#Temperature!=%var2% do var2 %value% endon on BME280#Humidity!=%var3% do var3 %value% endon on BME280#DewPoint!=%var4% do var4 %value% endon on BME280#Pressure!=%var5% do var5 %value% endon on Energy#Voltage!=%var6% do var6 %value% endon on Energy#Current!=%var7% do var7 %value% endon on Energy#Power!=%var8% do var8 %value% endon on Energy#Total!=%var9% do var9 %value% endon"}
17:40:49.978 MQT: stat/PowerMeter/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":462,"Free":49,"Rules":"on Time#Minute|5 do backlog publish narodmon/PowerMeter/sensors/DS18T1 %var1%; publish narodmon/PowerMeter/sensors/T1 %var2%; publish narodmon/PowerMeter/sensors/H1 %var3%; publish narodmon/PowerMeter/sensors/DPT1 %var4%; publish narodmon/PowerMeter/sensors/P1 %var5%; publish narodmon/PowerMeter/sensors/U4 %var6%; publish narodmon/PowerMeter/sensors/I4 %var7%; publish narodmon/PowerMeter/sensors/W4 %var8%; publish narodmon/PowerMeter/sensors/W5 %var9%; endon"}
17:40:50.172 MQT: stat/PowerMeter/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
```
- [x] Provide the output of this command: `Status 0`:
```lua
17:41:42.150 CMD: Status 0
17:41:42.158 MQT: stat/PowerMeter/STATUS = {"Status":{"Module":0,"DeviceName":"PowerMeter","FriendlyName":["PowerMeter"],"Topic":"PowerMeter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:41:42.164 MQT: stat/PowerMeter/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:14:39","StartupUTC":"2021-02-24T14:27:03","Sleep":50,"CfgHolder":4617,"BootCount":249,"BCResetTime":"2020-04-05T13:32:13","SaveCount":7374,"SaveAddress":"F8000"}}
17:41:42.170 MQT: stat/PowerMeter/STATUS2 = {"StatusFWR":{"Version":"9.3.1.1(sensors)","BuildDateTime":"2021-02-24T10:00:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
17:41:42.175 MQT: stat/PowerMeter/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["V@LEK.NET",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["01008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
17:41:42.187 MQT: stat/PowerMeter/STATUS4 = {"StatusMEM":{"ProgramSize":645,"Free":356,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","20000000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
17:41:42.195 MQT: stat/PowerMeter/STATUS5 = {"StatusNET":{"Hostname":"PowerMeter","IPAddress":"192.168.100.253","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"18:FE:34:D7:D7:D4","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:41:42.200 MQT: stat/PowerMeter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D7D7D4","MqttUser":"esp8266","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
17:41:42.207 MQT: stat/PowerMeter/STATUS7 = {"StatusTIM":{"UTC":"2021-02-24T14:41:42","Local":"2021-02-24T17:41:42","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+03:00","Sunrise":"09:41","Sunset":"20:25"}}
17:41:42.211 MQT: stat/PowerMeter/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:41:42.282 MQT: stat/PowerMeter/STATUS10 = {"StatusSNS":{"Time":"2021-02-24T17:41:42","ENERGY":{"TotalStartTime":"2021-02-19T00:00:00","Total":82.217,"Yesterday":16.077,"Today":0.166,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"DS18B20":{"Id":"0516823628FF","Temperature":9.2},"BME280":{"Temperature":-8.8,"Humidity":45.7,"DewPoint":-18.5,"Pressure":775.5},"PressureUnit":"mmHg","TempUnit":"C"}}
17:41:42.288 MQT: stat/PowerMeter/STATUS11 = {"StatusSTS":{"Time":"2021-02-24T17:41:42","Uptime":"0T00:14:39","UptimeSec":879,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Wifi":{"AP":1,"SSId":"V@LEK.NET","BSSId":"58:8B:F3:66:AD:E8","Channel":2,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### TO REPRODUCE
Flash the device with FW 9.3.1 tasmota release. Look in the Web-UI or MQTT output.

### EXPECTED BEHAVIOUR
The output contains the actual measurements from PZEM.

### SCREENSHOTS
[image]

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-02-24:
Hi,

The most important piece of information to troubleshoot your issue is missing from the part of the template you have deleted.

Please, set weblog to 4 for more debug information and the share the output of the console when your device starts.

Thanks.

valekrus, 2021-02-24:
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.050 CFG: Loaded from flash at F4, Count 7378
00:00:00.055 QPC: Count 1
00:00:00.061 SRC: Restart
00:00:00.063 Project tasmota PowerMeter Version 9.3.1.1(sensors)-2_7_4_9(2021-02-24T10:00:44)
00:00:00.108 SNS: Hardware Serial
00:00:00.124 DSB: Sensors found 1
00:00:00.136 SHT: Sensor did not ACK command
00:00:00.175 I2C: BME280 found at 0x76
00:00:00.438 WIF: Checking connection...
00:00:00.439 WIF: Attempting connection...
00:00:00.775 WIF: Connecting to AP1 V@LEK.NET Channel 2 BSSId 58:8B:F3:66:AD:E8 in mode 11n as PowerMeter...
00:00:01.502 WIF: Checking connection...
00:00:01.503 WIF: Attempting connection...
00:00:02.502 WIF: Checking connection...
00:00:02.503 WIF: Connected
00:00:02.754 HTP: Web server active on PowerMeter with IP address 192.168.100.253
00:00:03.249 RTC: UTC 2021-02-24T16:08:17, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
19:08:17.001 NRG: Energy reset by 4 seconds invalid data
19:08:17.083 RUL: ENERGY#TOTAL!=%VAR9% performs "var9 82.217"
```

ascillato, 2021-02-24:
Your PZEM is not sending any data. Check your wiring and your gpio assignments.

valekrus, 2021-02-24:
Ok, i will check it later. But it was working perfectly before update (tasmota 9.2.0). Currently it's difficult to do it, because this board is in outdoor case connected to my home mains input. To disassemble it i need to go outdoor (it's winter here now, so it's pretty cold and wet there), turn off power completletly and dismount it. Maybe there are some other ways to check? For example, can i try to OTA update it to previous version?

Jason2866, 2021-02-24:
Downgrade is not supported. Since your version you wanna try is still 9.x it will probably work.

ascillato, 2021-02-24:
Be aware that downgrading can left your device unusable as it is not supported.

kristofejro, 2021-02-24:
You can try downgrade to 9.3.0.
All of my DS18B20 on Sonoff Basic stopped working on 9.3.1.
After downgrade to 9.3.0 (first minimal -> then full) all is working back again.
But it is just my case :-)

Jason2866, 2021-02-27:
Could not reproduce.

[image]

```
15:53:29.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:29.313 CFG: Saved to flash at F7, Count 251, Bytes 4096
15:53:30.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:31.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:32.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:33.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:34.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:34.478 MQT: tele/sonoff-3193C7/STATE = {"Time":"2021-02-27T15:53:34","Uptime":"0T00:04:54","UptimeSec":294,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:08"}}
15:53:34.489 MQT: tele/sonoff-3193C7/SENSOR = {"Time":"2021-02-27T15:53:34","ENERGY":{"TotalStartTime":"2019-10-30T10:24:27","Total":0.006,"Yesterday":0.000,"Today":0.002,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":50.0,"Voltage":230,"Current":0.000}}
15:53:35.201 DMP: 01 04 14 08 FA 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 DD 93
15:53:36.201 DMP: 01 04 14 08 F9 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 99 D7
15:53:37.201 DMP: 01 04 14 08 FA 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 DD 93
15:53:38.201 DMP: 01 04 14 08 FA 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 DD 93
15:53:39.201 DMP: 01 04 14 08 F7 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 F3 3E
15:53:40.201 DMP: 01 04 14 08 F5 00 00 00 00 00 00 00 00 00 51 00 00 01 F4 00 00 00 00 8A 86
15:53:41.201 DMP: 01 04 14 08 F3 00 8E 00 00 01 30 00 00 00 51 00 00 01 F4 00 5D 00 00 D7 66
15:53:42.201 DMP: 01 04 14 08 F2 00 A7 00 00 01 7E 00 00 00 51 00 00 01 F4 00 64 00 00 1D 41
15:53:43.201 DMP: 01 04 14 08 F2 00 A7 00 00 01 7E 00 00 00 51 00 00 01 F4 00 64 00 00 1D 41
15:53:44.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 51 00 00 01 F4 00 64 00 00 22 3C
15:53:44.478 MQT: tele/sonoff-3193C7/STATE = {"Time":"2021-02-27T15:53:44","Uptime":"0T00:05:04","UptimeSec":304,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:08"}}
15:53:44.490 MQT: tele/sonoff-3193C7/SENSOR = {"Time":"2021-02-27T15:53:44","ENERGY":{"TotalStartTime":"2019-10-30T10:24:27","Total":0.006,"Yesterday":0.000,"Today":0.002,"Period":0,"Power":38,"ApparentPower":38,"ReactivePower":0,"Factor":1.00,"Frequency":50.0,"Voltage":229,"Current":0.167}}
15:53:45.201 DMP: 01 04 14 08 F5 00 A7 00 00 01 80 00 00 00 51 00 00 01 F4 00 64 00 00 FE 4A
15:53:46.201 DMP: 01 04 14 08 F6 00 A7 00 00 01 80 00 00 00 51 00 00 01 F4 00 64 00 00 BA 0E
15:53:47.201 DMP: 01 04 14 08 F6 00 A7 00 00 01 80 00 00 00 51 00 00 01 F4 00 64 00 00 BA 0E
15:53:48.201 DMP: 01 04 14 08 F6 00 A7 00 00 01 80 00 00 00 51 00 00 01 F4 00 64 00 00 BA 0E
15:53:48.501 WIF: Checking connection...
15:53:49.201 DMP: 01 04 14 08 F6 00 A7 00 00 01 7F 00 00 00 51 00 00 01 F4 00 64 00 00 EE F0
15:53:50.201 DMP: 01 04 14 08 F6 00 A7 00 00 01 80 00 00 00 51 00 00 01 F4 00 64 00 00 BA 0E
15:53:51.201 DMP: 01 04 14 08 F5 00 A7 00 00 01 7F 00 00 00 51 00 00 01 F4 00 64 00 00 AA B4
15:53:52.201 DMP: 01 04 14 08 F5 00 A7 00 00 01 7F 00 00 00 51 00 00 01 F4 00 64 00 00 AA B4
15:53:53.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 51 00 00 01 F4 00 64 00 00 22 3C
15:53:54.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 51 00 00 01 F4 00 64 00 00 22 3C
15:53:54.478 MQT: tele/sonoff-3193C7/STATE = {"Time":"2021-02-27T15:53:54","Uptime":"0T00:05:14","UptimeSec":314,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"88:C3:97:B1:1D:56","Channel":4,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:08"}}
15:53:54.488 MQT: tele/sonoff-3193C7/SENSOR = {"Time":"2021-02-27T15:53:54","ENERGY":{"TotalStartTime":"2019-10-30T10:24:27","Total":0.006,"Yesterday":0.000,"Today":0.002,"Period":0,"Power":38,"ApparentPower":38,"ReactivePower":0,"Factor":1.00,"Frequency":50.0,"Voltage":229,"Current":0.167}}
15:53:55.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 36 CC
15:53:56.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 36 CC
15:53:57.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 36 CC
15:53:58.201 DMP: 01 04 14 08 F2 00 A7 00 00 01 7E 00 00 00 52 00 00 01 F4 00 64 00 00 09 B1
15:53:59.201 DMP: 01 04 14 08 F2 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 0B 30
15:54:00.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 36 CC
15:54:01.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 36 CC
15:54:02.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7E 00 00 00 52 00 00 01 F4 00 64 00 00 34 4D
15:54:03.201 DMP: 01 04 14 08 F5 00 A8 00 00 01 80 00 00 00 52 00 00 01 F4 00 64 00 00 DF 49
15:54:04.201 DMP: 01 04 14 08 F5 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 BE 44
[... cut ...] "Voltage":229,"Current":0.167}}
15:54:15.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 36 CC
15:54:16.201 DMP: 01 04 14 08 F3 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F3 00 64 00 00 83 0C
15:54:17.201 DMP: 01 04 14 08 F4 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 83 B8
15:54:18.201 DMP: 01 04 14 08 F4 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F4 00 64 00 00 83 B8
15:54:19.201 DMP: 01 04 14 08 F4 00 A7 00 00 01 7F 00 00 00 52 00 00 01 F3 00 64 00 00 36 78
```

ascillato2, 2021-02-27:
I couldn't reproduce it either. Please, check that the configuration of GPIOs is correct. From your logs, your PZEM is not transmitting data to your Tasmota device.

ascillato2, 2021-03-02:
Please, when you have time and you can check the wiring, tell us if you could solve your issue. If not, please do not hesitate on asking to reopen. Thanks.

closed 2021-03-02 by ascillato2

valekrus, 2021-03-13:
After short power outage it's working now .So, maybe reboot of PZEM needed to work after update, or it was just PZEM glitch. Thank you for support!

JameZUK, 2021-04-20:
I have put one of these together today with a Sonoff TH module. It works fine for a but and then starts reading zero like this bug report. I will attempt to capture what I find and update this when I do.

JameZUK, 2021-04-20:
Update, 9.3.0 made zero difference so I enabled syslog and captured a failure. It seemed to stop working @ 17:11:45

```
18:42:01.056 CMD: Backlog Template; Module; GPIO 255
18:42:01.059 SRC: WebConsole from 192.168.1.227
18:42:01.062 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
18:42:01.140 SRC: Backlog
18:42:01.143 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
18:42:01.151 MQT: stat/tasmota_003E70/RESULT = {"NAME":"HW-655 PZEM","GPIO":[0,0,0,0,1536,0,0,0,0,0,1472,0,0,0],"FLAG":0,"BASE":1}
18:42:01.200 DMP: 01 04 14 08 78 1E 7E 00 00 3E D8 00 00 24 0C 00 00 00 F4 00 54 00 00 1E 28
18:42:01.203 PAC: PzemAc 1 error 9
18:42:01.368 SRC: Backlog
18:42:01.371 CMD: Group 0, Index 1, Command "MODULE", Data ""
18:42:01.377 MQT: stat/tasmota_003E70/RESULT = {"Module":{"4":"Sonoff TH"}}
18:42:01.519 WIF: Checking connection...
18:42:01.618 SRC: Backlog
18:42:01.621 CMD: Group 0, Index 1, Command "GPIO", Data "255"
```

```
18:43:51.032 CMD: Status 0
18:43:51.035 SRC: WebConsole from 192.168.1.227
18:43:51.038 CMD: Group 0, Index 1, Command "STATUS", Data "0"
18:43:51.045 MQT: stat/tasmota_003E70/STATUS = {"Status":{"Module":4,"DeviceName":"PowerMeter","FriendlyName":["PowerMeter"],"Topic":"tasmota_003E70","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:43:51.052 MQT: stat/tasmota_003E70/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:46:35","StartupUTC":"2021-04-20T16:57:16","Sleep":50,"CfgHolder":4617,"BootCount":44,"BCResetTime":"2021-04-20T10:32:13","SaveCount":108,"SaveAddress":"F7000"}}
18:43:51.061 MQT: stat/tasmota_003E70/STATUS2 = {"StatusFWR":{"Version":"9.3.1.4(tasmota)","BuildDateTime":"2021-04-20T12:21:52","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"429/699"}}
18:43:51.069 MQT: stat/tasmota_003E70/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"MqttLog":4,"SysLog":4,"LogHost":"syslog","LogPort":514,"SSId":["MEH",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
18:43:51.082 MQT: stat/tasmota_003E70/STATUS4 = {"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
18:43:51.092 MQT: stat/tasmota_003E70/STATUS5 = {"StatusNET":{"Hostname":"PowerMeter","IPAddress":"192.168.1.247","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"REMOVED","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:43:51.100 MQT: stat/tasmota_003E70/STATUS6 = {"StatusMQT":{"MqttHost":"domoticz","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_003E70","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:43:51.109 MQT: stat/tasmota_003E70/STATUS7 = {"StatusTIM":{"UTC":"2021-04-20T17:43:51","Local":"2021-04-20T18:43:51","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:48"}}
18:43:51.115 MQT: stat/tasmota_003E70/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:43:51.125 MQT: stat/tasmota_003E70/STATUS10 = {"StatusSNS":{"Time":"2021-04-20T18:43:51","ENERGY":{"TotalStartTime":"2021-04-20T10:32:13","Total":4.421,"Yesterday":0.000,"Today":4.421,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
18:43:51.134 MQT: stat/tasmota_003E70/STATUS11 = {"StatusSTS":{"Time":"2021-04-20T18:43:51","Uptime":"0T00:46:35","UptimeSec":2795,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MEH","BSSId":"REMOVED","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":2,"Downtime":"0T00:00:05"}}}
```

```
Apr 20 17:11:44 PowerMeter ESP-PAC: PzemAc 1 error 9
Apr 20 17:12:58 PowerMeter ESP-PAC: PzemAc 1 error 9
Apr 20 17:13:02 PowerMeter ESP-PAC: PzemAc 1 error 9
Apr 20 17:13:23 PowerMeter ESP-PAC: PzemAc 1 error 9
Apr 20 17:13:33 PowerMeter ESP-PAC: PzemAc 1 error 9
Apr 20 17:13:35 PowerMeter ESP-PAC: PzemAc 1 error 9
Apr 20 17:13:39 PowerMeter ESP-PAC: PzemAc 1 error 9
```

```
Apr 20 17:10:20 PowerMeter ESP-DMP: 01 04 14 09 6B 71 3A 00 00 0C 90 00 01 1E E5 00 00 01 F4 00 62 00 00 46 31
Apr 20 17:10:21 PowerMeter ESP-DMP: 01 04 14 09 6C 71 21 00 00 0C 5F 00 01 1E E7 00 00 01 F4 00 62 00 00 85 DC
Apr 20 17:10:22 PowerMeter ESP-DMP: 01 04 14 09 6C 71 25 00 00 0C 71 00 01 1E E9 00 00 01 F4 00 62 00 00 E2 93
Apr 20 17:10:23 PowerMeter ESP-DMP: 01 04 14 09 6D 71 48 00 00 0C DB 00 01 1E EB 00 00 01 F4 00 62 00 00 E2 29
Apr 20 17:10:24 PowerMeter ESP-DMP: 01 04 14 09 6D 71 46 00 00 0C D4 00 01 1E ED 00 00 01 F4 00 62 00 00 B9 E5
Apr 20 17:10:25 PowerMeter ESP-DMP: 01 04 14 09 6D 71 46 00 00 0C D4 00 01 1E EF 00 00 01 F4 00 62 00 00 A0 85
[... cut ...]
Apr 20 17:12:17 PowerMeter ESP-DMP: 01 04 14 09 7B 34 CD 00 00 77 D9 00 00 1F 5C 00 00 01 F4 00 5E 00 00 23 39
Apr 20 17:12:18 PowerMeter ESP-DMP: 01 04 14 09 7B 34 C9 00 00 77 D8 00 00 1F 5D 00 00 01 F4 00 5E 00 00 6E E9
Apr 20 17:12:19 PowerMeter ESP-DMP: 01 04 14 09 7B 34 C9 00 00 77 D8 00 00 1F 5E 00 00 01 F4 00 5E 00 00 7A 19
Apr 20 17:12:20 PowerMeter ESP-DMP: 01 04 14 09 7A 34 CA 00 00 77 C1 00 00 1F 5E 00 00 01 F4 00 5E 00 00 94 4C
Apr 20 17:12:21 PowerMeter ESP-DMP: 01 04 14 09 7A 34 BA 00 00 77 90 00 00 1F 5F 00 00 01 F4 00 5D 00 00 8E D6
```